thingsboard/thingsboard-gateway #165: Does not find org.thingsboard.common depenencey when compiling the project

Micckii, 2019-01-24:
IntelliJ IDEA does not find this depenencey when compiling the project
<dependency>
            <groupId>org.thingsboard.common</groupId>
            <artifactId>data</artifactId>
            <version>${thingsboard.version}</version>
        </dependency>
        <dependency>
            <groupId>org.thingsboard</groupId>
            <artifactId>netty-mqtt</artifactId>
            <version>${thingsboard.version}</version>
        </dependency>
[ERROR] Failed to execute goal on project gateway: Could not resolve dependencies for project org.thingsboard:gateway:jar:2.2.
1-SNAPSHOT: The following artifacts could not be resolved: org.thingsboard.common:data:jar:2.2.1-SNAPSHOT, org.thingsboard:net
ty-mqtt:jar:2.2.1-SNAPSHOT: Could not find artifact org.thingsboard.common:data:jar:2.2.1-SNAPSHOT -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.

mp-loki, 2019-01-24:
Please build the (main project)[https://github.com/thingsboard/thingsboard] first

Micckii, 2019-01-25:
It worked. Thank you

closed 2019-01-25 by Micckii

affoltep, 2019-09-18:
@mp-loki : 
I get the following similar error message:
````
[INFO] Scanning for projects...
[INFO]
[INFO] ----------------------< org.thingsboard:gateway >-----------------------
[INFO] Building Thingsboard Gateway 2.2.1-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
[WARNING] The POM for org.thingsboard.common:data:jar:2.2.1-SNAPSHOT is missing, no dependency information available
[WARNING] The POM for org.thingsboard:netty-mqtt:jar:2.2.1-SNAPSHOT is missing, no dependency information available
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  1.194 s
[INFO] Finished at: 2019-09-18T23:09:42+02:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal on project gateway: Could not resolve dependencies for project org.thingsboard:gateway:jar:2.2.1-SNAPSHOT: The following artifacts could not be resolved: org.thingsboard.common:data:jar:2.2.1-SNAPSHOT, org.thingsboard:netty-mqtt:jar:2.2.1-SNAPSHOT: Could not find artifact org.thingsboard.common:data:jar:2.2.1-SNAPSHOT -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/DependencyResolutionException
````
I could successful compile thingsboard 2.4.1 and now I try to compile the thingsboard-gateway 2.2.1rc but it fails. 
I would like to get a docker image, finally. Do I need to use 
mvn clean install -DskipTests -Ddockerfile.skip=false?

zouhr, 2019-09-19:
"I could successful compile thingsboard 2.4.1 and now I try to compile the thingsboard-gateway 2.2.1rc but it fails."

i have the same problem,  how to resolve ?

affoltep, 2019-09-19:
@zouhr - I had to change the thingsboard version in pom.xml on row 59:
````
       <thingsboard.version>2.4.1</thingsboard.version>
````

zouhr, 2019-09-19:
it work,thank you 

   <thingsboard.version>2.4.1-SNAPSHOT</thingsboard.version>

> @zouhr - I had to change the thingsboard version in pom.xml on row 59:
> 
> ```
>        <thingsboard.version>2.4.1</thingsboard.version>
> ```

martindenion, 2019-10-04:
Hi,
I have the same problem but when I change <thingsboard.version>2.4.1</thingsboard.version> in pom.xml it doesn't work.