thingsboard/thingsboard-gateway #170: Gateway 2.1.1.rc crashed

JozsefGuyas, 2019-02-14:
Hi!

I am using tb gateway 2.1.1.rc release, everything went fine, once gateway crashed. I checked the the service. It said  gateway was dead.
I have found this in the log file.

2019-02-14 01:39:46,493 [nioEventLoopGroup-2-1] WARN  o.t.g.s.gateway.MqttGatewayService - Lost connection to ThingsBoard.
2019-02-14 01:39:46,589 [MQTT Rec: cd4e6cce-50e1-4b83-b0d8-81d2291215f7] WARN  o.t.g.e.m.client.MqttBrokerMonitor - [localhost:1883] MQTT broker connection lost!
2019-02-14 01:39:46,648 [MQTT Rec: cd4e6cce-50e1-4b83-b0d8-81d2291215f7] WARN  o.t.g.e.m.client.MqttBrokerMonitor - [localhost:1883] MQTT broker connection failed!
org.eclipse.paho.client.mqttv3.MqttException: MqttException
	at org.eclipse.paho.client.mqttv3.internal.Token.waitForResponse(Token.java:148)
	at org.eclipse.paho.client.mqttv3.internal.Token.waitForCompletion(Token.java:108)
	at org.eclipse.paho.client.mqttv3.MqttToken.waitForCompletion(MqttToken.java:63)
	at org.thingsboard.gateway.extensions.mqtt.client.MqttBrokerMonitor.checkConnection(MqttBrokerMonitor.java:134)
	at org.thingsboard.gateway.extensions.mqtt.client.MqttBrokerMonitor.connectionLost(MqttBrokerMonitor.java:355)
	at org.eclipse.paho.client.mqttv3.internal.CommsCallback.connectionLost(CommsCallback.java:292)
	at org.eclipse.paho.client.mqttv3.internal.ClientComms.shutdownConnection(ClientComms.java:423)
	at org.eclipse.paho.client.mqttv3.internal.CommsReceiver.run(CommsReceiver.java:181)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.InterruptedException: null
	at java.lang.Object.wait(Native Method)
	at java.lang.Object.wait(Object.java:502)
	at org.eclipse.paho.client.mqttv3.internal.Token.waitForResponse(Token.java:143)
	... 14 common frames omitted
2019-02-14 01:39:46,650 [MQTT Con: cd4e6cce-50e1-4b83-b0d8-81d2291215f7] WARN  o.t.g.e.m.client.MqttBrokerMonitor - [localhost:1883] MQTT broker connection faied! Unable to connect to server
org.eclipse.paho.client.mqttv3.MqttException: Unable to connect to server
	at org.eclipse.paho.client.mqttv3.internal.TCPNetworkModule.start(TCPNetworkModule.java:94)
	at org.eclipse.paho.client.mqttv3.internal.ClientComms$ConnectBG.run(ClientComms.java:701)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.net.ConnectException: Connection refused (Connection refused)
	at java.net.PlainSocketImpl.socketConnect(Native Method)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at org.eclipse.paho.client.mqttv3.internal.TCPNetworkModule.start(TCPNetworkModule.java:84)
	... 8 common frames omitted

mp-loki, 2019-02-14:
The error log says it cannot connect to the external MQTT broker which is run on localhost:1883. 
Can you check that the broker on this port is up and running?

Also, The use of tb-gateway version 2.1.1 is discouraged, please use gateway version 2.2.1

JozsefGuyas, 2019-02-14:
I have checked my external broker, and it ran when gateway stopped. When I restarted the gateway it worked again.

mp-loki, 2019-02-14:
Ok please us know if you have any other issues

closed 2019-02-14 by mp-loki