thingsboard/thingsboard-gateway #171: Sigfox extension: confusing documentation

mfalkvidd, 2019-02-22:
https://thingsboard.io/docs/iot-gateway/sigfox/ says "Use general configuration to enable this extension." and links to https://thingsboard.io/docs/iot-gateway/configuration/

The general configuration at https://thingsboard.io/docs/iot-gateway/configuration/ says "SIGFOX Extension configuration is covered on the corresponding extension page." and links to https://thingsboard.io/docs/iot-gateway/sigfox/ creating a loop

https://thingsboard.io/docs/iot-gateway/getting-started/#step-10-connect-to-sigfox-backend says "Navigate to gateway configuration folder and edit tb-gateway.yml file" and "Change sigfox.enabled property value to true" but /etc/tb-gateway/conf/tb-gateway.yml does not contain anything about Sigfox.

Maybe the documentation can be cleaned up to only reference https://thingsboard.io/docs/iot-gateway/getting-started/#step-102-sigfox-extension-gui-configuration ?

yilmaznaslan, 2019-12-25:
Have you found a solution ?

zbeacon, 2020-01-02:
Hi @mfalkvidd, 

Thank you for your feedback. I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
 - Less footprint and memory impact
 - Easier to extend and customize
 - Improved functionality.

We don't have Sigfox protocol support in the new gateway. The Sigfox protocol configuration information has been deleted from the site, as it is not relevant. Perhaps we will make Sigfox protocol support in one of the following releases.

closed 2020-01-02 by zbeacon