thingsboard/thingsboard-gateway #175: Modbus TCP Server-side RPC

lazermen, 2019-03-14:
Не получается записать данные в устройство Modbus через Gateway (v2.2.1rc). Вместо ModBus-устройства использую симулятор mbslave. Данные с mbslave в ThingsBord (demo.thingsboard.io) попадают нормально, поэтому делаю вывод, что Gateway настроен правильно, хотя есть проблемы с попаданием данных в атрибуты [(Modbus TCP conficuration...).](https://github.com/thingsboard/thingsboard-gateway/issues/157) 
При попытки отправки RPC запроса через RPC call request (на входе данные: 
```
{
    "method": "write",
    "params": [{
        "tag": "WriteBigWord",
        "value": 123,
        "functionCode": 6,
        "address": 4,
        "byteOrder": "BIG"
    }]
}
```
метаданные:
```
{
    "deviceType": "gateway",
    "deviceName": "ModBus_Sensor",
    "ts": "1552564814437"
}
```
в журнале gateway (tb-gateway.log) появляется такая запись:

`2019-03-14 15:00:58,313 [pool-1-thread-3] ERROR o.t.g.s.gateway.MqttGatewayService - [ModBus_Sensor][13727] Failed to process rpc command
java.lang.RuntimeException: com.fasterxml.jackson.databind.JsonMappingException: No content to map due to end-of-input
 at [Source: ; line: 1, column: 0]
	at org.thingsboard.gateway.util.JsonTools.fromString(JsonTools.java:56)
	at org.thingsboard.gateway.extensions.modbus.rpc.RpcProcessor.onRpcCommand(RpcProcessor.java:74)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.lambda$null$23(MqttGatewayService.java:480)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
Caused by: com.fasterxml.jackson.databind.JsonMappingException: No content to map due to end-of-input
 at [Source: ; line: 1, column: 0]
	at com.fasterxml.jackson.databind.JsonMappingException.from(JsonMappingException.java:270)
	at com.fasterxml.jackson.databind.ObjectMapper._initForReading(ObjectMapper.java:3838)
	at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:3783)
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2861)
	at org.thingsboard.gateway.util.JsonTools.fromString(JsonTools.java:54)
	... 7 common frames omitted
`

если же отправлять запрос с dashboard через New RPC debug terminal 
[image]
, то в журнале появляется такая запись:

`2019-03-14 15:10:05,465 [pool-1-thread-6] ERROR o.t.g.s.gateway.MqttGatewayService - [ModBus_Sensor][13728] Failed to process rpc command
java.lang.RuntimeException: com.fasterxml.jackson.databind.JsonMappingException: Can not deserialize instance of org.thingsboard.gateway.extensions.modbus.conf.mapping.TagValueMapping out of START_ARRAY token
 at [Source: [[{tag:WriteCoil,functionCode:5,address:1,registerCount:1,byteOrder:BIG,bit:-1}]]; line: 1, column: 2] (through reference chain: java.util.ArrayList[0])
	at org.thingsboard.gateway.util.JsonTools.fromString(JsonTools.java:56)
	at org.thingsboard.gateway.extensions.modbus.rpc.RpcProcessor.onRpcCommand(RpcProcessor.java:74)
	at org.thingsboard.gateway.service.gateway.MqttGatewayService.lambda$null$23(MqttGatewayService.java:480)
	at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.util.concurrent.FutureTask.run(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
Caused by: com.fasterxml.jackson.databind.JsonMappingException: Can not deserialize instance of org.thingsboard.gateway.extensions.modbus.conf.mapping.TagValueMapping out of START_ARRAY token
 at [Source: [[{tag:WriteCoil,functionCode:5,address:1,registerCount:1,byteOrder:BIG,bit:-1}]]; line: 1, column: 2] (through reference chain: java.util.ArrayList[0])
	at com.fasterxml.jackson.databind.JsonMappingException.from(JsonMappingException.java:270)
	at com.fasterxml.jackson.databind.DeserializationContext.reportMappingException(DeserializationContext.java:1234)
	at com.fasterxml.jackson.databind.DeserializationContext.handleUnexpectedToken(DeserializationContext.java:1122)
	at com.fasterxml.jackson.databind.DeserializationContext.handleUnexpectedToken(DeserializationContext.java:1075)
	at com.fasterxml.jackson.databind.deser.BeanDeserializerBase.deserializeFromArray(BeanDeserializerBase.java:1370)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer._deserializeOther(BeanDeserializer.java:174)
	at com.fasterxml.jackson.databind.deser.BeanDeserializer.deserialize(BeanDeserializer.java:150)
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:287)
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:259)
	at com.fasterxml.jackson.databind.deser.std.CollectionDeserializer.deserialize(CollectionDeserializer.java:26)
	at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:3798)
	at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2861)
	at org.thingsboard.gateway.util.JsonTools.fromString(JsonTools.java:54)
	... 7 common frames omitted
`
 
Я не программирую на Java, но посмотрел код [TagValueMapping](https://github.com/thingsboard/thingsboard-gateway/blob/master/src/main/java/org/thingsboard/gateway/extensions/modbus/conf/mapping/TagValueMapping.java) и сравнил его с кодом [PollingTagMapping](https://github.com/thingsboard/thingsboard-gateway/blob/master/src/main/java/org/thingsboard/gateway/extensions/modbus/conf/mapping/PollingTagMapping.java) и мне кажется, что в TagValueMapping  не хватает двух строчек:
`import org.thingsboard.gateway.extensions.common.conf.mapping.DataTypeMapping;
import org.thingsboard.gateway.extensions.modbus.conf.ModbusExtensionConstants;`

Наверное я не прав, но как попробовать всё равно не знаю. Так что подскажите, что я делаю не так.
У кого-нибудь получалось записать данные в устройство Modbus с ThingsBoard через GateWay?

lazermen, 2019-04-26:
в RPC Debug terminale запрос такой:
`write '[{\"tag\":'true',\"value\":882,\"functionCode\":6,\"address\":0,\"byteOrder\":\"BIG\"}]'`
вместо 'true' можно поставить 'false' , роли не играет. 
Какой-то косяк видимо в коде, не обрабатываются значения string.

zbeacon, 2019-12-31:
Hi, @lazermen

I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
Less footprint and memory impact
Easier to extend and customize
Improved functionality.

You can send data to Modbus server/slave using RPC request over the new gateway.

closed 2019-12-31 by zbeacon