thingsboard/thingsboard-gateway #177: The data in opc-ua server is  Connected. but I can't see the data on Latest telemetry

liangguorui666, 2019-04-02:
tb-gateway.yml :

``` yml 
server:
  # Server bind address
  address: "0.0.0.0"
  # Server bind port
  port: "9090"
# Check new version updates parameters
updates:
  # Enable/disable updates checking.
  enabled: "${UPDATES_ENABLED:true}"
gateways:
  tenants:
    -
      label: "Tenant"
      reporting:
        interval: 60000
      persistence:
        type: file
        path: storage
        bufferSize: 1000
      connection:
        host: localhost
        port: 1883
        retryInterval: 3000
        maxInFlight: 1000
        security:
          accessToken: aUjiymacsKDykrFuumFu
      remoteConfiguration: true
      extensions:
        -
          id: "http"
          type: "HTTP"
          extensionConfiguration: http-config.json
#        -
#          id: "opc-ua"
#          type: "OPC UA"
#          extensionConfiguration: opc-config.json
opc:
  enabled: true
mqtt:
  enabled: true 
```
 extension： 
[image]

FreeOpcua Client
[image]

conselLog
[console.log](https://github.com/thingsboard/thingsboard-gateway/files/3033798/console.log)

  Latest telemetry  ：
[image]

these configurations not good
deviceNodePattern ：Root\.Views\.AirConditionerView\.AirConditioner_10
deviceNodePattern ：Root\\.Views\\.AirConditionerView\\.AirConditioner_10
deviceNodePattern ：Root\/Views\/AirConditionerView\/AirConditioner_10
deviceNodePattern ：Root/Views/AirConditionerView/AirConditioner_10
deviceNodePattern ：Root/.Views/.AirConditionerView/.AirConditioner_10
deviceNodePattern ：Root//.Views//.AirConditionerView//.AirConditioner_10

liangguorui666, 2019-04-03:
these configurations not good

liangguorui666, 2019-04-03:
these configurations not good
Root\\.Views\\.AirConditionerView\\.AirConditioner_1\\d+$
Root\.Views\.AirConditionerView\.AirConditioner_1\\d+$
Root\.Views\.AirConditionerView\.AirConditioner_1\d+$
Waiting for your .  Online
@mp-loki

mp-loki, 2019-04-03:
Please do the following:
1) backup your old tb-gateway.log file, 
2) set log level to DEBUG in your logback.xml for thingsboard section, 
3) restart your gateway and run your test again
4) attrach the new log file to this issue

liangguorui666, 2019-04-04:
Thank you very much
I've seen the data, but my  deviceNodePattern : .* 
[image]
[image]

How do I write regex  “deviceNodePattern”and ”deviceNamePattern”  to see the telemetry values of all devices
[image]

@mp-loki

zbeacon, 2019-12-31:
Hi, @liangguorui666 

I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
Less footprint and memory impact
Easier to extend and customize
Improved functionality.

You can try using a new gateway for your purposes instead of a Java gateway. You can read more about OPC-UA configuration in the new gateway [here](https://thingsboard.io/docs/iot-gateway/config/opc-ua/).

closed 2019-12-31 by zbeacon