thingsboard/thingsboard-gateway #179: I can not communicate with my devices through Modbus TCP ...

edme10, 2019-04-11:
Someone can guide me on how to configure the modbus TCP extension in TB-Gateway, I tried the instructions on the thingsboard page without success.

Any suggestion or guide

claib, 2019-11-03:
hi
same problem here ;-)
did you resolve ?

zbeacon, 2019-12-31:
Hi, @edme10 , @claib 

I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
 - Less footprint and memory impact
 - Easier to extend and customize
 - Improved functionality.

Unfortunately, remote configuration isn't implemented yet, we are going to implement it in next release. You can configure Modbus using [this guide](https://thingsboard.io/docs/iot-gateway/config/modbus/).

closed 2019-12-31 by zbeacon