thingsboard/thingsboard-gateway #182: Possible bug - Gateway MQTT extension GUI configuration

javierferreiraglez, 2019-04-24:
My configuration is:
    ThingsBoard 2.3.1
    ThingsBoard-Gateway 2.2.1rc
    Postgresql 11.2

There may be a possible bug that happens when you try to run your basic MQTT extension example for the Thingsboard Gateway ( https://thingsboard.io/docs/iot-gateway/getting-started/ - Step 8.3.1) following the GUI.

If you configure the extension following only the GUI and you try to publish a mqtt message, I was getting the following error in the TB gateway:

```
2019-04-24 22:49:54,409 [MQTT Rec: 5f208b98-fc5a-4ad3-9a8e-dd78df296091] DEBUG o.e.p.c.mqttv3.internal.ClientState - 5f208b98-fc5a-4ad3-9a8e-dd78df296091: received bytes count=1
2019-04-24 22:49:54,409 [MQTT Rec: 5f208b98-fc5a-4ad3-9a8e-dd78df296091] DEBUG o.e.p.c.mqttv3.internal.ClientState - 5f208b98-fc5a-4ad3-9a8e-dd78df296091: received bytes count=72
2019-04-24 22:49:54,409 [MQTT Rec: 5f208b98-fc5a-4ad3-9a8e-dd78df296091] DEBUG o.e.p.c.m.i.wire.MqttInputStream - null: Automatic Reconnect Successful: PUBLISH qos:0 retained:false dup:false topic:"sensors123" payload:[hex:7b2273657269616c4e756d626572223a22534e2d utf8:"{"serialNumber":"SN-" length:60]
2019-04-24 22:49:54,410 [MQTT Rec: 5f208b98-fc5a-4ad3-9a8e-dd78df296091] DEBUG o.e.p.c.mqttv3.internal.ClientState - 5f208b98-fc5a-4ad3-9a8e-dd78df296091: received key=0 message=PUBLISH qos:0 retained:false dup:false topic:"sensors123" payload:[hex:7b2273657269616c4e756d626572223a22534e2d utf8:"{"serialNumber":"SN-" length:60]
2019-04-24 22:49:54,411 [MQTT Rec: 5f208b98-fc5a-4ad3-9a8e-dd78df296091] DEBUG o.e.p.c.m.internal.CommsCallback - 5f208b98-fc5a-4ad3-9a8e-dd78df296091: new msg avail, notify workAvailable
2019-04-24 22:49:54,411 [MQTT Call: 5f208b98-fc5a-4ad3-9a8e-dd78df296091] DEBUG o.e.p.c.m.internal.CommsCallback - 5f208b98-fc5a-4ad3-9a8e-dd78df296091: call messageArrived key=0 topic=sensors123
2019-04-24 22:49:54,412 [MQTT Call: 5f208b98-fc5a-4ad3-9a8e-dd78df296091] TRACE o.t.g.e.m.c.c.m.MqttJsonConverter - Parsing json message: {"serialNumber":"SN-113", "model":"T1000", "temperature":11}
2019-04-24 22:49:54,412 [MQTT Call: 5f208b98-fc5a-4ad3-9a8e-dd78df296091] INFO  o.t.g.e.m.c.l.MqttTelemetryMessageListener - [sensors123] Failed to decode message: [123, 34, 115, 101, 114, 105, 97, 108, 78, 117, 109, 98, 101, 114, 34, 58, 34, 83, 78, 45, 49, 49, 51, 34, 44, 32, 34, 109, 111, 100, 101, 108, 34, 58, 34, 84, 49, 48, 48, 48, 34, 44, 32, 34, 116, 101, 109, 112, 101, 114, 97, 116, 117, 114, 101, 34, 58, 49, 49, 125]
java.lang.NullPointerException: null
	at org.thingsboard.gateway.extensions.mqtt.client.conf.mapping.MqttJsonConverter.convert(MqttJsonConverter.java:60)
	at org.thingsboard.gateway.extensions.mqtt.client.listener.MqttTelemetryMessageListener.messageArrived(MqttTelemetryMessageListener.java:42)
	at org.eclipse.paho.client.mqttv3.internal.CommsCallback.deliverMessage(CommsCallback.java:491)
	at org.eclipse.paho.client.mqttv3.internal.CommsCallback.handleMessage(CommsCallback.java:402)
	at org.eclipse.paho.client.mqttv3.internal.CommsCallback.run(CommsCallback.java:206)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
2019-04-24 22:49:54,413 [MQTT Call: 5f208b98-fc5a-4ad3-9a8e-dd78df296091] DEBUG o.e.p.c.m.internal.CommsCallback - 5f208b98-fc5a-4ad3-9a8e-dd78df296091: notify spaceAvailable
2019-04-24 22:49:54,413 [MQTT Call: 5f208b98-fc5a-4ad3-9a8e-dd78df296091] DEBUG o.e.p.c.m.internal.CommsCallback - 5f208b98-fc5a-4ad3-9a8e-dd78df296091: wait for workAvailable
2019-04-24 22:49:54,413 [MQTT Rec: 5f208b98-fc5a-4ad3-9a8e-dd78df296091] DEBUG o.e.p.c.m.internal.CommsReceiver - 5f208b98-fc5a-4ad3-9a8e-dd78df296091: network read message
2019-04-24 22:49:54,758 [http-nio-0.0.0.0-9090-ClientPoller-0] DEBUG o.apache.tomcat.util.net.NioEndpoint - timeout completed: keys processed=0; now=1556138994758; nextExpiration=1556138994756; keyCount=0; hasEvents=false; eval=false
```

On the other hand, if you configure the MQTT Extension by importing the json configuration file from the tutorial, the extension worked and the new device was created.

Comparing the configuration JSON extension configuration files, I saw that the field  **filterExpression** in the broker mapping was missing when you use the GUI to configure the extension. This missing field was the reason why you get the error when the extension tries to parse the mqtt message.

To avoid this error and if you want to still use only the GUI to configure the extension, in the ***Mapping - Filter expresion*** input box you place an space and delete it, this probably causes the field to initialize as an empty string ("").

If this is not a bug, you can remove this post.

NOTE: Attached you can find the extension configuration file, created when you configure it with the GUI

```json
[
	{
		"id": "TestMQTT",
		"type": "MQTT",
		"configuration": {
			"brokers": [
				{
					"host": "test.mosquitto.org",
					"port": 1883,
					"ssl": false,
					"retryInterval": 3000,
					"credentials": {
						"type": "anonymous"
					},
					"mapping": [
						{
							"topicFilter": "sensors123",
							"converter": {
								"attributes": [
									{
										"type": "string",
										"key": "model",
										"value": "${$.model}"
									}
								],
								"timeseries": [
									{
										"type": "double",
										"key": "temperature",
										"value": "${$.temperature}"
									}
								],
								"type": "json",
								"deviceNameJsonExpression": "${$.serialNumber}"
							}
						}
					],
					"connectRequests": [],
					"disconnectRequests": [],
					"attributeRequests": [],
					"attributeUpdates": [],
					"serverSideRpc": []
				}
			]
		}
	}
]
```

affoltep, 2019-09-20:
I can confirm this issue. It seems to be a bug

ssmaurya, 2019-11-09:
Fix being submitted for pull in main line. Added null check for filter expression

javierferreiraglez, 2019-11-14:
Thanks for the great job!

closed 2019-12-31 by zbeacon