thingsboard/thingsboard-gateway #183: Netty MQTT Client Failed

Anu111997, 2019-05-02:
Hi all,
I am trying to build thinksboard from source but getting the following error while running the command mvn clean install(I have also tried running with sudo mvn clean install).
[INFO] Thingsboard ........................................ SUCCESS [  0.701 s]
[INFO] Netty MQTT Client .................................. FAILURE [  1.070 s]
[INFO] Thingsboard Server Commons ......................... SKIPPED
[INFO] Thingsboard Server Common Data ..................... SKIPPED
[INFO] Thingsboard Server Common Messages ................. SKIPPED
[INFO] Thingsboard Server Queue components ................ SKIPPED
[INFO] Thingsboard Server Commons ......................... SKIPPED
[INFO] Thingsboard Server Common Transport components ..... SKIPPED
[INFO] Thingsboard MQTT Transport Common .................. SKIPPED
[INFO] Thingsboard HTTP Transport Common .................. SKIPPED
[INFO] Thingsboard CoAP Transport Common .................. SKIPPED
[INFO] Thingsboard Server DAO Layer ....................... SKIPPED
[INFO] Thingsboard Extensions ............................. SKIPPED
[INFO] Thingsboard Rule Engine API ........................ SKIPPED
[INFO] Thingsboard Rule Engine Components ................. SKIPPED
[INFO] Thingsboard Server Transport Modules ............... SKIPPED
[INFO] Thingsboard HTTP Transport Service ................. SKIPPED
[INFO] Thingsboard MQTT Transport Service ................. SKIPPED
[INFO] Thingsboard CoAP Transport Service ................. SKIPPED
[INFO] Thingsboard Server UI .............................. SKIPPED
[INFO] Thingsboard Server Tools ........................... SKIPPED
[INFO] ThingsBoard Server Application ..................... SKIPPED
[INFO] ThingsBoard Microservices .......................... SKIPPED
[INFO] ThingsBoard Docker Images .......................... SKIPPED
[INFO] ThingsBoard JavaScript Executor Microservice ....... SKIPPED
[INFO] ThingsBoard Web UI Microservice .................... SKIPPED
[INFO] ThingsBoard Node Microservice ...................... SKIPPED
[INFO] ThingsBoard Transport Microservices ................ SKIPPED
[INFO] ThingsBoard MQTT Transport Microservice ............ SKIPPED
[INFO] ThingsBoard HTTP Transport Microservice ............ SKIPPED
[INFO] ThingsBoard COAP Transport Microservice ............ SKIPPED
[INFO] ThingsBoard Black Box Tests ........................ SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  2.407 s
[INFO] Finished at: 2019-05-02T17:01:45+05:30
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.1:compile (default-compile) on project netty-mqtt: Fatal error compiling: java.lang.ExceptionInInitializerError: com.sun.tools.javac.code.TypeTags -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :netty-mqtt

please helpme

mp-loki, 2019-05-13:
What version are you building? Is it master? Any more errors in the log?

Anu111997, 2019-05-13:
Hi,
I am trying to build  ThingsBoard Server Application 2.3.1 from Github. As of now found this issue.

mp-loki, 2019-05-17:
Try to build with the command `mvn clean install -DskipTests -X` to get verbose output. probably it will help to identify the issue

kevz93, 2019-07-01:
Is this solved ? i have the same issue. Not sure if its JAVA_HOME path conflicting ?

samf48, 2019-08-19:
@mp-loki  Same issue here is verbose output of problem:

[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for Thingsboard 2.4.1-SNAPSHOT:
[INFO]
[INFO] Thingsboard ........................................ SUCCESS [04:06 min]
[INFO] Netty MQTT Client .................................. FAILURE [03:53 min]
[INFO] Thingsboard Server Commons ......................... SKIPPED
[INFO] Thingsboard Server Common Data ..................... SKIPPED
[INFO] Thingsboard Server Common Utils .................... SKIPPED
[INFO] Thingsboard Server Common Messages ................. SKIPPED
[INFO] Thingsboard Server Queue components ................ SKIPPED
[INFO] Thingsboard Server Commons ......................... SKIPPED
[INFO] Thingsboard Server Common Transport components ..... SKIPPED
[INFO] Thingsboard MQTT Transport Common .................. SKIPPED
[INFO] Thingsboard HTTP Transport Common .................. SKIPPED
[INFO] Thingsboard CoAP Transport Common .................. SKIPPED
[INFO] Thingsboard Server Common DAO API .................. SKIPPED
[INFO] Thingsboard Extensions ............................. SKIPPED
[INFO] Thingsboard Rule Engine API ........................ SKIPPED
[INFO] Thingsboard Server DAO Layer ....................... SKIPPED
[INFO] Thingsboard Rule Engine Components ................. SKIPPED
[INFO] Thingsboard Server Transport Modules ............... SKIPPED
[INFO] Thingsboard HTTP Transport Service ................. SKIPPED
[INFO] Thingsboard MQTT Transport Service ................. SKIPPED
[INFO] Thingsboard CoAP Transport Service ................. SKIPPED
[INFO] Thingsboard Server UI .............................. SKIPPED
[INFO] Thingsboard Server Tools ........................... SKIPPED
[INFO] ThingsBoard Server Application ..................... SKIPPED
[INFO] ThingsBoard Microservices .......................... SKIPPED
[INFO] ThingsBoard Docker Images .......................... SKIPPED
[INFO] ThingsBoard JavaScript Executor Microservice ....... SKIPPED
[INFO] ThingsBoard Web UI Microservice .................... SKIPPED
[INFO] ThingsBoard Node Microservice ...................... SKIPPED
[INFO] ThingsBoard Transport Microservices ................ SKIPPED
[INFO] ThingsBoard MQTT Transport Microservice ............ SKIPPED
[INFO] ThingsBoard HTTP Transport Microservice ............ SKIPPED
[INFO] ThingsBoard COAP Transport Microservice ............ SKIPPED
[INFO] ThingsBoard Black Box Tests ........................ SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  09:52 min
[INFO] Finished at: 2019-08-19T17:59:50Z
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.1:compile (default-compile) on project netty-mqtt: Fatal error compiling: java.lang.ExceptionInInitializerError: com.sun.tools.javac.code.TypeTags -> [Help 1]
org.apache.maven.lifecycle.LifecycleExecutionException: Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.1:compile (default-compile) on project netty-mqtt: Fatal error compiling
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:215)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:156)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:148)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:117)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:81)
    at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build (SingleThreadedBuilder.java:56)
    at org.apache.maven.lifecycle.internal.LifecycleStarter.execute (LifecycleStarter.java:128)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:305)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:192)
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:105)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:956)
    at org.apache.maven.cli.MavenCli.doMain (MavenCli.java:288)
    at org.apache.maven.cli.MavenCli.main (MavenCli.java:192)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0 (Native Method)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke (NativeMethodAccessorImpl.java:62)
    at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke (DelegatingMethodAccessorImpl.java:43)
    at java.lang.reflect.Method.invoke (Method.java:566)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced (Launcher.java:282)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:225)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:406)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:347)
Caused by: org.apache.maven.plugin.MojoExecutionException: Fatal error compiling
    at org.apache.maven.plugin.compiler.AbstractCompilerMojo.execute (AbstractCompilerMojo.java:796)
[... 169 lines cut ...]
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:117)
[... cut ...]
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:406)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:347)
[ERROR]
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :netty-mqtt

samf48, 2019-08-19:
Deleting the .m2 folder in the user/USERNAME/ directory as well as making sure that the JAVA_HOME variable on Windows was pointing to the JDK location fixed it. Additionally I just wiped out the entire Thingsboard source code folder and re-built it.  It seems like if a previous build fails it leave "stuff" behind so starting fresh is crucial.

closed 2019-12-31 by zbeacon

hunglmbk, 2020-10-12:
upgrade lombok version fixed this for my local build
        `<lombok.version>1.18.2</lombok.version>`