thingsboard/thingsboard-gateway #185: MQTT does not reconnect

mfalkvidd, 2019-05-09:
I have installed a separate MQTT broker (mosquitto) and configured TB-gateway to connect to it.

Everything was working fine, but one day TB-gateway stopped receiving messages. This was present in the log:

```
2019-04-29 10:15:05,161 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='GATEWAY', payload={"ts":1556532895594,"values":{"devicesOnline":21,"attributesUploaded":0,"telemetryUploaded":0}}, timestamp=0, topic='v1/devices/me/tele
metry'id=acf8ea36-300f-4fc4-befd-0134853c7e37, messageId=21775}]
2019-04-29 10:15:05,181 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2019-04-29 10:15:05,207 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Gateway statistics {"ts":1556532775594,"values":{"devicesOnline":21,"attributesUploaded":0,"telemetryUploaded":0}} reported!
2019-04-29 10:15:05,208 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Gateway statistics {"ts":1556532835594,"values":{"devicesOnline":21,"attributesUploaded":0,"telemetryUploaded":0}} reported!
2019-04-29 10:15:05,241 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Gateway statistics {"ts":1556532895594,"values":{"devicesOnline":21,"attributesUploaded":0,"telemetryUploaded":0}} reported!
2019-04-29 10:15:06,499 [MQTT Rec: 87cf3e28-4421-43cc-a60d-1c4557f2f900] WARN  o.t.g.e.m.client.MqttBrokerMonitor - [localhost:11883] MQTT broker connection lost!
2019-04-29 10:15:09,237 [MQTT Rec: 87cf3e28-4421-43cc-a60d-1c4557f2f900] INFO  o.t.g.s.gateway.MqttGatewayService - [3820536284][21776] Device Disconnected!
...
2019-04-29 10:15:09,977 [MQTT Rec: 87cf3e28-4421-43cc-a60d-1c4557f2f900] INFO  o.t.g.s.gateway.MqttGatewayService - [3820535528][21796] Device Disconnected!
2019-04-29 10:15:09,987 [MQTT Rec: 87cf3e28-4421-43cc-a60d-1c4557f2f900] DEBUG o.t.g.e.m.client.MqttBrokerMonitor - [localhost:11883] MQTT broker connection attempt!
2019-04-29 10:15:10,210 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='3820536284', payload={"device":"3820536284"}, timestamp=0, topic='v1/gateway/disconnect'id=4960c2ac-bf08-437f-9985-e694b838feae, messageId=21776}]
...
2019-04-29 10:15:10,389 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='3820535528', payload={"device":"3820535528"}, timestamp=0, topic='v1/gateway/disconnect'id=51f69d14-128a-47d9-9160-bcc9a9333148, messageId=21796}]
2019-04-29 10:15:10,389 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2019-04-29 10:15:05,161 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='GATEWAY', payload={"ts":1556532895594,"values":{"devicesOnline":21,"attributesUploaded":0,"telemetryUploaded":0}}, timestamp=0, topic='v1/devices/me/tele
metry'id=acf8ea36-300f-4fc4-befd-0134853c7e37, messageId=21775}]
2019-04-29 10:15:05,181 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2019-04-29 10:15:05,207 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Gateway statistics {"ts":1556532775594,"values":{"devicesOnline":21,"attributesUploaded":0,"telemetryUploaded":0}} reported!
2019-04-29 10:15:05,208 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Gateway statistics {"ts":1556532835594,"values":{"devicesOnline":21,"attributesUploaded":0,"telemetryUploaded":0}} reported!
2019-04-29 10:15:05,241 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Gateway statistics {"ts":1556532895594,"values":{"devicesOnline":21,"attributesUploaded":0,"telemetryUploaded":0}} reported!
2019-04-29 10:15:06,499 [MQTT Rec: 87cf3e28-4421-43cc-a60d-1c4557f2f900] WARN  o.t.g.e.m.client.MqttBrokerMonitor - [localhost:11883] MQTT broker connection lost!
2019-04-29 10:15:09,237 [MQTT Rec: 87cf3e28-4421-43cc-a60d-1c4557f2f900] INFO  o.t.g.s.gateway.MqttGatewayService - [3820536284][21776] Device Disconnected!
...
2019-04-29 10:15:09,977 [MQTT Rec: 87cf3e28-4421-43cc-a60d-1c4557f2f900] INFO  o.t.g.s.gateway.MqttGatewayService - [3820535528][21796] Device Disconnected!
2019-04-29 10:15:09,987 [MQTT Rec: 87cf3e28-4421-43cc-a60d-1c4557f2f900] DEBUG o.t.g.e.m.client.MqttBrokerMonitor - [localhost:11883] MQTT broker connection attempt!
2019-04-29 10:15:10,210 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='3820536284', payload={"device":"3820536284"}, timestamp=0, topic='v1/gateway/disconnect'id=4960c2ac-bf08-437f-9985-e694b838feae, messageId=21776}]
...
2019-04-29 10:15:10,389 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='3820535528', payload={"device":"3820535528"}, timestamp=0, topic='v1/gateway/disconnect'id=51f69d14-128a-47d9-9160-bcc9a9333148, messageId=21796}]
2019-04-29 10:15:10,389 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2019-04-29 10:15:11,272 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - [3820536284][21776] Device disconnect event is delivered!
...
2019-04-29 10:15:11,305 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - [3820535596][21782] Device disconnect event is delivered!
```

After this, TB-gateway has endless repeats of the folllwoing, with no attempt to reconnect:

```
2019-04-29 10:15:56,394 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Sending message [{deviceId='GATEWAY', payload={"ts":1556532955594,"values":{"devicesOnline":0,"attributesUploaded":0,"telemetryUploaded":0}}, timestamp=0, topic='v1/devices/me/telemetry'id=278d04ee-63f1-4bf4-86af-51b174d9fc60, messageId=21797}]
2019-04-29 10:15:56,395 [pool-2-thread-1] DEBUG o.t.g.service.MqttMessageSender - Waiting until all messages are sent before going to the next bucket
2019-04-29 10:15:56,417 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Gateway statistics {"ts":1556532955594,"values":{"devicesOnline":0,"attributesUploaded":0,"telemetryUploaded":0}} reported!
```

When I manually restarted TB-gateway, everything started working again.

If I manually shot down mosquitto, I can see that TB-gateway tries to reconnect every 3 seconds:
```thingsboard@thingsboard-1:~$ grep 'MQTT broker connection faied!' /var/log/tb-gateway/tb-gateway.log
2019-05-09 08:18:41,938 [MQTT Con: 380d405b-a69c-4e32-ba24-cf977186bd88] WARN  o.t.g.e.m.client.MqttBrokerMonitor - [localhost:11883] MQTT broker connection faied! Unable to connect to server
2019-05-09 08:18:44,945 [MQTT Con: 380d405b-a69c-4e32-ba24-cf977186bd88] WARN  o.t.g.e.m.client.MqttBrokerMonitor - [localhost:11883] MQTT broker connection faied! Unable to connect to server
2019-05-09 08:18:47,959 [MQTT Con: 380d405b-a69c-4e32-ba24-cf977186bd88] WARN  o.t.g.e.m.client.MqttBrokerMonitor - [localhost:11883] MQTT broker connection faied! Unable to connect to server
2019-05-09 08:18:50,966 [MQTT Con: 380d405b-a69c-4e32-ba24-cf977186bd88] WARN  o.t.g.e.m.client.MqttBrokerMonitor - [localhost:11883] MQTT broker connection faied! Unable to connect to server
2019-05-09 08:18:53,974 [MQTT Con: 380d405b-a69c-4e32-ba24-cf977186bd88] WARN  o.t.g.e.m.client.MqttBrokerMonitor - [localhost:11883] MQTT broker connection faied! Unable to connect to server
2019-05-09 08:18:56,978 [MQTT Con: 380d405b-a69c-4e32-ba24-cf977186bd88] WARN  o.t.g.e.m.client.MqttBrokerMonitor - [localhost:11883] MQTT broker connection faied! Unable to connect to server
```

and when I start mosquitto again, TB-gateway reconnects:
```
2019-05-09 08:22:15,475 [MQTT Call: 380d405b-a69c-4e32-ba24-cf977186bd88] INFO  o.t.g.e.m.client.MqttBrokerMonitor - [localhost:11883] MQTT broker connection established!
```

So it seems like the reconnection feature works, but the failure on 2019-04-29 10:15:06,499 did not cause any reconnection attempt. Any idea why?

mfalkvidd, 2019-05-09:
Full log file: https://gist.githubusercontent.com/mfalkvidd/127e8fe9cae39b275d1b8f0e1c516536/raw/ce2569cede2a9c8900cc2f366217244d9f4bbb8f/tb-gateway-log

ndctouko, 2019-11-08:
Have you figured this out yet? 
I have exactly the same problem, but I'm using AWS IoT Core as Broker.

toukke, 2019-11-08:
Have you figured this out yet?
I have exactly the same problem when using AWS IoT Core.

zbeacon, 2019-12-31:
Hi, @mfalkvidd 

I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
Less footprint and memory impact
Easier to extend and customize
Improved functionality.

You can try to using the new gateway instead of Java gateway. The new gateway supports autoreconnect to external brokers.

closed 2019-12-31 by zbeacon