thingsboard/thingsboard-gateway #186: client-side timestamp not posting to telemetry

samf48, 2019-05-09:
I've never been able to post the client-side time stamp successfully.
I tried sending client-side timestamps in the form of `mosquitto_pub -h localhost -p 1883 -t "sensors" -m "{"ts":1451649600512,"values":{"serialNumber":"SN-001", "model":"T1000", "temperature":36.6}}" `but I get nothing on the telemetry.  I'm on windows and have tried several different things.  The other thing I tried was using Node.JS and MQTT like so:
```
mqtt = require('mqtt');
var client  = mqtt.connect('mqtt://127.0.0.1:1885');
 
client.on('connect', function () {
  client.subscribe('sensors', function (err) {
    if (!err) {
		var obj1 = {'SensorName':'Virtual_Test_Beta1','model':'beta','IMEI':'352555570150826','NI':'001','protocol':'1','time':'2018','lat':'n/a','lon':'n/a','elev':'n/a','mode':'1','baro':'1389.05','airTemp':'36.7','h2oTemp':'22.0','battery':'4.8','rssi':'135','Pressure':'-6.35'};
		var obj2 = {'ts':1557282621221,'values':obj1};
		//obj1 = JSON.stringify(obj1);
		//obj1 = JSON.stringify('ts':'1557282621221','values':obj1);
      //client.publish('sensors', obj1)
	  client.publish('sensors', JSON.stringify(obj2))
    }
  })
})
 
client.on('message', function (topic, message) {
  // message is Buffer
  console.log(message.toString())
  client.end()
})
```

Am I doing something wrong or is this just not possible in the gateway's current API?

samf48, 2019-05-10:
I get this feedback in the .log file: 
```
2019-05-10 03:49:52,331 [MQTT Call: 3f732bef-d9e4-43ca-85f6-6c52cfbece4b] INFO  o.t.g.s.gateway.MqttGatewayService - [${$.serialNumber}] Device Connected!
2019-05-10 03:49:52,718 [pool-3-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - [${$.serialNumber}][${$.model}][466] Device connect event is reported to Thingsboard!
```
command I used:
`mosquitto_pub -h localhost -p 1885 -t "sensors" -m "{\"ts\":1557286276627,\"values\":{\"serialNumber\":\"SN-003\", \"model\":\"T3000\", \"temperature\":36.6}}" -d `

samf48, 2019-05-15:
Using the latest snapshot of the gateway (2.2.1) I am able to send the timestamps in a different way. Command I use:
mosquitto_pub -h localhost -p 1885 -t "sensors" -m "{\"serialNumber\":\"SN-003\", \"model\":\"T3000\", \"temperature\":36.6,\"ts\":1557286276627}"

closed 2019-05-15 by samf48

samf48, 2020-02-07:
@zbeacon is the new Python version able to accept client-side timestamps?

zbeacon, 2020-02-07:
Hi @samf48,

I added this ability in pre-release version 2.1.0.1 you can install it using pip:
```pip3 install thingsboard-gateway==2.1.0.1```
The gateway looking for "ts" or "timestamp" keys in incoming message.

samf48, 2020-02-07:
@zbeacon wow that was fast, works perfectly...Thanks!!