thingsboard/thingsboard-gateway #196: java.lang.NullPointerException: null

loganmc10, 2019-07-25:
I am trying to read MQTT coming from TheThingsNetwork, everything seems to be working fine, but I don't get any data, this is the error I get in the logs:

```
18:06:15.381 [MQTT Call: 59382b0a-6ff7-410d-ba0f-9f9579f15e80] INFO  o.t.g.e.m.c.l.MqttTelemetryMessageListener - [cbu-ttn-environment/devices/cbu-ttn-env-sensor/up] Failed to decode message: [123, 34, 97, 112, 112, 95, 105, 100, 34, 58, 34, 99, 98, 117, 45, 116, 116, 110, 45, 101, 110, 118, 105, 114, 111, 110, 109, 101, 110, 116, 34, 44, 34, 100, 101, 118, 95, 105, 100, 34, 58, 34, 99, 98, 117, 45, 116, 116, 110, 45, 101, 110, 118, 45, 115, 101, 110, 115, 111, 114, 34, 44, 34, 104, 97, 114, 100, 119, 97, 114, 101, 95, 115, 101, 114, 105, 97, 108, 34, 58, 34, 48, 48, 48, 52, 65, 51, 48, 66, 48, 48, 49, 65, 70, 50, 67, 55, 34, 44, 34, 112, 111, 114, 116, 34, 58, 50, 44, 34, 99, 111, 117, 110, 116, 101, 114, 34, 58, 49, 48, 51, 49, 44, 34, 112, 97, 121, 108, 111, 97, 100, 95, 114, 97, 119, 34, 58, 34, 68, 48, 103, 65, 107, 81, 108, 47, 34, 44, 34, 112, 97, 121, 108, 111, 97, 100, 95, 102, 105, 101, 108, 100, 115, 34, 58, 123, 34, 98, 97, 116, 116, 101, 114, 121, 34, 58, 51, 57, 49, 50, 44, 34, 101, 118, 101, 110, 116, 34, 58, 34, 105, 110, 116, 101, 114, 118, 97, 108, 34, 44, 34, 108, 105, 103, 104, 116, 34, 58, 49, 52, 53, 44, 34, 116, 101, 109, 112, 101, 114, 97, 116, 117, 114, 101, 34, 58, 50, 52, 46, 51, 49, 125, 44, 34, 109, 101, 116, 97, 100, 97, 116, 97, 34, 58, 123, 34, 116, 105, 109, 101, 34, 58, 34, 50, 48, 49, 57, 45, 48, 55, 45, 50, 53, 84, 49, 56, 58, 48, 54, 58, 49, 53, 46, 49, 54, 49, 53, 54, 48, 52, 56, 55, 90, 34, 44, 34, 102, 114, 101, 113, 117, 101, 110, 99, 121, 34, 58, 57, 48, 52, 46, 53, 44, 34, 109, 111, 100, 117, 108, 97, 116, 105, 111, 110, 34, 58, 34, 76, 79, 82, 65, 34, 44, 34, 100, 97, 116, 97, 95, 114, 97, 116, 101, 34, 58, 34, 83, 70, 55, 66, 87, 49, 50, 53, 34, 44, 34, 97, 105, 114, 116, 105, 109, 101, 34, 58, 53, 49, 52, 53, 54, 48, 48, 48, 44, 34, 99, 111, 100, 105, 110, 103, 95, 114, 97, 116, 101, 34, 58, 34, 52, 47, 53, 34, 44, 34, 103, 97, 116, 101, 119, 97, 121, 115, 34, 58, 91, 123, 34, 103, 116, 119, 95, 105, 100, 34, 58, 34, 99, 98, 117, 45, 116, 116, 110, 45, 103, 97, 116, 101, 119, 97, 121, 34, 44, 34, 103, 116, 119, 95, 116, 114, 117, 115, 116, 101, 100, 34, 58, 116, 114, 117, 101, 44, 34, 116, 105, 109, 101, 115, 116, 97, 109, 112, 34, 58, 52, 51, 57, 49, 49, 52, 49, 52, 55, 44, 34, 116, 105, 109, 101, 34, 58, 34, 50, 48, 49, 57, 45, 48, 55, 45, 50, 53, 84, 49, 56, 58, 48, 54, 58, 49, 53, 90, 34, 44, 34, 99, 104, 97, 110, 110, 101, 108, 34, 58, 51, 44, 34, 114, 115, 115, 105, 34, 58, 45, 52, 51, 44, 34, 115, 110, 114, 34, 58, 49, 48, 46, 53, 44, 34, 114, 102, 95, 99, 104, 97, 105, 110, 34, 58, 48, 125, 93, 125, 125]
java.lang.NullPointerException: null
        at org.thingsboard.gateway.extensions.mqtt.client.conf.mapping.MqttJsonConverter.convert(MqttJsonConverter.java:60)
        at org.thingsboard.gateway.extensions.mqtt.client.listener.MqttTelemetryMessageListener.messageArrived(MqttTelemetryMessageListener.java:42)
        at org.eclipse.paho.client.mqttv3.internal.CommsCallback.deliverMessage(CommsCallback.java:491)
        at org.eclipse.paho.client.mqttv3.internal.CommsCallback.handleMessage(CommsCallback.java:402)
        at org.eclipse.paho.client.mqttv3.internal.CommsCallback.run(CommsCallback.java:206)
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
```

mp-loki, 2019-07-26:
Please provide your mqtt-config.json (or export config from TB GUI if you are using that)

ssmaurya, 2019-11-09:
Seems to be similar to https://github.com/thingsboard/thingsboard-gateway/issues/182

closed 2019-12-31 by zbeacon