thingsboard/thingsboard-gateway #197: MODBUS is not supported!

demirizZz, 2019-07-26:
Helllo,
I try to use thingsboard-gateway with Modbus Slave. MQTT is works fine but Modbus is not working. 

tb-gateway.log: 

2019-07-26 13:48:24,990 [pool-3-thread-1] INFO  o.t.g.service.MqttMessageSender - Sending message [MqttPersistentMessage(id=40b4b3ee-ee0b-4998-9b46-09c80c248928, timestamp=0, deviceId=null, messageId=1, topic=v1/devices/m$
2019-07-26 13:48:25,047 [pool-2-thread-1] INFO  o.t.g.s.gateway.MqttHandlerImpl - Configuration arrived! {"client":{"appliedConfiguration":"[]"},"shared":{"configuration":"[{\"id\":\"Modbus\",\"type\":\"MODBUS\",\"configu$
2019-07-26 13:48:25,066 [pool-2-thread-1] INFO  o.t.g.service.TenantServiceRegistry - [Tenant] Updating extension configuration
2019-07-26 13:48:25,183 [pool-2-thread-1] INFO  o.t.g.service.TenantServiceRegistry - Initializing extension: [Modbus][MODBUS]
2019-07-26 13:48:25,185 [pool-2-thread-1] INFO  o.t.g.service.TenantServiceRegistry - Failed to read configuration attribute
java.lang.IllegalArgumentException: Extension: MODBUS is not supported!
        at org.thingsboard.gateway.service.TenantServiceRegistry.createExtensionServiceByType(TenantServiceRegistry.java:122)
        at org.thingsboard.gateway.service.TenantServiceRegistry.updateExtensionConfiguration(TenantServiceRegistry.java:80)
        at org.thingsboard.gateway.service.DefaultTenantManagerService.lambda$init$0(DefaultTenantManagerService.java:64)
        at org.thingsboard.gateway.service.gateway.MqttHandlerImpl.updateConfiguration(MqttHandlerImpl.java:164)
        at org.thingsboard.gateway.service.gateway.MqttHandlerImpl.onGatewayAttributesGet(MqttHandlerImpl.java:145)
        at org.thingsboard.gateway.service.gateway.MqttHandlerImpl.lambda$onMessage$0(MqttHandlerImpl.java:88)
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)

hasikp1, 2019-08-23:
What is the version you are using?

tohidprogram, 2019-08-28:
if you use latest version, you can use  Modbus  but still there is some minor problems.

TajSiddiqui, 2019-12-26:
I am also facing the similar issue.
I have installed the tb-gateway on win 10. The gateway is up and its getting connecting to the Thingsboard Server. I am using tb-gateway version as  v1.4.0
Now I am using the modbus-config extension from the TB server. The tb-gateway.log shows that the configuration data is arriving at the tb-gateway 
but the next thing tb-gateway throws is the error - Modbus extension not supported. I have set the remoteConfiguration: true so that the TB GUI 
shall able to send the configuration details and tb-gateway shall generate the modbus-config.json the tb-gateway is receveing
configuration data but not creating the modbus-config.json and its throwing the error Extension MODBUS is not supported
Attached is the tb-gateway.yml file

**tb-gateway.log snippets**

2019-12-26 18:48:06,722 [pool-2-thread-1] INFO  o.t.g.s.gateway.MqttHandlerImpl - Configuration arrived! {"shared":{"configuration":"[{\"id\":\"EM 1\",\"type\":\"MODBUS\",\"configuration\":{\"servers\":[{\"transport\":{\"type\":\"rtu\",\"portName\":\"COM1\",\"encoding\":\"rtu\",\"timeout\":5000,\"baudRate\":9600,\"dataBits\":7,\"stopBits\":1,\"parity\":\"even\"},\
.....
.....
2019-12-26 18:48:06,756 [pool-2-thread-1] INFO  o.t.g.service.TenantServiceRegistry - [Tenant] Updating extension configuration
2019-12-26 18:48:06,873 [pool-2-thread-1] INFO  o.t.g.service.TenantServiceRegistry - Initializing extension: [EM 1][MODBUS]
2019-12-26 18:48:06,878 [pool-2-thread-1] INFO  o.t.g.service.TenantServiceRegistry - Failed to read configuration attribute
java.lang.IllegalArgumentException: Extension: MODBUS is not supported!
...
...
2019-12-26 18:48:07,635 [pool-4-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - Gateway statistics {"ts":1577366286392,"values":{"devicesOnline":0,"attributesUploaded":0,"telemetryUploaded":0}} reported!
...
...
2019-12-26 18:48:08,519 [main] INFO  o.t.gateway.GatewayApplication - Started GatewayApplication in 11.296 seconds (JVM running for 13.031)

[tb-gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/4002653/tb-gateway.txt)

zbeacon, 2019-12-31:
Hi, @demirizZz 

I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
Less footprint and memory impact
Easier to extend and customize
Improved functionality.

You can try using a new gateway for your purposes instead of a Java gateway. You can read more [here](https://thingsboard.io/docs/iot-gateway/config/modbus/) about Modbus configuration.

We look forward to your feedback.

thucar, 2019-12-31:
I'm trying to get ModBus TCP working with the new gateway. Unsuccessfully so far...

This is my config:
```
{
  "server": {
    "name": "Modbus Connector",
    "type": "tcp",
    "host": "10.8.88.205",
    "port": 502,
    "timeout": 35,
    "method": "socket",
    "devices": [
      {
        "unitId": 0,
        "deviceName": "Komfovent C4",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": true,
        "attributes": [
          {
            "byteOrder": "BIG",
            "tag": "c4_config",
            "type": "integer",
            "functionCode": 3,
            "registerCount": 1,
            "address": 10003
          }
        ],
        "timeseries": [{
            "byteOrder": "BIG",
            "tag": "supply_air",
            "type": "integer",
            "functionCode": 3,
            "registerCount": 1,
            "address": 10300
        }],
        "rpc": {
        }
      }
    ]
  }
}
```
These are snips from Komfovent C4 manual. I'm trying to read the Supply Air temperature value in this instance.
[image]
[image]

My connector.log shows:
```
''2019-12-31 10:29:55' - DEBUG - modbus_connector - 198 - To modbus device Komfovent C4,
{'byteOrder': 'BIG', 'tag': 'supply_air', 'type': 'integer', 'functionCode': 3, 'registerCount': 1, 'address': 10300, 'deviceName': 'Komfovent C4'}'
''2019-12-31 10:29:55' - DEBUG - modbus_connector - 199 - With result Exception Response(132, 4, IllegalFunction)'
''2019-12-31 10:29:55' - ERROR - modbus_connector - 154 - 'ExceptionResponse' object has no attribute 'isError''
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 116, in __process_devices
    if input_data.isError():
AttributeError: 'ExceptionResponse' object has no attribute 'isError'
```

So it's claiming "IllegalFunction" however I can query and get correct responses from the Komfovent device by sending the requests manually. So I know for a fact that the query data (Function:3, register address:10300, registerCount:1) is definitely correct.

zbeacon, 2019-12-31:
Hi, @thucar 

My bad, i mixed up the function 3 and 4 codes, try use 4 function, we will fix that in next release, thank you for feedback.

thucar, 2019-12-31:
Hi @zbeacon,

I tried using function code 4 and am getting a different result now, unfortunately still not getting any data in ThingsBoard.

```
''2019-12-31 11:26:06' - DEBUG - modbus_connector - 198 - To modbus device Komfovent C4,
{'byteOrder': 'BIG', 'tag': 'supply_air', 'type': 'integer', 'functionCode': 4, 'registerCount': 1, 'address': 10300, 'deviceName': 'Komfovent C4'}'
''2019-12-31 11:26:06' - DEBUG - modbus_connector - 199 - With result ReadRegisterResponse (1)'
''2019-12-31 11:26:06' - ERROR - modbus_connector - 154 - 'ReadHoldingRegistersResponse' object has no attribute 'isError''
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 116, in __process_devices
    if input_data.isError():
AttributeError: 'ReadHoldingRegistersResponse' object has no attribute 'isError'
```

zbeacon, 2019-12-31:
@thucar that's better, please try clone repo, run following command and revert function code in config:
```bash
sudo pip3 uninstall thingsboard-gateway -y && sudo python3 ./thingsboard-gateway/thingsboard_gateway/setup.py install
```

thucar, 2019-12-31:
@zbeacon 

Done, this is the output now:

```
''2019-12-31 11:43:27' - DEBUG - modbus_connector - 198 - To modbus device Komfovent C4,
{'byteOrder': 'BIG', 'tag': 'supply_air', 'type': 'integer', 'functionCode': 3, 'registerCount': 1, 'address': 10300, 'deviceName': 'Komfovent C4'}'
''2019-12-31 11:43:27' - DEBUG - modbus_connector - 199 - With result ReadRegisterResponse (1)'
''2019-12-31 11:43:27' - ERROR - modbus_connector - 154 - 'ReadHoldingRegistersResponse' object has no attribute 'isError''
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0-py3.6.egg/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 116, in __process_devices
    if input_data.isError():
AttributeError: 'ReadHoldingRegistersResponse' object has no attribute 'isError'
```

zbeacon, 2019-12-31:
@thucar 
Got it, try clone and reinstall again, if problem won't be solved please email me ibarkov@thingsboard.io

closed 2019-12-31 by zbeacon