thingsboard/thingsboard-gateway #205: serial port exception

hasikp1, 2019-08-23:
I am receiving the following error when trying to read data using Modbus.
`2019-08-23 14:10:16,962 [pool-15-thread-1] ERROR o.t.g.extensions.modbus.ModbusClient - MBS[ttyUSB0] failed to read from MBD[speedo], tag [speed]
com.ghgande.j2mod.modbus.ModbusIOException: I/O exception - failed to read response for request [01 03 00 00 00 01] - getResponse serial port exception - Cannot read from serial port - truncated
	at com.ghgande.j2mod.modbus.io.ModbusRTUTransport.readResponseIn(ModbusRTUTransport.java:412)
	at com.ghgande.j2mod.modbus.io.ModbusSerialTransport.readResponse(ModbusSerialTransport.java:146)
	at com.ghgande.j2mod.modbus.io.ModbusSerialTransaction.execute(ModbusSerialTransaction.java:142)
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.readTag(ModbusClient.java:188)
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.lambda$readTags$7(ModbusClient.java:181)
	at java.util.LinkedList$LLSpliterator.forEachRemaining(LinkedList.java:1235)
	at java.util.stream.ReferencePipeline$Head.forEach(ReferencePipeline.java:580)
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.readTags(ModbusClient.java:181)
	at org.thingsboard.gateway.extensions.modbus.ModbusClient.lambda$null$4(ModbusClient.java:163)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
	at java.lang.Thread.run(Thread.java:745)
`
I was able to read data successfully using a  python script. What can be the issue?

zouhr, 2019-09-26:
there no crc in the request，this request 01 03 00 00 01 is wrong ，but i donot know howto resolve this problem。

zbeacon, 2019-12-31:
Hi, @hasikp1 

I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
Less footprint and memory impact
Easier to extend and customize
Improved functionality.

You can try using a new gateway for your purposes instead of a Java gateway.

We look forward to your feedback.

closed 2019-12-31 by zbeacon

hiteshpanchani, 2020-03-15:
Upgrade version of j2Mod dependecies in pom.xml 
<j2mod.version>2.5.8</j2mod.version>