thingsboard/thingsboard-gateway #206: MODBUS : it is not possible to write to slave device, json Node conversion is wrong

tohidprogram, 2019-08-28:
there is a bug in JSON data conversion  in implementation of function : onRpcCommand(String message) in MqttGatewayService.java

```
      JsonNode payload = fromString(message);
        String deviceName = payload.get("device").asText();
        Set<RpcCommandListener> listeners = rpcCommandSubs.stream()
                .filter(sub -> sub.matches(deviceName)).map(RpcCommandSubscription::getListener)
                .collect(Collectors.toSet());
        if (!listeners.isEmpty()) {
            JsonNode data = payload.get("data");
            RpcCommandData rpcCommand = new RpcCommandData();
            rpcCommand.setRequestId(data.get("id").asInt());
            rpcCommand.setMethod(data.get("method").asText());
           rpcCommand.setParams(data.get("params").asText());
```
It should be considered that 
public abstract String asText()
Method that will return a valid String representation of the container value, if the node is a value node (method **isValueNode()** returns true), otherwise empty String.
but in code when you create 
```
 JsonNode data = payload.get("data");
```
isValueNode() for following will be false
```
 rpcCommand.setParams(data.get("params").asText());
```
it is better to use ** toString()** to convert params to String for this case 
```
rpcCommand.setParams(data.get("params").toString());
```
Or  in creation of JsonNode  from data other approach be used.

mfalkvidd, 2019-09-02:
Looks like https://github.com/thingsboard/thingsboard-gateway/pull/192 is an attempt to fix this issue?

zbeacon, 2019-12-31:
Hi, @tohidprogram 

I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
Less footprint and memory impact
Easier to extend and customize
Improved functionality.

You can try using a new gateway for your purposes instead of a Java gateway.

We look forward to your feedback.

closed 2019-12-31 by zbeacon