thingsboard/thingsboard-gateway #207: failed to package or install

lincoln310, 2019-08-29:
[INFO] Scanning for projects...
[INFO]                                                                         
[INFO] ------------------------------------------------------------------------
[INFO] Building Thingsboard Gateway 2.2.1-SNAPSHOT
[INFO] ------------------------------------------------------------------------
[WARNING] The POM for org.thingsboard.common:data:jar:2.2.1-SNAPSHOT is missing, no dependency information available
[WARNING] The POM for org.thingsboard:netty-mqtt:jar:2.2.1-SNAPSHOT is missing, no dependency information available
Downloading: https://repo.maven.apache.org/maven2/com/bugsnag/bugsnag/maven-metadata.xml
Downloading: https://repo.eclipse.org/content/repositories/paho-releases/com/bugsnag/bugsnag/maven-metadata.xml
Downloading: https://jcenter.bintray.com/com/bugsnag/bugsnag/maven-metadata.xml
Downloading: http://repo.jenkins-ci.org/releases/com/bugsnag/bugsnag/maven-metadata.xml
Downloading: https://oss.sonatype.org/content/repositories/snapshots/com/bugsnag/bugsnag/maven-metadata.xml
Downloaded: https://repo.maven.apache.org/maven2/com/bugsnag/bugsnag/maven-metadata.xml (2 KB at 1.0 KB/sec)
Downloaded: https://jcenter.bintray.com/com/bugsnag/bugsnag/maven-metadata.xml (2 KB at 0.9 KB/sec)
Downloaded: https://oss.sonatype.org/content/repositories/snapshots/com/bugsnag/bugsnag/maven-metadata.xml (416 B at 0.1 KB/sec)
Downloading: https://oss.sonatype.org/content/repositories/snapshots/com/bugsnag/bugsnag/3.4.0-SNAPSHOT/maven-metadata.xml
Downloading: https://repo.eclipse.org/content/repositories/paho-releases/com/bugsnag/bugsnag/3.4.0-SNAPSHOT/maven-metadata.xml
Downloaded: https://oss.sonatype.org/content/repositories/snapshots/com/bugsnag/bugsnag/3.4.0-SNAPSHOT/maven-metadata.xml (2 KB at 0.2 KB/sec)
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 12.329 s
[INFO] Finished at: 2019-08-29T16:29:57+08:00
[INFO] Final Memory: 18M/257M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal on project gateway: Could not resolve dependencies for project org.thingsboard:gateway:jar:2.2.1-SNAPSHOT: The following artifacts could not be resolved: org.thingsboard.common:data:jar:2.2.1-SNAPSHOT, org.thingsboard:netty-mqtt:jar:2.2.1-SNAPSHOT: Could not find artifact org.thingsboard.common:data:jar:2.2.1-SNAPSHOT -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/DependencyResolutionException

Process finished with exit code 1

mp-loki, 2019-08-29:
Download and build the [main project](https://github.com/thingsboard/thingsboard) source code first

lincoln310, 2019-08-29:
@mp-loki this process cost much more....

closed 2019-08-29 by lincoln310