thingsboard/thingsboard-gateway #214: Gateway server side RPC not giving message values

asadshaik1103, 2019-09-15:
Hey everyone,
I am not getting message value in my device application script when I send a post command to the device. I followed the exact sample from mqtt extension configuration's server side rpc example  https://thingsboard.io/docs/iot-gateway/mqtt/#server-side-rpc-commands  
I am running a node script which sends a connect message to gateway. I am able to see devicesOnline = 1 in Latest Telemetry, but when I am send a post command like this one     curl -v -X POST -d @rpc-request.json $SERVER_BASE_URL/api/plugins/rpc/twoway/$DEVICE_ID \
--header "Content-Type:application/json" \
--header "X-Authorization: Bearer $JWT_TOKEN"  my device application's node script gives the output:
connected
request.topic: sensor/SN-001/request/echo/5
request.body: 
response.topic: sensor/SN-001/response/echo/5

'5' is the request id that increments after every post command, but the request.body always shows empty, i take it should display 'Hello' as per the request body's "method.params" but nothing appears.
I also installed Thingsboard IoT gateway 2.2.1 release.
Any help is appreciated

zbeacon, 2019-12-31:
Hi, @hasikp1 

I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
Less footprint and memory impact
Easier to extend and customize
Improved functionality.

You can try using a new gateway for your purposes instead of a Java gateway.

We look forward to your feedback.

zbeacon, 2019-12-31:
Hi, @asadshaik1103 

I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
Less footprint and memory impact
Easier to extend and customize
Improved functionality.

You can try using a new gateway for your purposes instead of a Java gateway.

We look forward to your feedback.

closed 2019-12-31 by zbeacon