thingsboard/thingsboard-gateway #217: Extension MQTT : assigning device type and dynamic mapping

martindenion, 2019-09-25:
Hello,
I set up a network of sensors in a building.
On each card (Zolertia RE-Mote) are connected different sensors, sometimes only brightness and other times humidity and pressure for example. I send these values via MQTT to JSON via Thingsboard Gateway.
First problem, when I create my device via MQTT, I can't assign it a type, the type is then "default" on Thingsboard
Second problem, given the diversity of my cards, some cards will return different information (key in the MQTT Thingsboard Gateway extension). For example, if I have entered the following 3 keys on the extension: temperature, humidity and pressure (in this order), to return only one pressure value, I am required to assign the temperature and humidity values at least once.
I would actually like to have a "dynamic" mapping on my extension and not "static" as it seems to be now.
Can you help me?

martindenion, 2019-10-04:
I will probably start a implementation of Thingsboard Gateway which support a dynamic mapping, as I explained in the previous comment

ssmaurya, 2019-11-09:
Device type default seems to be issue similar to  #70.

martindenion, 2019-11-24:
Hi, I have now the answer to my issue. Because Thingsboard IoT Gateway does not provide dynamic mapping, I'm developing my own app called [https://github.com/martindenion/dmway](dmway).

closed 2019-11-24 by martindenion