thingsboard/thingsboard-gateway #224: tb gatway not working with external broker

zahradehghanpur, 2019-10-20:
I configured tb-gateway.yml and set the thingsboard host and port and also set the mosquitto broker as an external broker and put it i the Extenstion of Gateway device and everything works fine . I can add device with mosquitto_pub  on the external broker but sometimes when I pub a new device  , tb-gateway can't sub it from the broker and doesn't make  this log " 2019-10-20 10:20:50,415 [MQTT Call: 19bed83e-fa90-4a49-af2c-c485885b450d] INFO  o.t.g.s.gateway.MqttGatewayService - [127.0.0.1] Device Connected!
2019-10-20 10:20:51,009 [pool-4-thread-1] INFO  o.t.g.s.gateway.MqttGatewayService - [127.0.0.1][9] Device connect event is reported to Thingsboard!"
Is this a bug of the tb-gateway?

ssmaurya, 2019-11-09:
I have also observed this issue intermittently. Since this issue is affecting my deployment, I will look into this and try to address. This seems to be issue with external broker connection as I could see that gateway statistics are getting published to platform but none of the messages on external broker get processed by gateway. @zahradehghanpur , if you have logs please share. I need to confirm if the issue that I am seeing is related

zbeacon, 2019-12-31:
Hi, @zahradehghanpur 

I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
Less footprint and memory impact
Easier to extend and customize
Improved functionality.

For example if you want send RPC request from to broker on 127.0.0.1 and you use [demo.thingsboard.io](http://demo.thingsboard.io), you should configure the gateway using [this guide](https://thingsboard.io/docs/iot-gateway/getting-started/) and [configure a MQTT connector](https://thingsboard.io/docs/iot-gateway/config/mqtt/). In your case, you should add to a section "serverSideRpc" in the mqtt configuration file following strings:

    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "/sensor/${deviceName}/request/${methodName}/",
      "valueExpression": "${params}"
    }

If the gateway receives RPC request to device with this configuration, it will send data from RPC "params" to topic "/sensor/SN-0002/request/no-reply/" on external broker.

closed 2019-12-31 by zbeacon