thingsboard/thingsboard-gateway #226: issue about thingsboard-gateway-rpc(mqtt)

worldmaomao, 2019-10-23:
After configuring mqtt extension on the gateway.
I can register new device and post device telemetry by external mqtt broker.
But I can not send rpc request from thingsboard to device which is managed by mqtt gateway.
Is it a bug? 
Or could you tell me how to do it?I can not find any more detail info about gateway rpc from https://thingsboard.io/docs/guides/,  please give me some guides or documents.
Thank you!

thucar, 2019-10-27:
I'm struggling with this as well. I've set up "Server side RPC" section for my MQTT gateway device as follows:
[image]

I'm trying to have it publish a string "ON" or "OFF" to a Tasmota device at cmnd/devicename/POWER

So far all attempts have failed - I have not been able to get it to publish anything. All the while I have incoming data from all my Tasmota devices working perfectly.

What am I missing here?

masterchen, 2019-11-01:
well , i can managed to send rpc to device . the key point is when you use RPC widget, you MUST change the rpc method getValue or setValue to match the Mothed Filter.  if you use the MQTT extension config from offical , you must change the setValue to echo, this should work!

You can try it!

But i came a problem, i can't change the Method filter to other expression. once changed , the status of extension says "not sync".@thucar  @worldmaomao

thucar, 2019-11-02:
@masterchen Thank you!!! Finally all the pieces of the puzzle fall into place and make sense.

Btw, as for your issue of extension saying "not sync" for me a quick restart of the tb-gateway service fixes that.

masterchen, 2019-11-02:
@thucar restart gateway works,thanks!

cdyangzhenyu, 2019-11-19:
Hello @thucar @masterchen, Do you have specific examples of how to configure RPC of gateway and switch control. I can't get any messages from the mqtt broker of the gateway. I'm subscribed to  "#". Thanks!

thucar, 2019-11-20:
@cdyangzhenyu which part is not working for you? Getting data from MQTT Gateway into ThingsBoard, or sending RPC from ThingsBoard to physical devices via MQTT Gateway?

cdyangzhenyu, 2019-11-20:
@thucar sending RPC from ThingsBoard to physical devices via MQTT Gateway not work.
How to debug this process, the thingsboard and gateway not any useful log. The gateway's mqtt also has no any message. I think I may have a wrong configuration, but I can't find the reason.

cdyangzhenyu, 2019-11-20:
@thucar When I click switch control, the message is sent to which queue. I use the official extended configuration to subscribe to the sensor/${devicename}/request/${methodname}/${requestid} queue on the mqtt of gateway, or the '#' queue, and I can't get the message. I want to find out where this practice message was sent and in which queue it can be seen. Switch control uses RPC type, set method uses no-reply, get method uses echo.

cdyangzhenyu, 2019-11-20:
### The tb gateway python code:
```python
import json
import paho.mqtt.client as mqtt

MQTT_HOST = "127.0.0.1"
MQTT_PORT = 1883

def on_message(client, userdata, msg):
    print 'Topic: ' + msg.topic + '\nMessage: ' + str(msg.payload)
    data = json.loads(msg.payload)

def on_connect(client, userdata, rc, *extra_params):
    print('Connected with result code ' + str(rc))
    # Subscribing to receive RPC requests
    client.subscribe('sensor/SN-0002/request/no-reply/+')

client = mqtt.Client()
# Register connect callback
client.on_connect = on_connect
# Registed publish message callback
client.on_message = on_message
client.connect(MQTT_HOST, MQTT_PORT, 60)
client.loop_forever()
```

zbeacon, 2019-12-31:
Hi, @worldmaomao 

I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
Less footprint and memory impact
Easier to extend and customize
Improved functionality.

If your request is still relevant, i can help you. You should try to use a new gateway, to solve this problem. Main flow of RPC requests from ThingsBoard to broker over the gateway is:
ThingsBoard sends RPC request to the gateway -> The gateway checks it configuration -> gateway sends data to topic in configuration. About configuration for MQTT you can read more here.

For example if you want send RPC request from to broker on 127.0.0.1 and you use demo.thingsboard.io, you should configure the gateway using this guide and configure a MQTT connector. In your case, you should add to a section "serverSideRpc" in the mqtt configuration file following strings:

    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "/sensor/${deviceName}/request/${methodName}/",
      "valueExpression": "${params}"
    }

If the gateway receives RPC request to device with this configuration, it will send data from RPC "params" to topic "/sensor/SN-0002/request/no-reply/" on external broker.

Please try to use the new gateway for your purposes.

closed 2020-01-23 by zbeacon

lillianphyoe, 2020-11-22:
How can I subscribe topic via external mqtt broker (mosquitto) from thingasbaord gateway with mqtt rpc please?

akumart21, 2024-06-13:
In my java client which subscribed to topic `v1/devices/me/rpc/request/+`. Over the period of time I see no messages coming in to java client by RPC from thingsboard rule chain . No specific error in client code or thingsboard logs. Restarting  `tb-mqtt-0` pod solves the problem.
Just wondering if any thresholds configurations am missing.