thingsboard/thingsboard-gateway #230: Working for Mod bus single tag in time series but not for multiple

ankitfidel, 2019-11-14:
Modbus.json 
{
  "server": {
    "name": "Modbus Default Server",
    "type": "tcp",
    "host": "10.128.31.39",
    "port": 502,
    "timeout": 35,
    "rtuOverTcp": false,
    "devices": [
      {
        "unitId": 4,
        "deviceName": "FM1.1 G30.1_04",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": true,
        "attributes": [
        ],
        "timeseries": [
          {
            "byteOrder": "BIG",
            "tag": "Voltage",
            "type": "long",
            "functionCode": 4,
            "registerCount": 1,
            "address": 18436
          },
          {
            "byteOrder": "BIG",
            "tag": "ph Voltage",
            "type": "long",
            "functionCode": 4,
            "registerCount": 1,
            "address": 18438
          }
        ],
        "rpc": {
          "turnLightOn": {
            "address": 4,
            "bit": 2,
            "value": true
          },
          "turnLightOff": {
            "address": 4,
            "bit": 2,
            "value": false
          },
          "getCPULoad": {
            "tag": "Integer",
            "value": 42,
            "functionCode": 16,
            "address": 0,
            "unitId": 1,
            "byteOrder": "BIG",
            "registerCount": 1
          }
        }
      }
    ]
  }
}
Getting Error as below 
''2019-11-14 14:55:01' - DEBUG - tb_device_mqtt - 163 - Data published to ThingsBoard!'
''2019-11-14 14:55:01' - DEBUG - tb_device_mqtt - 344 - Subscribed to * with id 1'
''2019-11-14 14:55:01' - DEBUG - tb_gateway_mqtt - 155 - Subscribed to *|* with id 1'
''2019-11-14 14:55:01' - INFO - tb_client - 47 - (<paho.mqtt.client.Client object at 0x7ffbbddb7c88>, None, 16, "Sending PUBLISH (d0, q1, r0, m9), 'b'v1/devices/me/telemetry'', ... (82 bytes)")'
''2019-11-14 14:55:01' - INFO - tb_client - 47 - (<paho.mqtt.client.Client object at 0x7ffbbddb7c88>, None, 16, "Sending PUBLISH (d0, q1, r0, m10), 'b'v1/devices/me/telemetry'', ... (42 bytes)")'
''2019-11-14 14:55:02' - DEBUG - modbus_connector - 198 - To modbus device FM1.1 G30.1_04, 
{'byteOrder': 'BIG', 'tag': 'Voltage', 'type': 'long', 'functionCode': 4, 'registerCount': 1, 'address': 18436, 'deviceName': 'FM1.1 G30.1_04'}'
''2019-11-14 14:55:02' - DEBUG - modbus_connector - 199 - With result ReadRegisterResponse (1)'
''2019-11-14 14:55:02' - INFO - tb_client - 47 - (<paho.mqtt.client.Client object at 0x7ffbbddb7c88>, None, 16, 'Received PUBACK (Mid: 9)')'
''2019-11-14 14:55:02' - DEBUG - tb_device_mqtt - 163 - Data published to ThingsBoard!'
''2019-11-14 14:55:02' - DEBUG - modbus_connector - 198 - To modbus device FM1.1 G30.1_04, 
{'byteOrder': 'BIG', 'tag': 'ph Voltage', 'type': 'long', 'functionCode': 4, 'registerCount': 1, 'address': 18438, 'deviceName': 'FM1.1 G30.1_04'}'
''2019-11-14 14:55:02' - DEBUG - modbus_connector - 199 - With result ReadRegisterResponse (1)'
''2019-11-14 14:55:02' - DEBUG - modbus_connector - 122 - Checking timeseries for device FM1.1 G30.1_04'
''2019-11-14 14:55:02' - ERROR - modbus_connector - 153 - list index out of range'
Traceback (most recent call last):

zbeacon, 2019-11-14:
Hello, @ankitfidel ,

We rewrote the telemetry/attribute array processing logic.
Please try last pre-released version (you can upgrade the library, using following command):
pip3 install --upgrade thingsboard-gateway --user

Please try and give some feedback.

adamfadil, 2019-11-21:
Hello @zbeacon ,
in the pre-released python version, how to build it and generate a DEB package.
thank you in advance for your support.

zbeacon, 2019-11-21:
Hi, @adamfadil ,
You should download sources and run generate_deb_package.sh, i added it in 69753db commit.
Please try it and give feedback.

adamfadil, 2019-11-21:
Hi @zbeacon, 
I got this error message when i run  ./generate_deb_package.sh 

dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
Traceback (most recent call last):
  File "setup.py", line 40, in <module>
    'thingsboard-gateway = thingsboard_gateway.tb_gateway:daemon'
  File "/home/ad/.local/lib/python3.6/site-packages/setuptools/__init__.py", line 145, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3/dist-packages/stdeb/command/bdist_deb.py", line 48, in run
    util.process_command(syscmd,cwd=target_dirs[0])
  File "/usr/lib/python3/dist-packages/stdeb/util.py", line 183, in process_command
    check_call(args, cwd=cwd)
  File "/usr/lib/python3/dist-packages/stdeb/util.py", line 46, in check_call
    raise CalledProcessError(retcode)
stdeb.util.CalledProcessError: 2
Adding the files, scripts and permissions in the package
cp: cannot create directory 'deb_dist/thingsboard-gateway-2.0.*/debian/python3-thingsboard-gateway': No such file or directory
cp: cannot stat 'for_build/var': No such file or directory
cp: cannot create directory 'deb_dist/thingsboard-gateway-2.0.*/debian/python3-thingsboard-gateway': No such file or directory
chown: cannot access 'deb_dist/thingsboard-gateway-2.0.*/debian/python3-thingsboard-gateway/': No such file or directory
chown: cannot access 'deb_dist/thingsboard-gateway-2.0.*/debian/python3-thingsboard-gateway/var/': No such file or directory
chmod: cannot access 'deb_dist/thingsboard-gateway-2.0.*/debian/python3-thingsboard-gateway/DEBIAN/preinst': No such file or directory
chown: cannot access 'deb_dist/thingsboard-gateway-2.0.*/debian/python3-thingsboard-gateway/DEBIAN/preinst': No such file or directory
dpkg-deb: error: failed to open package info file 'deb_dist/thingsboard-gateway-2.0.*/debian/python3-thingsboard-gateway//DEBIAN/control' for reading: No such file or directory
cp: cannot stat 'deb_dist/thingsboard-gateway-2.0.*/debian/python3-thingsboard-gateway.deb': No such file or directory
[ad@ad-machine thingsboard-gateway (develop/2.4-python *+)]$

zbeacon, 2019-11-21:
It looks, like you run it not from thingsboard-gateway folder, this script uses relative path to build the package, please try - clone repository - change directory to thingsboard-gateway(root directory of the branch) - run command `bash ./generate_deb_package.sh`

adamfadil, 2019-11-21:
in my opinion, i think in the source code, the for_build/var  is missing

zbeacon, 2019-11-21:
Oh, really, i'm sorry. Git doesn't upload directories with no file. I added empty file, to upload it, please try again :)

adamfadil, 2019-11-21:
I changed the *  from  thingsboard-gateway-2.0.*  in  generate_deb_package.sh to hingsboard-gateway-2.0.8.1.   now the directory is generated.  but some file still missing in for_build/DEBIAN   directory .  thank you for seeing the message : 

dpkg-deb: error: failed to open package info file 'deb_dist/thingsboard-gateway-2.0.0/debian/python3-thingsboard-gateway//DEBIAN/control' for reading: No such file or directory

zbeacon, 2019-11-21:
Hmm, in this error - in the path one extra backslash: python3-thingsboard-gateway//DEBIAN

Maybe it's mistake or if in the your file no extra backslashes - problem may being caused earlier and some file didn't generate. In this case - check your output after running the script, before this error.

closed 2019-11-27 by zbeacon