thingsboard/thingsboard-gateway #231: Does the gateway support RPC forwarding?

cdyangzhenyu, 2019-11-21:
@ashvayka hi, Does the gateway support RPC forwarding?  I don't see any instructions about gateway RPC [here](https://thingsboard.io/docs/iot-gateway/getting-started/). 
My understanding of the gateway is that the gateway can start an mqtt service, let the sensor device send the data to the gateway, and the gateway will process and send it to the server of thingsboard.
According to the official documents, at present, the data can be transmitted to thingsboard server through the gateway, but the device cannot be controlled through the server.
I started a switch control to control the LED lights on the device. With the default configuration, it can't succeed. Switch control uses RPC type, set method uses no-reply, get method uses echo.
I want to know: when I click the switch, where is the RPC message sent? Is it the mqtt of thingsboard or the mqtt of gateway. No matter listening to `sensor/${devicename}/request/ ${methodname}/+` or listening `#`, no message can be obtained.
There is also the problem of how to locate. At present, there is no useful information in the log.
Here is the code that the gateway listens for mqtt and calls back(not work):
```python
import json
import paho.mqtt.client as mqtt

MQTT_HOST = "127.0.0.1"
MQTT_PORT = 1883

def on_message(client, userdata, msg):
    print 'Topic: ' + msg.topic + '\nMessage: ' + str(msg.payload)
    data = json.loads(msg.payload)

def on_connect(client, userdata, rc, *extra_params):
    print('Connected with result code ' + str(rc))
    # Subscribing to receive RPC requests
    client.subscribe('sensor/SN-0002/request/no-reply/+')

client = mqtt.Client()
# Register connect callback
client.on_connect = on_connect
# Registed publish message callback
client.on_message = on_message
client.connect(MQTT_HOST, MQTT_PORT, 60)
client.loop_forever()
```

cdyangzhenyu, 2019-11-21:
I directly listen to the `v1/devices/me/RPC/request/` queue of thingsboard's mqtt and it can receive data. Does the gateway still forward messages to the local mqtt?

```shell
mosquitto_sub -h ${THINGSBOARD_IP} -p 1883  -t v1/devices/me/rpc/request/+ -u ${DEVICE_TOKEN}
{"method":"no-reply","params":true}
{"method":"no-reply","params":false}
```

zbeacon, 2019-12-31:
Hi, @cdyangzhenyu ,
I am glad to inform you that we have released new version of the Gateway based on Python. This is a major rewrite and improvement from Java version. Main benefits:
Less footprint and memory impact
Easier to extend and customize
Improved functionality.

If your request is still relevant, i can help you. You should try to use a new gateway, to solve this problem. Main flow of RPC requests from ThingsBoard to broker over the gateway is:
ThingsBoard sends RPC request to the gateway -> The gateway checks it configuration -> gateway sends data to topic in configuration. About configuration for MQTT you can read more [here](https://thingsboard.io/docs/iot-gateway/config/mqtt/).

For example if you want send RPC request from to broker on 127.0.0.1 and you use [demo.thingsboard.io](http://demo.thingsboard.io), you should configure the gateway using [this guide](https://thingsboard.io/docs/iot-gateway/getting-started/) and [configure a MQTT connector](https://thingsboard.io/docs/iot-gateway/config/mqtt/). In your case, you should add to a section "serverSideRpc" in the mqtt configuration file following strings:
```json

    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestTopicExpression": "/sensor/${deviceName}/request/${methodName}/",
      "valueExpression": "${params}"
    }
```
If the gateway receives RPC request to device with this configuration, it will send data from RPC "params" to topic "/sensor/SN-0002/request/no-reply/" on external broker.

closed 2020-01-29 by zbeacon