thingsboard/thingsboard-gateway #233: Error while instaling , corrupted file ?

jdobruchowski, 2020-01-02:
Clean centos , while installing tb-gateway : 

`wget https://github.com/thingsboard/thingsboard-gateway/releases/download/2.0.0.1/python3-thingsboard-gateway.rpm`
`yum install -y ./python3-thingsboard-gateway.rpm`

>  Total size: 32 k
> Installed size: 32 k
> Downloading packages:
> Running transaction check
> Running transaction test
> Transaction test succeeded
> Running transaction
> thingsboard_gateway:x:996:994:ThingsBoard-Gateway Service:/var/lib/thingsboard_gateway:/bin/bash
> Removing password for user thingsboard_gateway.
> passwd: Success
> WARNING: Running pip install with root privileges is generally not a good idea. Try pip3 install --user instead.
> Requirement already satisfied: thingsboard-gateway in /usr/local/lib/python3.6/site-packages
> Requirement already satisfied: cffi in /usr/local/lib64/python3.6/site-packages (from thingsboard-gateway)
> Requirement already satisfied: pip in /usr/lib/python3.6/site-packages (from thingsboard-gateway)
> Requirement already satisfied: jsonschema==3.1.1 in /usr/local/lib/python3.6/site-packages (from thingsboard-gateway)
> Requirement already satisfied: lxml in /usr/local/lib64/python3.6/site-packages (from thingsboard-gateway)
> Requirement already satisfied: opcua in /usr/local/lib/python3.6/site-packages (from thingsboard-gateway)
> Requirement already satisfied: paho-mqtt in /usr/local/lib/python3.6/site-packages (from thingsboard-gateway)
> Requirement already satisfied: pymodbus in /usr/local/lib/python3.6/site-packages (from thingsboard-gateway)
> Requirement already satisfied: pyserial in /usr/local/lib/python3.6/site-packages (from thingsboard-gateway)
> Requirement already satisfied: pytz in /usr/local/lib/python3.6/site-packages (from thingsboard-gateway)
> Requirement already satisfied: PyYAML in /usr/local/lib64/python3.6/site-packages (from thingsboard-gateway)
> Requirement already satisfied: simplejson in /usr/local/lib/python3.6/site-packages (from thingsboard-gateway)
> Requirement already satisfied: pyrsistent in /usr/local/lib64/python3.6/site-packages (from thingsboard-gateway)
> Requirement already satisfied: pycparser in /usr/local/lib/python3.6/site-packages (from cffi->thingsboard-gateway)
> Requirement already satisfied: setuptools in /usr/lib/python3.6/site-packages (from jsonschema==3.1.1->thingsboard-gateway)
> Requirement already satisfied: attrs>=17.4.0 in /usr/local/lib/python3.6/site-packages (from jsonschema==3.1.1->thingsboard-gateway)
> Requirement already satisfied: importlib-metadata in /usr/local/lib/python3.6/site-packages (from jsonschema==3.1.1->thingsboard-gateway)
> Requirement already satisfied: six>=1.11.0 in /usr/local/lib/python3.6/site-packages (from jsonschema==3.1.1->thingsboard-gateway)
> Requirement already satisfied: python-dateutil in /usr/local/lib/python3.6/site-packages (from opcua->thingsboard-gateway)
> Requirement already satisfied: zipp>=0.5 in /usr/local/lib/python3.6/site-packages (from importlib-metadata->jsonschema==3.1.1->thingsboard-gateway)
> Requirement already satisfied: more-itertools in /usr/local/lib/python3.6/site-packages (from zipp>=0.5->importlib-metadata->jsonschema==3.1.1->thingsboard-gateway)
>   Installing : thingsboard-gateway-2.0-1.noarch                                                                                                                                                                                          1/1
> Error unpacking rpm package thingsboard-gateway-2.0-1.noarch
> error: unpacking of archive failed on file /var/lib/thingsboard_gateway/extensions/serial/custom_serial_connector.py;5e0db162: cpio: read
>   Verifying  : thingsboard-gateway-2.0-1.noarch                                                                                                                                                                                          1/1
> 
> Failed:
>   thingsboard-gateway.noarch 0:2.0-1
> 
> Complete!
> 

**Service won't start** 

`
[root@localhost etc]# systemctl status thingsboard-gateway
`
> ● thingsboard-gateway.service - ThingsBoard Gateway
>    Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; disabled; vendor preset: disabled)
>    Active: failed (Result: start-limit) since Thu 2020-01-02 04:10:21 EST; 7s ago
>   Process: 12554 ExecStop=/bin/kill -INT $MAINPID (code=exited, status=1/FAILURE)
>   Process: 12548 ExecStart=/usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon() (code=exited, status=1/FAILURE)
>  Main PID: 12548 (code=exited, status=1/FAILURE)
> Jan 02 04:10:21 localhost.localdomain systemd[1]: thingsboard-gateway.service: control process exited, code=exited status=1
> Jan 02 04:10:21 localhost.localdomain systemd[1]: Unit thingsboard-gateway.service entered failed state.
> Jan 02 04:10:21 localhost.localdomain systemd[1]: thingsboard-gateway.service failed.
> Jan 02 04:10:21 localhost.localdomain systemd[1]: thingsboard-gateway.service holdoff time over, scheduling restart.
> Jan 02 04:10:21 localhost.localdomain systemd[1]: start request repeated too quickly for thingsboard-gateway.service
> Jan 02 04:10:21 localhost.localdomain systemd[1]: Failed to start ThingsBoard Gateway.
> Jan 02 04:10:21 localhost.localdomain systemd[1]: Unit thingsboard-gateway.service entered failed state.
> Jan 02 04:10:21 localhost.localdomain systemd[1]: thingsboard-gateway.service failed.
> 

Trying to run from console : 
`
/usr/bin/python3 -c "from thingsboard_gateway.tb_gateway import daemon; daemon()"`

> Traceback (most recent call last):
>   File "<string>", line 1, in <module>
>   File "/usr/local/lib/python3.6/site-packages/thingsboard_gateway/tb_gateway.py", line 24, in daemon
>     TBGatewayService("/etc/thingsboard-gateway/config/tb_gateway.yaml")
>   File "/usr/local/lib/python3.6/site-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 41, in __init__
>     logging.config.fileConfig(self.__config_dir + "logs.conf")
>   File "/usr/lib64/python3.6/logging/config.py", line 84, in fileConfig
>     handlers = _install_handlers(cp, formatters)
>   File "/usr/lib64/python3.6/logging/config.py", line 148, in _install_handlers
>     h = klass(*args)
>   File "/usr/lib64/python3.6/logging/handlers.py", line 202, in __init__
>     BaseRotatingHandler.__init__(self, filename, 'a', encoding, delay)
>   File "/usr/lib64/python3.6/logging/handlers.py", line 57, in __init__
>     logging.FileHandler.__init__(self, filename, mode, encoding, delay)
>   File "/usr/lib64/python3.6/logging/__init__.py", line 1032, in __init__
>     StreamHandler.__init__(self, self._open())
>   File "/usr/lib64/python3.6/logging/__init__.py", line 1061, in _open
>     return open(self.baseFilename, self.mode, encoding=self.encoding)
> FileNotFoundError: [Errno 2] No such file or directory: '/etc/logs/service.log'
> 

`
mkdir /etc/logs`

sloves the issiue .

But whole section in documentation about default log location is  wrong . Logs by default are stored in /etc/logs   not in /var/logs/thingsboard-gateway  ( per https://thingsboard.io/docs/iot-gateway/configuration/ ) 

also 
 
https://thingsboard.io/docs/iot-gateway/install/rpm-installation/
download link is wrong ( 404 error ) it needs to point to current version 
( https://github.com/thingsboard/thingsboard-gateway/releases/download/2.0.0.1/python3-thingsboard-gateway.rpm )

zbeacon, 2020-01-02:
Hi, @jdobruchowski 

Thank you, for your feedback. We will fix this and write to you.

zbeacon, 2020-01-02:
@jdobruchowski 
Thanks, once again, I fixed the rpm package and it works correctly.
The link has also been fixed.

closed 2020-01-02 by zbeacon