thingsboard/thingsboard-gateway #234: Possible Regex issue with OPC UA Connector - IndexError: list index out of range

kevz93, 2020-01-07:
Hello!
Below is the versions Im using:
Thingsboard @ 2.4.1
Gateway @ 2.0.0.2
KepEx Server 6.X Demo data

I am not able to scan for devices using the opc-ua.json config file. The logs are as below:
Thingsboard :
```
2020-01-07 13:28:20.520  INFO 30115 --- [ntLoopGroup-4-3] o.t.s.t.mqtt.MqttTransportHandler        : [d7ce6301-84b2-40a9-b93a-39d2ee5c63c5] Processing connect msg for client: !
2020-01-07 13:28:20.520  INFO 30115 --- [ntLoopGroup-4-3] o.t.s.t.mqtt.MqttTransportHandler        : [d7ce6301-84b2-40a9-b93a-39d2ee5c63c5] Processing connect msg for client with user name: N530rKRR83dXx0gmZoiJ!
2020-01-07 13:28:20.523  INFO 30115 --- [ool-8-worker-21] o.t.s.t.mqtt.MqttTransportHandler        : [d7ce6301-84b2-40a9-b93a-39d2ee5c63c5] Client connected!
2020-01-07 13:28:20.548  WARN 30115 --- [ntLoopGroup-4-3] o.t.s.t.mqtt.MqttTransportHandler        : [d7ce6301-84b2-40a9-b93a-39d2ee5c63c5] Failed to subscribe to [v1/gateway/attributes/response/+][AT_MOST_ONCE]
2020-01-07 13:28:20.548  WARN 30115 --- [ntLoopGroup-4-3] o.t.s.t.mqtt.MqttTransportHandler        : [d7ce6301-84b2-40a9-b93a-39d2ee5c63c5] Failed to subscribe to [v1/gateway/rpc/+][AT_MOST_ONCE]

```
Gateway:
```
cat /var/log/thingsboard-gateway/connector.log 
''2020-01-07 13:28:22' - INFO - opcua_connector - 86 - Starting OPC-UA Connector'
''2020-01-07 13:28:22' - INFO - opcua_connector - 102 - OPC-UA connector OPC-UA Default Server Connector connected to server 192.168.0.112:49321/'
''2020-01-07 13:28:23' - ERROR - opcua_connector - 203 - list index out of range'
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 174, in __search_name
    if re.search(int_node.split('\\.')[recursion_level-2], ch.get_display_name().Text):
IndexError: list index out of range
''2020-01-07 13:28:24' - ERROR - opcua_connector - 259 - list index out of range'
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 214, in __search_tags
    if re.search(int_node.split('\\.')[recursion_level-2], ch.get_display_name().Text):
IndexError: list index out of range
''2020-01-07 13:28:24' - DEBUG - opcua_connector - 108 - Subscriptions: {}'
''2020-01-07 13:28:24' - DEBUG - opcua_connector - 109 - Available methods: {}'
```
```
cat /var/log/thingsboard-gateway/service.log
''2020-01-07 13:28:20' - DEBUG - tb_utility - 61 - Load connector for opcua class name - OpcUaConnector from /usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0-py3.6.egg/thingsboard_gateway/connectors/opcua'
''2020-01-07 13:28:20' - DEBUG - tb_utility - 66 - ModuleSpec(name='OpcUaConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f16d1c171d0>, origin='/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_converter.py')'
''2020-01-07 13:28:20' - DEBUG - tb_utility - 72 - <module 'OpcUaConnector' from '/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_converter.py'>'
''2020-01-07 13:28:20' - DEBUG - tb_utility - 66 - ModuleSpec(name='OpcUaConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f16d1bc91d0>, origin='/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py')'
''2020-01-07 13:28:20' - DEBUG - tb_utility - 72 - <module 'OpcUaConnector' from '/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py'>'
''2020-01-07 13:28:22' - DEBUG - tb_gateway_service - 380 - Loaded devices:
```
opcua.json:
```
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "192.168.0.112:49321/",
    "scanPeriodInMillis": 10000,
    "timeoutInMillis": 5000,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Channel1\\.Device\\d+$",
        "deviceNamePattern": "Device ${_System._DeviceId}",
        "attributes": [
          {
            "key": "Tag2",
            "path": "${Tag2}"
          }
        ],
        "timeseries": [
          {
            "key": "Tag1",
            "path": "${Tag1}"
          },
          {
            "key": "Tag2",
            "path": "${Tag2}"
          }
        ]
      }
    ]
  }
}

```

zbeacon, 2020-01-07:
Hi, @kevz93 

Thank you for your feedback.
We know about this problem and work to solve it. Also please try use some variable in device object on the OPC-UA server as a name.

zbeacon, 2020-01-09:
Hi, @kevz93 

I created a fix for this problem, please try installing the gateway from the sources and give feedback after that.

Thank you.

kevz93, 2020-01-09:
Hey,
Thank you so much for the quick response and fix.
I have tried running from source, my connection to OPC UA server is being timed out. Is this an expected behavior from running from source (./tb_gatway).
Do I need to install it also ?

zbeacon, 2020-01-09:
I mean try to install from sources using [this guide](https://thingsboard.io/docs/iot-gateway/install/source-installation/)

You should just clone, run installation and restart the gateway. Timeout connection can be if you run 2 gateways at same time.

kevz93, 2020-01-13:
I have tried connecting. The previous error seems to be resolved but connection keeps dropping out. I will try again on another system and report to ensure its not system related issue.

zbeacon, 2020-01-13:
> I have tried connecting. The previous error seems to be resolved but connection keeps dropping out. I will try again on another system and report to ensure its not system related issue.

Hi, @kevz93 

Thank you for your tries, if problem won't be solved please attach logs to next message.

kevz93, 2020-01-17:
hello! The regex error seems to be resolved. But  Im still not able to scan through OPC tree and see on thingsboard. 

When I connect to thingsboard : (notice the 'fail to subscribe' messages)
```

2020-01-17 17:16:51.120  INFO 844 --- [tLoopGroup-4-12] o.t.s.t.mqtt.MqttTransportHandler        : [dc703039-dc45-4b1f-8697-e938ad44418a] Processing connect msg for client: !

2020-01-17 17:16:51.120  INFO 844 --- [tLoopGroup-4-12] o.t.s.t.mqtt.MqttTransportHandler        : [dc703039-dc45-4b1f-8697-e938ad44418a] Processing connect msg for client with user name: 1nff5X7kTtpxkYbEO2AH!

2020-01-17 17:16:51.122  INFO 844 --- [ool-8-worker-17] o.t.s.t.mqtt.MqttTransportHandler        : [dc703039-dc45-4b1f-8697-e938ad44418a] Client connected!

2020-01-17 17:16:51.286  WARN 844 --- [tLoopGroup-4-12] o.t.s.t.mqtt.MqttTransportHandler        : [dc703039-dc45-4b1f-8697-e938ad44418a] Failed to subscribe to [v1/gateway/attributes/response/+][AT_MOST_ONCE]

2020-01-17 17:16:51.286  WARN 844 --- [tLoopGroup-4-12] o.t.s.t.mqtt.MqttTransportHandler        : [dc703039-dc45-4b1f-8697-e938ad44418a] Failed to subscribe to [v1/gateway/rpc/+][AT_MOST_ONCE]
```

OPC-python output when started:
```
sudo python3 ./thingsboard_gateway/tb_gateway.py
''2020-01-17 17:18:26' - DEBUG - tb_utility - 61 - Load connector for opcua class name - OpcUaConnector from /usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0-py3.6.egg/thingsboard_gateway/connectors/opcua'
''2020-01-17 17:18:26' - DEBUG - tb_utility - 66 - ModuleSpec(name='OpcUaConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fb392e2f630>, origin='/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_converter.py')'
''2020-01-17 17:18:26' - DEBUG - tb_utility - 72 - <module 'OpcUaConnector' from '/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_converter.py'>'
''2020-01-17 17:18:26' - DEBUG - tb_client - 78 - connecting to ThingsBoard'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, "Sending CONNECT (u1, p0, wr0, wq0, wf0, c1, k60) client_id=b''")'
''2020-01-17 17:18:26' - DEBUG - tb_utility - 66 - ModuleSpec(name='OpcUaConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fb392e17668>, origin='/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py')'
''2020-01-17 17:18:26' - DEBUG - tb_utility - 72 - <module 'OpcUaConnector' from '/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py'>'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, 'Received CONNACK (1, 0)')'
''2020-01-17 17:18:26' - DEBUG - tb_client - 61 - Gateway connected to ThingsBoard'
''2020-01-17 17:18:26' - INFO - tb_device_mqtt - 182 - connection SUCCESS'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, "Sending SUBSCRIBE (d0, m1) [(b'v1/devices/me/attributes', 1)]")'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, "Sending SUBSCRIBE (d0, m2) [(b'v1/devices/me/attributes/response/+', 1)]")'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, "Sending SUBSCRIBE (d0, m3) [(b'v1/devices/me/rpc/request/+', 0)]")'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, "Sending SUBSCRIBE (d0, m4) [(b'v1/devices/me/rpc/response/+', 1)]")'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, "Sending SUBSCRIBE (d0, m5) [(b'v1/gateway/attributes', 1)]")'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, "Sending SUBSCRIBE (d0, m6) [(b'v1/gateway/attributes/response/+', 0)]")'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, "Sending SUBSCRIBE (d0, m7) [(b'v1/gateway/rpc/+', 0)]")'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, 'Received SUBACK')'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, 'Received SUBACK')'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, 'Received SUBACK')'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, 'Received SUBACK')'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, 'Received SUBACK')'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, 'Received SUBACK')'
''2020-01-17 17:18:26' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7fb392e176d8>, None, 16, 'Received SUBACK')'
''2020-01-17 17:18:33' - INFO - opcua_connector - 86 - Starting OPC-UA Connector'
''2020-01-17 17:18:33' - DEBUG - tb_gateway_service - 380 - Loaded devices:
 {}'
''2020-01-17 17:18:33' - DEBUG - tb_device_mqtt - 349 - Subscribed to * with id 1'
[... cut ...]
    "name": "OPC-UA Default Server",
[... 8 lines cut ...]
      {
        "deviceNodePattern": "Channel1\\.Device1",
        "deviceNamePattern": "Device {$_System._DeviceId}",
        "attributes": [
          {
            "key": "Tag2",
            "path": "${Tag2}"
          }
        ],
        "timeseries": [
          {
            "key": "Tag1",
            "path": "${Tag1}"
          },
          {
            "key": "Tag2",
            "path": "${Tag2}"
          }
        ]
      }
    ]
  }
}

```

zbeacon, 2020-01-17:
Hi, @kevz93 

Please change "deviceNamePattern" expression to "Device ${_System._DeviceId}" and try to run the gateway again.

kevz93, 2020-01-20:
Thank you @zbeacon for pointing that out. I have updated logs from the OPC connector:
```
2020-01-20 11:39:48' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, 'Received CONNACK (1, 0)')'
''2020-01-20 11:39:48' - DEBUG - tb_client - 61 - Gateway connected to ThingsBoard'
''2020-01-20 11:39:48' - INFO - tb_device_mqtt - 182 - connection SUCCESS'
''2020-01-20 11:39:48' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, "Sending SUBSCRIBE (d0, m1) [(b'v1/devices/me/attributes', 1)]")'
''2020-01-20 11:39:48' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, "Sending SUBSCRIBE (d0, m2) [(b'v1/devices/me/attributes/response/+', 1)]")'
''2020-01-20 11:39:48' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, "Sending SUBSCRIBE (d0, m3) [(b'v1/devices/me/rpc/request/+', 0)]")'
''2020-01-20 11:39:48' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, "Sending SUBSCRIBE (d0, m4) [(b'v1/devices/me/rpc/response/+', 1)]")'
''2020-01-20 11:39:48' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, "Sending SUBSCRIBE (d0, m5) [(b'v1/gateway/attributes', 1)]")'
''2020-01-20 11:39:48' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, "Sending SUBSCRIBE (d0, m6) [(b'v1/gateway/attributes/response/+', 0)]")'
''2020-01-20 11:39:48' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, "Sending SUBSCRIBE (d0, m7) [(b'v1/gateway/rpc/+', 0)]")'
''2020-01-20 11:39:48' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, 'Received SUBACK')'
''2020-01-20 11:39:48' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, 'Received SUBACK')'
''2020-01-20 11:39:48' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, 'Received SUBACK')'
''2020-01-20 11:39:48' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, 'Received SUBACK')'
''2020-01-20 11:39:48' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, 'Received SUBACK')'
''2020-01-20 11:39:48' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, 'Received SUBACK')'
''2020-01-20 11:39:48' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, 'Received SUBACK')'
''2020-01-20 11:39:50' - INFO - opcua_connector - 86 - Starting OPC-UA Connector'
''2020-01-20 11:39:50' - DEBUG - tb_gateway_service - 380 - Loaded devices:
 {}'
''2020-01-20 11:39:50' - DEBUG - tb_device_mqtt - 349 - Subscribed to * with id 1'
''2020-01-20 11:39:50' - DEBUG - tb_gateway_mqtt - 155 - Subscribed to *|* with id 1'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 93 - KEPServerEX'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 94 - 2'
''2020-01-20 11:39:50' - INFO - opcua_connector - 104 - OPC-UA connector OPC-UA Default Server Connector connected to server 192.168.0.116:49321/'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _ActiveTagCount'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _ClientCount'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _Date'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _Date_Day'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _Date_DayOfWeek'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _Date_Month'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _Date_Year2'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _Date_Year4'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _DateTime'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _DateTimeLocal'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _ExpiredFeatures'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _FullProjectName'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _LicensedFeatures'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _OpcClientNames'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _ProductName'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _ProductVersion'
''2020-01-20 11:39:50' - DEBUG - opcua_connector - 192 - _ProjectName'
''2020-01-20 11:39:51' - DEBUG - opcua_connector - 192 - _ProjectTitle'
''2020-01-20 11:39:51' - DEBUG - opcua_connector - 192 - _Time'
''2020-01-20 11:39:51' - DEBUG - opcua_connector - 192 - _Time_Hour'
''2020-01-20 11:39:51' - DEBUG - opcua_connector - 192 - _Time_Hour24'
''2020-01-20 11:39:51' - DEBUG - opcua_connector - 192 - _Time_Minute'
''2020-01-20 11:39:51' - DEBUG - opcua_connector - 192 - _Time_PM'
''2020-01-20 11:39:51' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, "Sending PUBLISH (d0, q1, r0, m8), 'b'v1/devices/me/telemetry'', ... (96 bytes)")'
''2020-01-20 11:39:51' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, "Sending PUBLISH (d0, q1, r0, m9), 'b'v1/devices/me/telemetry'', ... (40 bytes)")'
''2020-01-20 11:39:51' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, 'Received PUBACK (Mid: 8)')'
[... cut ...]
''2020-01-20 11:40:52' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, "Sending PUBLISH (d0, q1, r0, m11), 'b'v1/devices/me/telemetry'', ... (40 bytes)")'
''2020-01-20 11:40:52' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, 'Received PUBACK (Mid: 10)')'
''2020-01-20 11:40:52' - DEBUG - tb_client - 54 - (<paho.mqtt.client.Client object at 0x7f8bb204c6d8>, None, 16, 'Received PUBACK (Mid: 11)')'
```

I am still not able to see the devices. =(    Im using thingsboard 2.4.1

zbeacon, 2020-01-20:
Hi @kevz93 

Got it, It has been caused because OPC-UA connector was looking for interesting nodes only in objects node in the OPC-UA server.
Please upgrade to version 2.0.0.6 and try to run it with config below. Also please add object to the device with name "Tag2" and some value. The gateway will not add device to the ThingsBoard if it doesn't have any interest parameter.

```json
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "192.168.0.116:49321/",
    "scanPeriodInMillis": 10000,
    "timeoutInMillis": 5000,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Channel1\\.Device1",
        "deviceNamePattern": "Device ${_System\\._DeviceId}",
        "attributes": [
          {
            "key": "Tag2",
            "path": "${Tag2}"
          }
        ],
        "timeseries": [
          {
            "key": "Tag1",
            "path": "${Tag1}"
          },
          {
            "key": "Tag2",
            "path": "${Tag2}"
          }
        ]
      }
    ]
  }
}
```

kevz93, 2020-01-20:
Thank you for your response. I have pulled the latest changes from master. Running with same config. Now getting a logging error :
```
''2020-01-20 15:09:29' - DEBUG - tb_client - 94 - connecting to ThingsBoard'
''2020-01-20 15:09:29' - DEBUG - tb_device_mqtt - 343 - Subscribed to * with id 1'
''2020-01-20 15:09:29' - DEBUG - tb_gateway_mqtt - 172 - Subscribed to *|* with id 1'
Traceback (most recent call last):
  File "./thingsboard_gateway/tb_gateway.py", line 28, in <module>
    main()
  File "./thingsboard_gateway/tb_gateway.py", line 20, in main
    TBGatewayService(path.dirname(path.abspath(__file__)) + '//config//tb_gateway.yaml')
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0.0.5-py3.6.egg/thingsboard_gateway/gateway/tb_gateway_service.py", line 64, in __init__
    self.remote_handler = TBLoggerHandler(self)
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0.0.5-py3.6.egg/thingsboard_gateway/gateway/tb_logger.py", line 24, in __init__
    super().__init__(logging.getLevelName(self.__current_log_level))
AttributeError: 'TBLoggerHandler' object has no attribute '_TBLoggerHandler__current_log_level'

```

Can you also explain " The gateway will not add device to the ThingsBoard if it doesn't have any interest parameter. " ?

zbeacon, 2020-01-20:
>> I have pulled the latest changes from master. Running with same config. 
Please try again.

>>  The gateway will not add device to the ThingsBoard if it doesn't have any interest parameter.
I mean, that if the gateway hasn't found any attribute or telemetry node in the OPC-UA server, it will not publish device to the ThingsBoard. 

I have added logging when the gateway found device name. The logging message will be like: "Full device name is: Device 1"

kevz93, 2020-01-20:
I am getting error :
``` AttributeError: 'TBLoggerHandler' object has no attribute '_TBLoggerHandler__current_log_level' ```

kevz93, 2020-01-20:
Thank you for your time and patience @zbeacon  
I am getting a new traceback log:
```
sudo python3 ./thingsboard_gateway/tb_gateway.py
''2020-01-20 16:49:36' - DEBUG - tb_client - 94 - connecting to ThingsBoard'
''2020-01-20 16:49:36' - DEBUG - tb_device_mqtt - 343 - Subscribed to * with id 1'
''2020-01-20 16:49:36' - DEBUG - tb_gateway_mqtt - 172 - Subscribed to *|* with id 1'
''2020-01-20 16:49:36' - DEBUG - tb_logger - 36 - Added remote handler to log service'
''2020-01-20 16:49:36' - DEBUG - tb_logger - 36 - Added remote handler to log storage'
''2020-01-20 16:49:36' - DEBUG - tb_logger - 36 - Added remote handler to log extension'
''2020-01-20 16:49:36' - DEBUG - tb_logger - 36 - Added remote handler to log connector'
''2020-01-20 16:49:36' - DEBUG - tb_logger - 36 - Added remote handler to log tb_connection'
''2020-01-20 16:49:36' - DEBUG - tb_utility - 66 - Load connector for opcua class name - OpcUaConnector from /usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0.0.5-py3.6.egg/thingsboard_gateway/connectors/opcua'
''2020-01-20 16:49:36' - DEBUG - tb_utility - 71 - ModuleSpec(name='OpcUaConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fef33702a20>, origin='/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0.0.5-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_converter.py')'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, "Sending CONNECT (u1, p0, wr0, wq0, wf0, c1, k60) client_id=b''")'
''2020-01-20 16:49:36' - DEBUG - tb_utility - 77 - <module 'OpcUaConnector' from '/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0.0.5-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_converter.py'>'
''2020-01-20 16:49:36' - DEBUG - tb_utility - 71 - ModuleSpec(name='OpcUaConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fef326af2b0>, origin='/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0.0.5-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py')'
''2020-01-20 16:49:36' - DEBUG - tb_utility - 77 - <module 'OpcUaConnector' from '/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0.0.5-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py'>'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, 'Received CONNACK (1, 0)')'
''2020-01-20 16:49:36' - DEBUG - tb_client - 70 - TB client <paho.mqtt.client.Client object at 0x7fef337027f0> connected to ThingsBoard'
''2020-01-20 16:49:36' - INFO - tb_device_mqtt - 184 - connection SUCCESS'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, "Sending SUBSCRIBE (d0, m1) [(b'v1/devices/me/attributes', 1)]")'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, "Sending SUBSCRIBE (d0, m2) [(b'v1/devices/me/attributes/response/+', 1)]")'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, "Sending SUBSCRIBE (d0, m3) [(b'v1/devices/me/rpc/request/+', 0)]")'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, "Sending SUBSCRIBE (d0, m4) [(b'v1/devices/me/rpc/response/+', 1)]")'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, "Sending SUBSCRIBE (d0, m5) [(b'v1/gateway/attributes', 1)]")'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, "Sending SUBSCRIBE (d0, m6) [(b'v1/gateway/attributes/response/+', 0)]")'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, "Sending SUBSCRIBE (d0, m7) [(b'v1/gateway/rpc/+', 0)]")'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, 'Received SUBACK')'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, 'Received SUBACK')'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, 'Received SUBACK')'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, 'Received SUBACK')'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, 'Received SUBACK')'
''2020-01-20 16:49:36' - DEBUG - tb_gateway_mqtt - 63 - Service subscription to topic v1/gateway/attributes - successfully completed.'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, 'Received SUBACK')'
''2020-01-20 16:49:36' - DEBUG - tb_gateway_mqtt - 63 - Service subscription to topic v1/gateway/attributes/response - successfully completed.'
''2020-01-20 16:49:36' - DEBUG - tb_client - 58 - (<paho.mqtt.client.Client object at 0x7fef337027f0>, None, 16, 'Received SUBACK')'
''2020-01-20 16:49:36' - DEBUG - tb_gateway_mqtt - 63 - Service subscription to topic v1/gateway/rpc - successfully completed.'
''2020-01-20 16:49:38' - INFO - opcua_connector - 86 - Starting OPC-UA Connector'
''2020-01-20 16:49:38' - DEBUG - tb_gateway_service - 454 - Loaded devices:
 {}'
''2020-01-20 16:49:38' - DEBUG - opcua_connector - 93 - KEPServerEX'
[... cut ...]
  File "/usr/lib/python3.6/sre_parse.py", line 416, in _parse_sub
    not nested and not items))
  File "/usr/lib/python3.6/sre_parse.py", line 488, in _parse
    sourceget()
  File "/usr/lib/python3.6/sre_parse.py", line 255, in get
    self.__next()
  File "/usr/lib/python3.6/sre_parse.py", line 245, in __next
    self.string, len(self.string) - 1) from None
sre_constants.error: bad escape (end of pattern) at position 8
''2020-01-20 16:49:39' - ERROR - opcua_connector - 311 - bad escape (end of pattern) at position 8'
```

kevz93, 2020-01-20:
I think  this is a regex issue I will try again.

kevz93, 2020-01-20:
It was a regex issue, based on this https://bugs.python.org/issue29015.
The connector instructions should be updated to use '\\\\' or r'\\'.

zbeacon, 2020-01-20:
Ok, thanks for your research. Please leave feedback about the data processing.

kevz93, 2020-01-22:
Im still figuring how the recursive search works with namePattern. I get more results in scan when I switch to "objects" instead of "root" in opcua-connector.py

zbeacon, 2020-01-22:
Hi @kevz93 

I have added option to the OPC-UA configuration - 
```
    "showMap": true,
```
Just add this parameter to the "server" configuration section and you will can see in the console nodes are beign checked. I have also rewritten a little bit connector. Please try to use it (Branch develop/2.4-python in the repository). It will scan from root node and for example use something like following configuration to get name of device from ```deviceName``` node (```Root.Objects.MyObject1.MyStringVariable```) and read telemetry and attributes tags ( from ```Root.Objects.MyObject1.MyVariable1```, ```Root.Objects.MyObject1.MyVariable2```, etc.) from deviceNode (```Root.Objects.MyObject1```) :

```
      {
        "deviceNodePattern": "Root\\.Objects\\.MyObject\\d+",
        "deviceNamePattern": "Device ${Root\\.Objects\\.MyObject\\d+\\.MyStringVariable}",
        "attributes": [
          {
            "key": "Tag1",
            "path": "${MyVariable1}"
          }
        ],
        "timeseries": [
          {
            "key": "Tag3",
            "path": "${MyVariable2}"
          },
          {
            "key": "Tag2",
            "path": "${MyVariable3}"
          }
        ]
      }
```

Please try and give feedback.

kevz93, 2020-01-24:
Still getting an error : 
```
''2020-01-24 15:00:18' - ERROR - opcua_connector - 238 - list index out of range'
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0.0.7-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 185, in __search_name
    re.search(TBUtility.get_value(interest_node[int_node]["deviceNodePattern"].split("\\.")[recursion_level+1], get_tag=True), ch.get_display_name().Text):
IndexError: list index out of range
```

and

```
''2020-01-24 15:00:18' - ERROR - opcua_connector - 318 - list index out of range'
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0.0.7-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 284, in __search_tags
    if re.search(name_to_check.replace('$', ''), current_var_path.split(".")[-2]):
IndexError: list index out of range
```

zbeacon, 2020-01-24:
@kevz93 

Ok, try to set full path to the objects and variables (From Root) like: 
```
        "deviceNodePattern": "Root\\.Objects\\.MyObject\\d+",
        "deviceNamePattern": "Device ${Root\\.Objects\\.MyObject\\d+\\.MyVariable22}",
```

If you are using the last version from sources, i have changed search from Root node ( For purpose, if some server variables are required.) I am also working on updating documentation.

zbeacon, 2020-01-24:
@kevz93 

We just published new pre release (2.0.0.8), it will be available in the pip in few minutes. Try to update your gateway to the new version and check.

kevz93, 2020-02-04:
Hello,
Sorry for the late response. I have upgraded to 2.0.0.9.
The search throws an error when a match is found:
```
''2020-02-04 14:57:49' - INFO - opcua_connector - 178 - Looking for name: Root.Objects.Demo.Massfolder_Static'
''2020-02-04 14:57:52' - INFO - opcua_connector - 178 - Looking for name: Root.Objects.Demo.Massfolder_Dynamic'
''2020-02-04 14:57:56' - INFO - opcua_connector - 178 - Looking for name: Root.Objects.Demo.DynamicNodes'
''2020-02-04 14:57:59' - INFO - opcua_connector - 178 - Looking for name: Root.Objects.Demo.BoilerDemo'
''2020-02-04 14:58:04' - INFO - opcua_connector - 178 - Looking for name: Root.Objects.Demo.BoilerDemo.Boiler1'
''2020-02-04 14:58:06' - DEBUG - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f5e49678908>, None, 16, 'Sending PINGREQ')'
''2020-02-04 14:58:06' - DEBUG - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f5e49678908>, None, 16, 'Received PINGRESP')'
''2020-02-04 14:58:07' - DEBUG - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f5e49678908>, None, 16, "Sending PUBLISH (d0, q1, r0, m10), 'b'v1/devices/me/telemetry'', ... (70 bytes)")'
''2020-02-04 14:58:07' - DEBUG - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f5e49678908>, None, 16, "Sending PUBLISH (d0, q1, r0, m11), 'b'v1/devices/me/telemetry'', ... (40 bytes)")'
''2020-02-04 14:58:07' - DEBUG - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f5e49678908>, None, 16, 'Received PUBACK (Mid: 10)')'
''2020-02-04 14:58:07' - DEBUG - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f5e49678908>, None, 16, 'Received PUBACK (Mid: 11)')'
''2020-02-04 14:58:08' - INFO - opcua_connector - 178 - Looking for name: Root.Objects.Demo.BoilerDemo.Boiler1.Fill'
''2020-02-04 14:58:13' - INFO - opcua_connector - 178 - Looking for name: Root.Objects.Demo.BoilerDemo.Boiler1.FillLevelSensor'
''2020-02-04 14:58:13' - ERROR - opcua_connector - 239 - list index out of range'
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0.0.9-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 186, in __search_name
    re.search(TBUtility.get_value(interest_node[int_node]["deviceNodePattern"].split("\\.")[recursion_level+1], get_tag=True), ch.get_display_name().Text):
IndexError: list index out of range
```

Search is based on : 
```
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "opc.tcp://opcuaserver.com:48010",
    "scanPeriodInMillis": 10000,
    "showMap": true,
    "timeoutInMillis": 5000,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Demo\\.BoilerDemo\\.Boiler1",
        "deviceNamePattern": "Device ${Root\\.Objects\\.Demo\\.BoilerDemo\\.Boiler1\\.HeaterStatus}",
        "attributes": [
          {
            "key": "HeaterStatus",
            "path": "${Demo.BoilerDemo.Boiler1.HeaterStatus}"
          }
        ],
        "timeseries": [
          {
            "key": "Temperature",
            "path": "${Demo.BoilerDemo.Boiler1.TemperatureSensor.Temperature}"
          }
        ]
      }
    ]
  }
}

```

zbeacon, 2020-02-04:
Hello,

Please try following config:

{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "opc.tcp://opcuaserver.com:48010",
    "scanPeriodInMillis": 10000,
    "showMap": true,
    "timeoutInMillis": 5000,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Demo\\.BoilerDemo\\.Boiler1",
        "deviceNamePattern": "Device
${Root\\.Objects\\.Demo\\.BoilerDemo\\.Boiler1\\.HeaterStatus}",
        "attributes": [
          {
            "key": "HeaterStatus",
            "path": "${HeaterStatus}"
          }
        ],
        "timeseries": [
          {
            "key": "Temperature",
            "path": "${TemperatureSensor\\.Temperature}"
          }
        ]
      }
    ]
  }
}

-- 
<https://thingsboard.io/>ThingsBoard Inc. <https://thingsboard.io/>

*Ilya Barkov *|| Python Developer

ibarkov@thingsboard.io

https://thingsboard.io

*pleaseDON'TprintTHISemail  *

We kindly urge you that this e-mail may contain confidential and privileged
information. It is intended only for the addressee. Unless you are the
named addressee (or authorized to receive for the addressee) you are hereby
notified that any disclosure, copying, distribution, or reliance upon the
contents of this e-mail is strictly prohibited. If you received it in
error, please notify me immediately and then destroy it.

kevz93, 2020-02-05:
Two TRACEBACKS:
```
''2020-02-05 10:45:33' - INFO - opcua_connector - 178 - Looking for name: Root.Objects.Demo.BoilerDemo.Boiler1.FillLevelSensor'
''2020-02-05 10:45:34' - ERROR - opcua_connector - 239 - list index out of range'
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0.0.9-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 186, in __search_name
    re.search(TBUtility.get_value(interest_node[int_node]["deviceNodePattern"].split("\\.")[recursion_level+1], get_tag=True), ch.get_display_name().Text):
IndexError: list index out of range

```

and 

```
''2020-02-05 10:47:01' - ERROR - opcua_connector - 264 - 'deviceName''
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0.0.9-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 261, in __search_tags
    self.__available_object_resources[interest_node[int_node]["deviceName"]]["methods"].append({method.get_display_name().Text: method,
KeyError: 'deviceName'
''2020-02-05 10:47:05' 
```

also after subscriptions which I can see in logs happened :
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.0.0.9-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 245, in __search_tags
    current_var_path = '.'.join(x.split(":")[1] for x in ch.get_path(20000, True))
  File "/home/hacklab/.local/lib/python3.6/site-packages/opcua/common/node.py", line 417, in get_path
    path = [el.get_browse_name().to_string() for el in path]
  File "/home/hacklab/.local/lib/python3.6/site-packages/opcua/common/node.py", line 417, in <listcomp>
    path = [el.get_browse_name().to_string() for el in path]
  File "/home/hacklab/.local/lib/python3.6/site-packages/opcua/common/node.py", line 76, in get_browse_name
    result = self.get_attribute(ua.AttributeIds.BrowseName)
  File "/home/hacklab/.local/lib/python3.6/site-packages/opcua/common/node.py", line 273, in get_attribute
    result = self.server.read(params)
  File "/home/hacklab/.local/lib/python3.6/site-packages/opcua/client/ua_client.py", line 337, in read
    data = self._uasocket.send_request(request)
  File "/home/hacklab/.local/lib/python3.6/site-packages/opcua/client/ua_client.py", line 83, in send_request
    data = future.result(self.timeout)
  File "/usr/lib/python3.6/concurrent/futures/_base.py", line 434, in result
    raise TimeoutError()
concurrent.futures._base.TimeoutError
```

zbeacon, 2020-02-05:
Hi @kevz93 

I rewrote a recursive search. I committed this in the development branch (development / 2.4-python), you can update the gateway or wait a few days until we release this patch in the release version.

zbeacon, 2020-02-06:
Hi @kevz93 , 

We ave released new version. I rewrote search, please try and give feedback.

kevz93, 2020-02-10:
Hey,
 I have tried new release as well as develop branch. Still getting error:
```
""2020-02-10 16:23:40" - ERROR - [opcua_connector.py] - opcua_connector - 182 - 'NoneType' object is not subscriptable"
Traceback (most recent call last):
  File "K:\thingsboard-gateway\thingsboard_gateway\connectors\opcua\opcua_connector.py", line 175, in __scan_nodes_from_config 
    if device_info["deviceNode"] is not None:
TypeError: 'NoneType' object is not subscriptable
```

zbeacon, 2020-02-10:
@kevz93 Ok, i added a handler for this exception in develop branch. But also this means that you have a wrong "deviceNode" configuration and the gateway cannot find it in the OPC-UA server.

kevz93, 2020-02-11:
Ok I will check it and confirm on that. Is it possible if you can enable showMap= true on develop ?

zbeacon, 2020-02-11:
I removed this function, when i wrote search. But if you think this feature is useful, I can add it.

kevz93, 2020-02-11:
yes, if you can add it to 'develop' it'll be great cause it'll help me debug =)

zbeacon, 2020-02-11:
@kevz93 

I have add "showMap" feature, please try to use it.

zbeacon, 2020-02-17:
Hi @kevz93 

Is problem, described in this issues has been resolved? Could i close it?

kevz93, 2020-02-19:
hey, I dont see any output with showMap=true  
Ive tried master and develop.

zbeacon, 2020-02-19:
Hi @kevz93 

Please check file "logs.conf" and set the level everywhere in it "**DEBUG**" instead of "**INFO**".

kevz93, 2020-02-19:
I did. For some reason its not showing up. Search is empty. It comes Device node is None. 
However, before, for same config I was able to see device created.

zbeacon, 2020-02-19:
Please share your logs and configs, i will check the problem.

kevz93, 2020-02-19:
opcua.json
```
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "opc.tcp://opcuaserver.com:48010",
    "scanPeriodInMillis": 10000,
    "showMap": true,
    "timeoutInMillis": 5000,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Demo\\.BoilerDemo\\.Boiler1",
        "deviceNamePattern": "Device ${Root\\.Objects\\.Demo\\.BoilerDemo\\.Boiler1\\.HeaterStatus}",
        "attributes": [
          {
            "key": "HeaterStatus",
            "path": "${HeaterStatus}"
          }
        ],
        "timeseries": [
          {
            "key": "Temperature",
            "path": "${Root\\.Objects\\.Demo\\.BoilerDemo\\.Boiler1\\.TemperatureSensor\\.Temperature}"
          }
        ],
        "attributes_updates": [
          {
            "attributeOnThingsBoard": "HeaterStatus1",
            "attributeOnDevice": "{Root\\.Objects\\.Demo\\.BoilerDemo\\.Boiler1\\.HeaterStatus"
          }]
      }
    ]
  }
}
```

```
sudo python3 ./thingsboard_gateway/tb_gateway.py | grep opcua_connector 
""2020-02-19 14:06:53" - DEBUG - [tb_utility.py] - tb_utility - 68 - ModuleSpec(name='OpcUaConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f21cc6ebd30>, origin='/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.1.0.7-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py')"
""2020-02-19 14:06:53" - DEBUG - [tb_utility.py] - tb_utility - 74 - <module 'OpcUaConnector' from '/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.1.0.7-py3.6.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py'>"
""2020-02-19 14:06:53" - INFO - [opcua_connector.py] - opcua_connector - 87 - Starting OPC-UA Connector"
""2020-02-19 14:07:02" - DEBUG - [opcua_connector.py] - opcua_connector - 98 - urn:UnifiedAutomation:CppDemoServer:UANodeSetXmlImport"
""2020-02-19 14:07:03" - DEBUG - [opcua_connector.py] - opcua_connector - 99 - 7"
""2020-02-19 14:07:03" - INFO - [opcua_connector.py] - opcua_connector - 109 - OPC-UA connector OPC-UA Connector connected to server opc.tcp://opcuaserver.com:48010"
""2020-02-19 14:07:07" - DEBUG - [opcua_connector.py] - opcua_connector - 340 - Search in Root\\.Objects"
""2020-02-19 14:07:11" - DEBUG - [opcua_connector.py] - opcua_connector - 340 - Search in Root\\.Objects\\.Demo"
""2020-02-19 14:07:39" - DEBUG - [opcua_connector.py] - opcua_connector - 340 - Search in Root\\.Objects\\.Demo\\.BoilerDemo"
""2020-02-19 14:08:21" - DEBUG - [opcua_connector.py] - opcua_connector - 340 - Search in Root\\.Objects\\.Demo"
""2020-02-19 14:08:28" - ERROR - [opcua_connector.py] - opcua_connector - 321 - Device node not found with expression: Root\.Objects\.Demo\.BoilerDemo\.Boiler1"
""2020-02-19 14:08:28" - ERROR - [opcua_connector.py] - opcua_connector - 194 - Device node is None, please check your configuration."
""2020-02-19 14:08:28" - DEBUG - [opcua_connector.py] - opcua_connector - 195 - Current device node is: Root\.Objects\.Demo\.BoilerDemo\.Boiler1"
""2020-02-19 14:08:28" - DEBUG - [opcua_connector.py] - opcua_connector - 199 - [{'Root\\.Objects\\.Demo\\.BoilerDemo\\.Boiler1': {'deviceNodePattern': 'Root\\.Objects\\.Demo\\.BoilerDemo\\.Boiler1', 'deviceNamePattern': 'Device ${Root\\.Objects\\.Demo\\.BoilerDemo\\.Boiler1\\.HeaterStatus}', 'attributes': [{'key': 'HeaterStatus', 'path': '${HeaterStatus}'}], 'timeseries': [{'key': 'Temperature', 'path': '${Root\\.Objects\\.Demo\\.BoilerDemo\\.Boiler1\\.TemperatureSensor\\.Temperature}'}], 'attributes_updates': [{'attributeOnThingsBoard': 'HeaterStatus1', 'attributeOnDevice': '{Root\\.Objects\\.Demo\\.BoilerDemo\\.Boiler1\\.HeaterStatus'}]}}]"
""2020-02-19 14:08:28" - DEBUG - [opcua_connector.py] - opcua_connector - 114 - Subscriptions: {}"
""2020-02-19 14:08:28" - DEBUG - [opcua_connector.py] - opcua_connector - 115 - Available methods: {}"
```

zbeacon, 2020-02-19:
@kevz93 

I found the problem with interpretation a method node as a variable node. Please try to use version from master branch with f70800b commit.

closed 2020-02-26 by zbeacon

kevz93, 2020-03-07:
Hello, I have tried the updated commit and looks like the recursive fucntion is able to search and register new nodes ! 
Thank you so much for the support.

Will there be a feature to write to an OPC server ? or will it be part of uplink data convertors ?