thingsboard/thingsboard-gateway #236: thingsboard-gateway connect to MQTT Broker over SSL

Bruce1221, 2020-01-18:
my mqtt.json document like this:

"broker": {
      "name":"EMQX",
      "host":"testname.com",
      "port":8883,
      "security": {
        "type": "basic",
        "username": "user",
        "password": "pass"
      }
    },
    "mapping": [

zbeacon, 2020-01-18:
Hi @Bruce1221, 

You can configure the ssl-connection to the MQTT broker using (this guide)[https://thingsboard.io/docs/iot-gateway/config/mqtt/?mqttConnectorCredentialsConfig=tls#subsection-security]. Your security configuration section will look like:
```json
  "security":{
    "caCert": "/etc/thingsboard-gateway/ca.pem",
    "privateKey": "/etc/thingsboard-gateway/privateKey.pem",
    "cert": "/etc/thingsboard-gateway/certificate.pem"
  }
```

Bruce1221, 2020-01-19:
I mean i want use TLS but not X509.
My MQTT broker use TLS and I connect it successful use MQTTfx enable the ssl/tls.

zbeacon, 2020-01-19:
@Bruce1221  

Try downloading the certificate from the server (it is used to confirm the correct operation of the server, I mean the CA certificate, add it to the gateway and specify the path to it in the caCert parameter in the configuration, you can omit the rest of the parameters).  Automatic receipt of this key has not yet been implemented, but I will add this to the list of tasks for the next releases.

Bruce1221, 2020-01-19:
It can't work,when config caCert it will log warn.and the username password will not use when connect.So,it can not connect with MQTT Broker when Broker use TLS.
Is anyway to solve it?

zbeacon, 2020-01-19:
I will try to reproduce this problem and give feedback to you. Please attach the warning from logs to your message.

Bruce1221, 2020-01-19:
Thank you.the connector log likes
''2020-01-19 15:50:06' - ERROR - mqtt_connector - 55 - caCert and cert parameters must be in config if you need to use the SSL. Please add it and try again.'
''2020-01-19 15:50:06' - DEBUG - mqtt_connector - 186 - "mqtt1" was disconnected.'
''2020-01-19 15:50:07' - DEBUG - mqtt_connector - 186 - "mqtt1" was disconnected.'

zbeacon, 2020-01-19:
@Bruce1221 

Please try with a latest version of the thingsboard-gateway library. To upgrade to the latest version, you can use a following command:
```bash
pip install thingsboard-gateway --upgrade
```
Notate: if you have installed the gateway as a daemon - you need to run command above with **sudo**.

Bruce1221, 2020-01-19:
I just install latest version yesterday.
I noticed that the code didn't set username and password when just use ssl to check the Broker?
maybe it should like :
```
mqttc.tls_set(ca_certs="ca.cer", certfile=None, keyfile=None, cert_reqs=ssl.CERT_REQUIRED, tls_version=ssl.PROTOCOL_TLSv1.2, ciphers=None)
mqttc.username_pw_set(username, password)
```

zbeacon, 2020-01-19:
>> I just install latest version yesterday.

I have fixed it few minutes ago. And i have changed username/password and ssl processing in 
7938457 commit

Bruce1221, 2020-01-19:
There is another problem,log as
'2020-01-19 18:44:42' - ERROR - mqtt_connector - 97 - [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:105

my certificate is an .cer document ,
[image]

zbeacon, 2020-01-19:
Ok, got it, then try to upgrade from sources now (branch master in the gateway repository) run:
```bash
git clone https://github.com/thingsboard/thingsboard-gateway.git
cd thingsboard-gateway
Python3 setup.py install
```

Change your security configuration section to:

```
"security":{
    "type":"tls"
    }
```

And try to run.

Bruce1221, 2020-01-20:
OK,Thank you.it's work.
Just need open tls,certificate with CA.

closed 2020-01-20 by Bruce1221

frasavino, 2021-03-01:
> Ok, got it, then try to upgrade from sources now (branch master in the gateway repository) run:
> 
> ```shell
> git clone https://github.com/thingsboard/thingsboard-gateway.git
> cd thingsboard-gateway
> Python3 setup.py install
> ```
> 
> Change your security configuration section to:
> 
> ```
> "security":{
>     "type":"tls"
>     }
> ```
> 
> And try to run.

Hi, I have a similar problem. I'm trying to connect to an external MQTT broker which has a self-signed certificate. I have an username and a password but I'm not able to connect to it because of the problem of this self-signed certificate. I have followed these steps but i don't know how to configure the security section. Might you help me? Thanks a lot ....

sonvt8, 2021-08-04:
Dear,
I have the same problem and unable to connect to MQTT broker with security type "basic", although the test messages I transfer between 2 terminal using same parameters are successfully delivered. The connector.log show the error like this:

`""2021-08-03 12:09:38" - DEBUG - [mqtt_connector.py] - mqtt_connector - 308 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0x7f4fad357400>, None, 1)" ""2021-08-03 12:11:39" - DEBUG - [mqtt_connector.py] - mqtt_connector - 308 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0x7f4fad357400>, None, 1)"`

someone help me??