thingsboard/thingsboard-gateway #237: Modbus Connector Multiple Servers

sanru95, 2020-01-20:
Hey! I'm trying to set up Thingsboard Gateway to work as a hub for a group of devices. The connector that I use is Modbus TCP. Setting up only one server (one IP) works perfectly (except the `sendDataOnlyOnChange` feature). 
The only option I have to add more devices is the `unitId` within a server (same IP). I find hard to believe that there is no option to add multiple servers, as it´s the most common use case for Modbus TCP. Different IPs for different devices.
Is this a lacking feature or just a bad configuration? Thank you

zbeacon, 2020-01-20:
Hi @sanru95 

Thank you for your interest in the gateway. This is not a flaw or a bad configuration - we decided to put it in the main configuration file - you can create several connectors with the same type but with different configuration files. Each connector will connect to its server - this allows you to differentiate the zones of responsibility of the connectors. And allowed to keep the gateway operational if one of the connectors was incorrectly configured.

sanru95, 2020-01-20:
Thank you so much for your quick answer. I tested the configuration of my devices adding different Modbus connectors to the main configuration file and it´s working flawlessly. 
On the other hand, `sendDataOnlyOnChange` doesn´t seem to be working. Changing that attribute doesn´t make any difference.

zbeacon, 2020-01-23:
Hi @sanru95 

I have fixed feature ```sendDataOnlyOnChange```, please download repository and install the gateway from it. To install the gateway from sources, please use [this guide](https://thingsboard.io/docs/iot-gateway/install/source-installation/). Please try and give feedback.

sanru95, 2020-01-23:
Hi @zbeacon, 
I tried installing the gateway from [this guide](https://thingsboard.io/docs/iot-gateway/install/source-installation/) but I get an error and cannot run properly the gateway. 
After following the steps (from a fresh formatted SD card in a Raspberry Pi), I call `sudo python3 ./thingsboard_gateway/tb_gateway.py` and I get this log:

`Traceback (most recent call last):
  File "./thingsboard_gateway/tb_gateway.py", line 28, in <module>
    main()
  File "./thingsboard_gateway/tb_gateway.py", line 20, in main
    TBGatewayService(path.dirname(path.abspath(__file__)) + '//config//tb_gatewa  y.yaml')
  File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway-2.0.0.7-py3.7  .egg/thingsboard_gateway/gateway/tb_gateway_service.py", line 45, in __init__
    logging.config.fileConfig(self._config_dir + "logs.conf")
  File "/usr/lib/python3.7/logging/config.py", line 79, in fileConfig
    handlers = _install_handlers(cp, formatters)
  File "/usr/lib/python3.7/logging/config.py", line 145, in _install_handlers
    h = klass(*args, **kwargs)
  File "/usr/lib/python3.7/logging/handlers.py", line 199, in __init__
    BaseRotatingHandler.__init__(self, filename, 'a', encoding, delay)
  File "/usr/lib/python3.7/logging/handlers.py", line 54, in __init__
    logging.FileHandler.__init__(self, filename, mode, encoding, delay)
  File "/usr/lib/python3.7/logging/__init__.py", line 1092, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python3.7/logging/__init__.py", line 1121, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/thingsboard-ga  teway/logs/service.log'`

What´s the problem?

zbeacon, 2020-01-23:
@sanru95  

Please check - is a directory with name "logs" exists in the folder "/home/pi/thingsboard-gateway/".
If it doesn't exist, just create it and try to run the gateway again.

sanru95, 2020-01-23:
@zbeacon 

I can confirm that the `sendDataOnlyOnChange` feature is now working properly. Thank you for your quick answer. When do you plan to have the new release with these updates?

zbeacon, 2020-01-23:
I think we will release a pre-release next week. We are also going to release version 2.1 of the gateway in 1-2 weeks.
Library in pip with these updates will be available after pre-release has been published.

zbeacon, 2020-01-23:
Thank you for your research and participation in the development of the gateway. If you will have any questions please email us or write an issue.

closed 2020-01-23 by zbeacon