thingsboard/thingsboard-gateway #238: Bug with sendDataOnlyOnChange fix

sanru95, 2020-01-23:
After some hours of trying the fixed `sendDataOnlyOnChange` feature, I found that values that equal 0 are not being sent to the cloud. I tried changing  `"sendDataOnlyOnChange" : true` and `"sendDataOnlyOnChange" : false` and in both cases I got the same result . 
I read values from a Modbus Server and there is a couple of parameters that change from 0 to 100, but the value 0 never arrives.
I hope it´s a quick fix!

zbeacon, 2020-01-23:
Hi @sanru95 

Are any errors in logs?

sanru95, 2020-01-23:
Yes, It shows this error:

[image]

zbeacon, 2020-01-23:
Hi @sanru95 

The bug was in Modbus uplink converter, please reinstall from sources and check.

zbeacon, 2020-01-24:
@sanru95 

You can install from pip3 - we have published new pre release.

sanru95, 2020-01-24:
Do you mean installing normally with [this guide](https://thingsboard.io/docs/iot-gateway/install/deb-installation/)?? And then  `sudo pip3 install thingsboard-gateway --upgrade` ?  Or reinstall from source?

zbeacon, 2020-01-24:
If the pre release has been published, you can update the gateway using ```sudo pip3 install thingsboard-gateway --upgrade```. But if you want the newest version, you can install from sources.

sanru95, 2020-01-24:
I was doubting because I updated the gateway using `sudo pip3 install thingsboard-gateway --upgrade` but the `sendDataOnlyOnChange` feature was not working, like in the previous pre-release. 
Now no matter the value of that parameter, values are being pushed to the cloud.

zbeacon, 2020-01-24:
Please reinstall from sources and check. If the problem was with some atrribute, it must be resolved.

sanru95, 2020-01-24:
I reinstall from source and got the same problem. Data is pushed to the cloud no matter the value of the parameter `sendDataOnlyOnChange`.

zbeacon, 2020-01-24:
@sanru95 

Ok, attach please logs to a comment.

sanru95, 2020-01-24:
[image]

zbeacon, 2020-01-24:
It's really strange. Ok, i will try to repeat this issue. I wanna please you also fully remove ThingsBoard gateway from libraries and install it from sources. To remove it please run 2 commands:
```pip3 uninstall thingsboard-gateway -y``` and ```pip3 uninstall thingsboard-gateway -y```

sanru95, 2020-01-24:
Now installing from sources it working perfectly!

zbeacon, 2020-01-24:
This problem was with different scopes of installation, e.g.:
Version 2.0.0.6 has been installed in root scope and version 2.0.0.8 has been installed as a user library and you ran it as sudo python3 tb_gateway.py

sanru95, 2020-01-24:
Thank you for your answer 👍 . I would suggest adding a keep-alive message or similar. Because even if the device is connected to the gateway and responding, if nothing changes it´s shown as inactive by the Thingsboard platform.

closed 2020-01-24 by sanru95