thingsboard/thingsboard-gateway #239: mqtt deviceNameTopicExpression issue

andihas, 2020-01-24:
I have tested thingsobard-gateway v2.0.0.7 and v2.0.0.8
Sending  an MQTT message to the topic "/sensor/SN-001/data" is not working.

`mosquitto_pub -t "/sensor/SN-001/data" -m '{"sensorType": "Thermometer", "sensorModel": "T1000", "temp": 42, "hum": 60}'`
[log.txt](https://github.com/thingsboard/thingsboard-gateway/files/4108925/log.txt)

This MQTT message is working:

`mosquitto_pub -t "/sensor/data" -m '{"serialNumber": "SN-001", "sensorType": "Thermometer", "sensorModel": "T1000", "temp": 42, "hum": 58}'`

zbeacon, 2020-01-25:
Hi @andihas,

Thank you for your interest in ThingsBoard IoT gateway. 
We have found the issue in the code, to solve it please install the latest version from sources. To install the gateway from sources you can use [this guide](https://thingsboard.io/docs/iot-gateway/install/source-installation/).

Please check and give feedback.

andihas, 2020-01-25:
Hi @zbeacon,
Thank you for your help.

I have downloaded the latest Version and tested it again.

`mosquitto_pub -t "/sensor/SN-001/data" -m '{"sensorType": "Thermometer", "sensorModel": "T1000", "temp": 42, "hum": 60}'`

now there is an different error:
[log.txt](https://github.com/thingsboard/thingsboard-gateway/files/4112121/log.txt)

zbeacon, 2020-01-27:
Hi @andihas,

We have released a new pre-release, please update your version to 2.0.0.9, check it and give feedback.

andihas, 2020-01-27:
Hi @zbeacon,

Thank you, now the MQTT Message its working and creates a new Device in Thingsboard.
For testing i deleted the device in Thingsboard and tested  the same Message again. 
The Device was not created.
When i restart thingsboard_gateway and send the Message the Device appears again.
There are no errors in the log.

zbeacon, 2020-01-27:
@andihas 

The gateway saves connected devices in the file "connected_devices.json" and if device has already been in this file, the gateway thinks it also in the ThingsBoard. You can remove this file and restart the gateway, it will create the device again.

andihas, 2020-01-27:
Okay thank you for the Information.

closed 2020-01-27 by andihas