thingsboard/thingsboard-gateway #240: mqtt json key mapping Question

andihas, 2020-01-30:
Hello,
Is it possible to map a more complex json structure like this example?

`{
   "serialNumber":"SN-001",
   "sensorType":"Thermometer",
   "sensorModel":"T1000",
   "data":{
      "temp":42,
      "hum":58
}`

and if it is possible how can i map it?

`        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${data.temp}"
          }]"` 

is not working.

zbeacon, 2020-01-31:
Hi @andihas 

Thank you for your interest in ThingsBoard IoT Gateway.
We used something similar in the early stages of development, but it really reduces the performance of the gateway. We used JSONPath to analyze the structure as you provided. I will add this function again after we solve the problem with its performance.

I will write here when this feature is added.

zbeacon, 2020-02-06:
Hi, @andihas 

I have added to MQTT converter ability to use JSONpath syntax. Please update your gateway to version 2.1 using following command:
```pip3 install thingsboard-gateway --upgrade```
and try to use configuration like:
"timeseries": [ { "type": "double", "key": "temperature", "value": "${$data.temp}" }]"

zbeacon, 2020-02-10:
Hi @andihas 

I have found the problem and fixed it in the pre-release 2.1.0.2. Please update you instance of the gateway and try to search methods. You can update using pip:
```
pip3 install thingsboard-gateway==2.1.0.2
```

andihas, 2020-02-12:
Hi @zbeacon,
Thank you!
I updated thingsboard-gateway and tried again, but with no success.

I tested with this MQTT message:

>mosquitto_pub -t "/sensor/data" -m '{ "serialNumber":"SN-001", "sensorType":"Thermometer", "sensorModel":"T1000", "data":{"temp":66, "hum":66 }}'

config:
>
       "timeseries": [
           {
             "type": "double",
             "key": "temperature",
             "value": "${$data.temp}"
           },
           {
             "type": "double",
             "key": "humidity",
             "value": "${$data.hum}"
           }
         ]

Log output:
>''2020-02-12 13:37:30' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0x7f1c9a91f6a0>, None, 16, "Received PUBLISH (d0, q0, r0, m0), '/sensor/data', ...  (108 bytes)")'

> ''2020-02-12`` 13:37:30' - DEBUG - json_mqtt_uplink_converter - 79 - "$data.temp" key not found in message: {'serialNumber': 'SN-001', 'sensorType': 'Thermometer', 'sensorModel': 'T1000', 'data': {'temp': 66, 'hum': 66}}'

>''2020-02-12 13:37:30' - DEBUG - json_mqtt_uplink_converter - 79 - "$data.hum" key not found in message: {'serialNumber': 'SN-001', 'sensorType': 'Thermometer', 'sensorModel': 'T1000', 'data': {'temp': 66, 'hum': 66}}'

zbeacon, 2020-02-14:
@andihas 

Thanks, I checked one more time and create a little fix for this, you can apply it using installation from sources. Also please use something like ```${$.data.hum}``` in value field.

andihas, 2020-02-14:
@zbeacon,
I installed from source and tested the fix, but with no success. There was no error in the logs.

mqtt:
> mosquitto_pub -t "/sensor/data" -m '{ "serialNumber":"SN-001", "sensorType":"Thermometer", "sensorModel":"T1000", "data":{"temp":66, "hum":66 }}'

config: 
>   "timeseries": [
       {
         "type": "double",
         "key": "temperature",
         "value": "${$.data.temp}"
       },
       {
         "type": "double",
         "key": "humidity",
         "value": "${$.data.hum}"
       }
     ]

I also tested the basic example, there is also a problem:
mqtt:
>mosquitto_pub -t "/sensor/SN-001/data" -m '{ "sensorType":"Thermometer", "sensorModel":"T1000", "hum":12, "temp":35}'

error:
>""2020-02-14 17:27:24" - ERROR - [json_mqtt_uplink_converter.py] - json_mqtt_uplink_converter - 81 - Error in converter, for config: 
{"type": "json", "deviceNameTopicExpression": "(?<=sensor/)(.*?)(?=/data)", "deviceTypeTopicExpression": "Thermometer", "timeout": 60000, "attributes": [{"type": "string", "key": "model", "value": "${sensorModel}"}], "timeseries": [{"type": "double", "key": "temperature", "value": "${temp}"}, {"type": "double", "key": "humidity", "value": "${hum}"}]}
 and message: 
{'sensorType': 'Thermometer', 'sensorModel': 'T1000', 'hum': 12, 'temp': 35}
"
""2020-02-14 17:27:24" - ERROR - [json_mqtt_uplink_converter.py] - json_mqtt_uplink_converter - 82 - replace() argument 2 must be str, not int"
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/thingsboard_gateway-2.1.0.6-py3.8.egg/thingsboard_gateway/connectors/mqtt/json_mqtt_uplink_converter.py", line 77, in convert
    dict_result["telemetry"].append({ts["key"]: ts["value"].replace('${' + tag + '}', ts_value)})
TypeError: replace() argument 2 must be str, not int

ToyotaCorollary, 2020-02-15:
@andihas: I got a similar TypeError, "argument must be str, not double", which I fixed by putting quote marks around the numbers in my mosquitto_pub -- in your case, try 

mosquitto_pub -t "/sensor/data" -m '{ "serialNumber":"SN-001", "sensorType":"Thermometer", "sensorModel":"T1000", "data":{"temp":"66", "hum":"66" }}'

The gateway (or Thingsboard itself) manages to interpret them as numbers anyhow.

closed 2020-02-17 by zbeacon

andihas, 2020-02-17:
@zbeacon:
Yes, the basic example is working with and without quote marks now.
mqtt:

>  mosquitto_pub -t "/sensor/SN-001/data" -m '{ "sensorType":"Thermometer", "sensorModel":"T1000", "hum":66, "temp":66}'

>  mosquitto_pub -t "/sensor/SN-001/data" -m '{ "sensorType":"Thermometer", "sensorModel":"T1000", "hum":"66", "temp":"66"}'

Unfortunately the extended example does not work for me.
I tried both:

> mosquitto_pub -t "/sensor/data" -m '{ "serialNumber":"SN-001", "sensorType":"Thermometer", "sensorModel":"T1000", "data":{"temp":66, "hum":66 }}'

> mosquitto_pub -t "/sensor/data" -m '{ "serialNumber":"SN-001", "sensorType":"Thermometer", "sensorModel":"T1000", "data":{"temp":"66", "hum":"66" }}'

[mqtt.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/4216262/mqtt.json.txt)
many thanks for your help.

falconhunter80, 2022-08-31:
@andihas 
This configuration is working for me with release 2.9

```
   "timeseries": [
       {
         "type": "double",
         "key": "temperature",
         "value": "${data.temp}"
       },
       {
         "type": "double",
         "key": "humidity",
         "value": "${data.hum}"
       }
     ]
```

commit 6149c667 (2020-02-17): Fix #240