thingsboard/thingsboard-gateway #241: opc ua deviceNodePattern

ki-do, 2020-01-31:
Hi,

is deviceNodePattern a pattern for NodeIds on the opc ua server side? does it work with the numeric NodeIds or is only for matching strings?

Thanks and regards

zbeacon, 2020-02-04:
Hi @ki-do,

At the moment, there is no possibility to use NodeIds in the gateway. But we will add this ability in one of the next releases.

ki-do, 2020-02-04:
thanks for the info. I leave my next question here as the topic seems to be the same. I'm still struggling with the deviceNodePattern and deviceNamePattern settings, being not able to set them to communicate with my server. my namespace looks like:
Root
-Objects
--MyObject
---MySubObject
----myMethod

I thought, in the opcua.json I do something like:

`"mapping": [                                                                                                                                                                                                                                                                                {                                                                                                                                                                                                                                                                                           "deviceNodePattern": "MyObject\.MySubObject\.myMethod+$",                                                                                                                                                                                                                                                    "deviceNamePattern": "MyObject\.MySubObject+$",                                                                                                                                                                                                                                              "rpc_methods": [                                                                                                                                                                                                                                                                            {                                                                                                                                                                                                                                                                                           "method": "start",                                                                                                                                                                                                                                                                        "arguments": []                                                                                                                                                                                                                                                                         }                                                                                                                                                                                                                                                                                       ]                                                                                                                                                                                                                                                                                       }`

so that gateway goes via browse pathes but neither this or any other modifications I thought of didn't work out.

Thx!

ki-do, 2020-02-07:
@zbeacon any ideas how do i bind a method call? In the meantime I did an exercise, running the freeOpcUA server instead of my custom one (which is open62541). There it does browse through the nodes. However, it doesn't find the methods (Available methods list in the connector.log remains empty).

`
''2020-02-07 16:32:23' - DEBUG - opcua_connector - 113 - Subscriptions: {}'                                                                                                                                                                                                               ''2020-02-07 16:32:23' - DEBUG - opcua_connector - 115 - Available methods: {}'
`

here's my opcua.json:
`
{                                                                                                                                                                                                                                                                                           "server": {                                                                                                                                                                                                                                                                                 "url": "opc.tcp://localhost:4840/freeopcua/server/",                                                                                                                                                                                                                                      "scanPeriodInMillis": 10000,                                                                                                                                                                                                                                                              "timeoutInMillis": 5000,                                                                                                                                                                                                                                                                  "security": "None",                                                                                                                                                                                                                                                                       "identity": {                                                                                                                                                                                                                                                                               "type": "anonymous"                                                                                                                                                                                                                                                                     },                                                                                                                                                                                                                                                                                        "mapping": [                                                                                                                                                                                                                                                                                {                                                                                                                                                                                                                                                                                           "deviceNodePattern": "Root\\.Objects\\.Device1",                                                                                                                                                                                                                                          "deviceNamePattern": "Device ${Root\\.Objects\\.Device1\\.serialNumber}",                                                                                                                                                                                                                 "attributes": [                                                                                                                                                                                                                                                                             {                                                                                                                                                                                                                                                                                           "key": "temperature",                                                                                                                                                                                                                                                                     "path": "${TemperatureAndHumiditySensor\\.Temperature}"                                                                                                                                                                                                                                 }       [... cut ...]                                                                                                                                                                                                         }                                                                                                                                                                                                                                                                                       }`

Thx for the help

zbeacon, 2020-02-10:
Hi @ki-do 

It was a problem with methods names receiving from the OPC-UA server. Please update the gateway to 2.1.0.3 version and try again to run.

You can update using following command:
```
pip3 install thingsboard-gateway --upgrade
```

ki-do, 2020-02-11:
hi @zbeacon thanks for the update. I am still not sure if it works properly, unfortunately. Looks like I still do not understand completely what does deviceNode and deviceNamePatterns do. The only way i can see something in the log under available methods is when I set both patterns to "w*". Then I got 

`''2020-02-11 18:12:56' - DEBUG - opcua_connector - 109 - Available methods: {'w*': {'methods': [{'GetMonitoredItems': Node(NumericNodeId(i=11492)), 'node': Node(NumericNodeId(i=2253))}, {'ResendData': Node(NumericNodeId(i=12873)), 'node': Node(NumericNo deId(i=2253))}, {'SetSubscriptionDurable': Node(NumericNodeId(i=12749)), 'node': Node(NumericNodeId(i=2253))}, {'RequestServerStateChange': Node(NumericNodeId(i=12886)), 'node': Node(NumericNodeId(i=2253))}, {'mymethod': Node(NumericNodeId(ns=2;i=20)),  'node': Node(NumericNodeId(ns=2;i=12))}, {'multiply': Node(NumericNodeId(ns=2;i=23)), 'node': Node(NumericNodeId(ns=2;i=12))}], 'variables': []}}'`

so the list of all the method nodes on the example server. I'd like to filter out only the multiply method out of it, but any other settings in the patterns lead to 

> Available methods: {}

any suggestions?

zbeacon, 2020-02-11:
@ki-do  

Please try use e.g. "multipl." to find the method. It looks like a problem with regular expression. Please try and give feedback, if it will work properly i will release version with fix.

ki-do, 2020-02-11:
> ''2020-02-11 18:28:20' - DEBUG - opcua_connector - 109 - Available methods: {'multipl.': {'methods': [{'GetMonitoredItems': Node(NumericNodeId(i=11492)), 'node': Node(NumericNodeId(i=2253))}, {'ResendData': Node(NumericNodeId(i=12873)), 'node': Node(Num ericNodeId(i=2253))}, {'SetSubscriptionDurable': Node(NumericNodeId(i=12749)), 'node': Node(NumericNodeId(i=2253))}, {'RequestServerStateChange': Node(NumericNodeId(i=12886)), 'node': Node(NumericNodeId(i=2253))}, {'mymethod': Node(NumericNodeId(ns=2;i= 20)), 'node': Node(NumericNodeId(ns=2;i=12))}, {'multiply': Node(NumericNodeId(ns=2;i=23)), 'node': Node(NumericNodeId(ns=2;i=12))}], 'variables': []}}'
--

zbeacon, 2020-02-11:
@ki-do  

Ok, Please specify which version you are using?  
The last one installed from the source?

ki-do, 2020-02-11:
v2.1.0.4 is it

zbeacon, 2020-02-12:
@ki-do 

Thanks, i found problem, please install from develop or master branch and check.

ki-do, 2020-02-12:
i'm afraid, still no luck 

>Feb 12 11:16:35 iot-pi python3[3172]: ''2020-02-12 11:16:35' - INFO - opcua_connector - 87 - Starting OPC-UA Connector'                                                                                                                                                                   >Feb 12 11:16:35 iot-pi python3[3172]: ''2020-02-12 11:16:35' - DEBUG - tb_gateway_service - 459 - Loaded devices:                                                                                                                                                                         >Feb 12 11:16:35 iot-pi python3[3172]:  {}'                                                                                                                                                                                                                                                >Feb 12 11:16:35 iot-pi python3[3172]: ''2020-02-12 11:16:35' - DEBUG - tb_gateway_service - 265 - Send data Thread has been started successfully.'                                                                                                                                        >Feb 12 11:16:35 iot-pi python3[3172]: ''2020-02-12 11:16:35' - DEBUG - opcua_connector - 94 - urn:freeopcua:example'                                                                                                                                                                      >Feb 12 11:16:35 iot-pi python3[3172]: ''2020-02-12 11:16:35' - DEBUG - opcua_connector - 95 - 3'                                                                                                                                                                                          >Feb 12 11:16:35 iot-pi python3[3172]: ''2020-02-12 11:16:35' - INFO - opcua_connector - 105 - OPC-UA connector OPC-UA Connector connected to server localhost:4840/freeopcua/server/'                                                                                                     >Feb 12 11:16:35 iot-pi python3[3172]: ''2020-02-12 11:16:35' - DEBUG - opcua_connector - 326 - Search in Root\\.Objects'                                                                                                                                                                  >Feb 12 11:16:36 iot-pi python3[3172]: ''2020-02-12 11:16:36' - ERROR - opcua_connector - 304 - Device name node not found with expression: multipl.'                                                                                                                                      >Feb 12 11:16:36 iot-pi python3[3172]: ''2020-02-12 11:16:36' - ERROR - opcua_connector - 190 - Device node is None, please check your configuration.'                                                                                                                                     >Feb 12 11:16:36 iot-pi python3[3172]: ''2020-02-12 11:16:36' - DEBUG - opcua_connector - 191 - Current device node is: Root\.Objects\.MyObject'                                                                                                                                           >Feb 12 11:16:36 iot-pi python3[3172]: ''2020-02-12 11:16:36' - DEBUG - opcua_connector - 195 - [{'Root\\.Objects\\.MyObject': {'deviceNodePattern': 'Root\\.Objects\\.MyObject', 'deviceNamePattern': 'multipl.', 'attributes': [{'key': 'State', 'path': '${Root\\.Objects\\.Device0001\ Feb 12 11:16:36 iot-pi python3[3172]: ''2020-02-12 11:16:36' - DEBUG - opcua_connector - 110 - Subscriptions: {}'                                                                                                                                                                         >Feb 12 11:16:36 iot-pi python3[3172]: ''2020-02-12 11:16:36' - DEBUG - opcua_connector - 111 - Available methods: {}'

zbeacon, 2020-02-12:
If you use default freeopcua server example - please try to run with following configuration for OPC-UA connector:
```json
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "opc.tcp://127.0.0.1:4840/freeopcua/server/",
    "scanPeriodInMillis": 10000,
    "timeoutInMillis": 5000,
    "showMap": true,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.MyObject",
        "deviceNamePattern": "Device ${Root\\.Objects\\.MyObject\\.MyStringVariable}",
        "attributes": [
          {
            "key": "temperature",
            "path": "${MyVariable}"
          }
        ],
        "timeseries": [
          {
            "key": "humidity",
            "path": "${MyVariable}"
          },
          {
            "key": "batteryLevel",
            "path": "${MyVariable}"
          }
        ],
        "rpc_methods": [
          {
            "method": "multiply",
            "arguments": [2, 4]
          }
        ],
        "attributes_updates": [
          {
            "attributeOnThingsBoard": "deviceName",
            "attributeOnDevice": "Root\\.Objects\\.MyObject\\.MyStringVariable"
          }
        ]
      }
    ]
  }
}```

In errors from your message - the gateway cannot find device node.

Also it can been caused by bed installation from sources. please try to run following commands and install from sources again:
```
pip3 uninstall thingsboard-gateway -y
sudo pip3 uninstall thingsboard-gateway -y
```

ki-do, 2020-02-12:
yeap, it was the device name causing the problems. now i get it without reinstalling the gateway. Thanks!

>Feb 12 11:48:32 iot-pi python3[3612]: ''2020-02-12 11:48:32' - DEBUG - opcua_connector - 195 - [{'Root\\.Objects\\.MyObject': {'deviceNodePattern': 'Root\\.Objects\\.MyObject', 'deviceNamePattern': 'Device ${Root\\.Objects\\.MyObject\\.MyStringVariable}', 'attributes': [{'key': 'S >Feb 12 11:48:32 iot-pi python3[3612]: ''2020-02-12 11:48:32' - DEBUG - opcua_connector - 110 - Subscriptions: {}'                                                                                                                                                                         >Feb 12 11:48:32 iot-pi python3[3612]: ''2020-02-12 11:48:32' - DEBUG - opcua_connector - 111 - Available methods: {'Device Really nice string': {'methods': [{'multiply': Node(NumericNodeId(ns=2;i=23)), 'node': Node(NumericNodeId(ns=2;i=12)), 'arguments': [2, 4]}]}}'

Now I come back to my question, what should I put there as a device name if I have a different namespace structure? please recall my example, where I have something like:

>Root
-Objects
--MyObject
---MySubObject
----myMethod

how should I configure gateway to get to myMethod?

zbeacon, 2020-02-12:
@ki-do 

Ok, please update to version from master branch and try to use for this purpose or full path:
```
"method": "Root\\.Objects\\.MyObject\\.MySubObject\\.myMethod"
```
or relative path from object node:
```
"method": "MySubObject\\.myMethod"
```

ki-do, 2020-02-12:
didn't work out this way.

opcua.json

```
{
  "server": {
    "name": "FreeOpcUaExampleServer",
    "url": "127.0.0.1:4840/freeopcua/server/",
    "scanPeriodInMillis": 10000,
    "timeoutInMillis": 5000,
    "showMap" : true,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.MyObject\\.MySubObject",
        "deviceNamePattern": "MySubObject",
        "rpc_methods": [
          {
            "method": "Root\\.Objects\\.MyObject\\.MySubObject\\.myMethod",
            "arguments": []
          }
        ]
      }
    ]
  }
}

```

log:

```

Feb 12 15:54:11 iot-pi python3[8017]: ''2020-02-12 15:54:11' - INFO - opcua_connector - 105 - OPC-UA connector OPC-UA Connector connected to server 127.0.0.1:4840/freeopcua/server/'
Feb 12 15:54:11 iot-pi python3[8017]: ''2020-02-12 15:54:11' - DEBUG - opcua_connector - 328 - Search in Root\\.Objects'
Feb 12 15:54:11 iot-pi python3[8017]: ''2020-02-12 15:54:11' - DEBUG - opcua_connector - 328 - Search in Root\\.Objects\\.MyObject'
Feb 12 15:54:11 iot-pi python3[8017]: ''2020-02-12 15:54:11' - ERROR - opcua_connector - 306 - Device name node not found with expression: MySubObject'
Feb 12 15:54:11 iot-pi python3[8017]: ''2020-02-12 15:54:11' - ERROR - opcua_connector - 190 - Device node is None, please check your configuration.'
Feb 12 15:54:11 iot-pi python3[8017]: ''2020-02-12 15:54:11' - DEBUG - opcua_connector - 191 - Current device node is: Root\.Objects\.MyObject\.MySubObject'
Feb 12 15:54:11 iot-pi python3[8017]: ''2020-02-12 15:54:11' - DEBUG - opcua_connector - 195 - [{'Root\\.Objects\\.MyObject\\.MySubObject': {'deviceNodePattern': 'Root\\.Objects\\.MyObject\\.MySubObject', 'deviceNamePattern': 'MySubObject', 'rpc_methods': [{'method': 'Root\\.Objects\
Feb 12 15:54:11 iot-pi python3[8017]: ''2020-02-12 15:54:11' - DEBUG - opcua_connector - 110 - Subscriptions: {}'
Feb 12 15:54:11 iot-pi python3[8017]: ''2020-02-12 15:54:11' - DEBUG - opcua_connector - 111 - Available methods: {}'

```

a sidenote: it seems, it doesn't care what's written there in the server url after IP:PORT.. my server doesn't have now this freeopcua/server endpoint, but still connector connects successfully.. could be an issue, when there is no endpoint runnning at IP:PORT directly..

zbeacon, 2020-02-12:
deviceNameNode must contains a path to a string variable with device name. Please change it and try again.

ki-do, 2020-02-12:
what is a "device name"? may i use any string for that purpose?

zbeacon, 2020-02-12:
@ki-do  Yes, device name is a value for ThingsBoard. It will show it when you open "devices" tab in UI.

ki-do, 2020-02-13:
i'd suggest then to have an opportunity to somehow hardcode it in json, as it's not opc ua specific and you can't assume, every opc ua server will have such a node for thingsboard. even more, when I tried to use /Objects/Server/ServerStatus/BuildInfo/ManufacturerName for it (as I believe any server must have it since it's NS 0), the freeopcua browse fails to find it, as one of the variables in this path is of type ServerStatus, which the stack doesn't know.
But ok, for now, I created a string with a dummy name.

I still can't connect to my method and the reason for it is the freeopcua stack again. You use get_path method to check if the paths are the same. My method, however, is a part of a custom object, which is defined as an ObjectType in my opc ua server. Therefore, the node has several paths and with the get_path you get the first one, the stack finds. In my case, it first finds the one in the ObjectTypes and then your regex match fails, obviously.

It will be also a problem for the simple methods (which are not part of the complex objects as mine). Imagine, you have 

```
Root
-Objects
--MyObject
---run
--My2ndObject
---run

```

it'll never find the second browse path. Any ideas, except fixing the opc ua stack? :-)

zbeacon, 2020-02-14:
@ki-do 

I added ability to hardcode the davice name and device type. Please install version 2.1.0.6 or use the version from the master branch on the github repository.

You are absolutely right, but, unfortunately, we use the following logic - when the method name in OPC-UA is equal to the method name in the RPC request from ThingsBoard, it processes the data. If we search for all methods with the same name and call them, the gateway will not correctly handle this RPC call.
To solve this problem, we can add another optional parameter to the configuration (e.g. rpcMethodName) and specify the method path for each method.
Or we can also develop logic, when we find more than one method with this name, we will send an array with the results of methods call.

Please give feedback that it will be more convenient from these options.

ki-do, 2020-02-14:
> To solve this problem, we can add another optional parameter to the configuration (e.g. rpcMethodName) and specify the method path for each method.
> Or we can also develop logic, when we find more than one method with this name, we will send an array with the results of methods call.

the latter is definitely not what you wanna do :-) imagine you have several objects that have run() method and then when you wanna run one of those, you immediately start all of them. normally, a method call should contain two nodeIds: one of the method and one of the method object (parent) that contains the corresponding method. if this is what you mean by your first option then I'd say yes, this is a way to go. but in this case i wouldn't see this parameter as an optional one.

zbeacon, 2020-02-14:
@ki-do  In this case you can use subObject as a general node for device, this will increase devices count on ThingsBoard but you can separate RPC calls.   
If we add identifiers of the general node to RPC, we need to specify this nodes when we use RPC call from server. This will make the method call harder.   
We can add parameter rpcMethod, it will contain method that will be called from ThingsBoard and it will be unique, e.g. run0, run1 etc. When you call method run0, the gateway will looking for key run0 in configuration and run the method in node 1, run1 = run in node 2 etc. Maybe, this is solution.   
What do you think?

ki-do, 2020-02-18:
Hi @zbeacon 

> In this case you can use subObject as a general node for device, this will increase devices count on ThingsBoard but you can separate RPC calls.

Sorry for the delay. I'll give this a try. It sounds like the most natural solution for me, even though it can lead to the explosion of the number of "devices" in my case.

> If we add identifiers of the general node to RPC, we need to specify this nodes when we use RPC call from server. This will make the method call harder.

This is how the method call is defined in the opc ua spec part 4. The request must contain objectId (The NodeId shall be that of the Object or ObjectType on which the Method is invoked), methodId (NodeId of the Method to invoke) and input arguments.

>We can add parameter rpcMethod, it will contain method that will be called from ThingsBoard and it will be unique, e.g. run0, run1 etc. When you call method run0, the gateway will looking for key run0 in configuration and run the method in node 1, run1 = run in node 2 etc. Maybe, this is solution.

The order of the Objects in the namespace is not clear, actually. I don't think this will work for a general use-case.

I'll try the first option you suggested and report ;-)

Thanks!

ki-do, 2020-02-25:
Hi @zbeacon 

I tried the approach you suggested. with the following namespace model on my server side:

```
Root
-Objects
--Object1
---SubObject
----method1 (NodeId: 4;7001)
...
----method7 (NodeId: 4;7007)
--Object2
---SubObject
----method1 (NodeId: 4;7001)
...
----method7 (NodeId: 4;7007)
-Types
--ObjectTypes
...
--------MyType
---------SubObject
----------method1 (NodeId: 4;7001)
...
----------method7 (NodeId: 4;7007)
```

So I have a defined MyType in the types folder and then I instantiate it two times in my Object folder, namely, creating Object1 and Object2. OPC UA instantiates the objects with all my methods defined for MyType. Note, that the nodeIds for the methods remain the same for the type and all its instances as these are the same methods for all the objects of the type. However, one must differentiate between the method calls for different object instances (as discussed above). So, what happens then after I configure opcua gateway like this in the opcua.json: 

```

    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Object1",
        "deviceNamePattern": "Root\\.Objects\\.DeviceName",
        "attributes": [
        ],
        "timeseries": [
        ],
        "rpc_methods": [
          {
            "method": "Root\\.Objects\\.Object1\\.SubObject\\.method1",
            "arguments": []
          },
          {
            "method": "Root\\.Objects\\.Object1\\.SubObject\\.method2",
            "arguments": []
          },
          {
            "method": "Root\\.Objects\\.Object1\\.SubObject\\.method3",
            "arguments": []
          },
          {
            "method": "Root\\.Objects\\.Object1\\.SubObject\\.method4",
            "arguments": []
          },
          {
            "method": "Root\\.Objects\\.Object1\\.SubObject\\.method5",
            "arguments": []
          },
          {
            "method": "Root\\.Objects\\.Object1\\.SubObject\\.method6",
            "arguments": []
          },
          {
            "method": "Root\\.Objects\\.Object1\\.SubObject\\.method7",
            "arguments": []
          }
        ],
        "attributes_updates": [
[... 59 lines cut ...]
''2020-02-25 11:06:43' - DEBUG - opcua_connector - 316 - current_node Node(FourByteNodeId(ns=4;i=58192)) chinld node Node(FourByteNodeId(ns=4;i=7003))'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 320 - new_node_path Root\\.Types\\.ObjectTypes\\.BaseObjectType\\.TopologyElementType\\.BlockType\\.CtrlProgramOrganizationUnitType\\.CtrlFunctionBlockType\\.MyType\\.SubObject\\.method3'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 321 - new_node Node(FourByteNodeId(ns=4;i=7003))'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 316 - current_node Node(FourByteNodeId(ns=4;i=58192)) chinld node Node(FourByteNodeId(ns=4;i=7001))'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 320 - new_node_path Root\\.Types\\.ObjectTypes\\.BaseObjectType\\.TopologyElementType\\.BlockType\\.CtrlProgramOrganizationUnitType\\.CtrlFunctionBlockType\\.MyType\\.SubObject\\.method1'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 321 - new_node Node(FourByteNodeId(ns=4;i=7001))'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 316 - current_node Node(FourByteNodeId(ns=4;i=58192)) chinld node Node(FourByteNodeId(ns=4;i=7006))'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 320 - new_node_path Root\\.Types\\.ObjectTypes\\.BaseObjectType\\.TopologyElementType\\.BlockType\\.CtrlProgramOrganizationUnitType\\.CtrlFunctionBlockType\\.MyType\\.SubObject\\.method6'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 321 - new_node Node(FourByteNodeId(ns=4;i=7006))'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 316 - current_node Node(FourByteNodeId(ns=4;i=58192)) chinld node Node(FourByteNodeId(ns=4;i=7007))'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 320 - new_node_path Root\\.Types\\.ObjectTypes\\.BaseObjectType\\.TopologyElementType\\.BlockType\\.CtrlProgramOrganizationUnitType\\.CtrlFunctionBlockType\\.MyType\\.SubObject\\.method7'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 321 - new_node Node(FourByteNodeId(ns=4;i=7007))'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 316 - current_node Node(FourByteNodeId(ns=4;i=58192)) chinld node Node(FourByteNodeId(ns=4;i=7004))'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 320 - new_node_path Root\\.Types\\.ObjectTypes\\.BaseObjectType\\.TopologyElementType\\.BlockType\\.CtrlProgramOrganizationUnitType\\.CtrlFunctionBlockType\\.MyType\\.SubObject\\.method4'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 321 - new_node Node(FourByteNodeId(ns=4;i=7004))'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 316 - current_node Node(FourByteNodeId(ns=4;i=58192)) chinld node Node(FourByteNodeId(ns=4;i=7005))'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 320 - new_node_path Root\\.Types\\.ObjectTypes\\.BaseObjectType\\.TopologyElementType\\.BlockType\\.CtrlProgramOrganizationUnitType\\.CtrlFunctionBlockType\\.MyType\\.SubObject\\.method5'
''2020-02-25 11:06:44' - DEBUG - opcua_connector - 321 - new_node Node(FourByteNodeId(ns=4;i=7005))'
```

So it looks like it founds the methods in the types folder (by nodeId?) and gives up without looking into the node that I specified. I thought if I specify the absolute path starting from Root will handle this but it seems not to be the case. Or have I done something wrong?

Thank you (and I will disappear for the next 2.5 weeks, but will be back then and looking forward to your answer :-) )

Cheers

zbeacon, 2020-03-02:
Hi @ki-do 

I have implemented ability to use OPC-UA nodes namespace and identifiers to search into develop branch.I will add it in the next release. But if you want to test this feature, you can install the gateway from the develop branch and just replace `"path": "${Root\\.Objects\\.Device}"` to something like `"path":"${ns=4;i=7001}"`. Also you can use different flags (i, s, g, b).

closed 2020-03-26 by zbeacon