thingsboard/thingsboard-gateway #242: development guide

agtokty, 2020-02-06:
Hi all, is there any development guide ?

zbeacon, 2020-02-06:
Hi @agtokty 

Unfortunatly, there is no available development guide, except [this guide](https://thingsboard.io/docs/iot-gateway/custom/) for custom connectors/converters development. I am going to create a development guide, but unfortunately, I don’t have time right now. Because we are going to implement some about 3-4 different connectors and i will have a time to write development documentation.

You can read the guide above and if you will have some questions, you can write us.

agtokty, 2020-02-06:
Hi @zbeacon , 

Thanks for reply, actually I followed that guide before and it is very useful, I also write(edit) serial connector and converter for my arduino project. 

But the .deb setup file is from older version(2.0) in [here](https://thingsboard.io/docs/iot-gateway/install/deb-installation/) . I just want to use and test the newer versions(pre-releases) of the gateway. So, I need to know how to build the project to generate .deb file.  Since I didn't know much about python development and creating setup before, I think it will be better to have development guide to make contributions. Btw, you can show me some links/tutorials about building python project or I try to find it myself :)

zbeacon, 2020-02-06:
@agtokty 

Please try to use a script "generate_deb_package.sh" in the root of the repository to genereta the debian package. You need to download sources to generate deb package. You can also run it like following, it will remove previous built:
``` bash generate_deb_package.sh clean```

Debian package will download the newest version of the gateway from pip during installation.

vimalgupta, 2020-02-07:
I have tried the above command but it has thrown below error.
Building DEB package
Creating sources for DEB package...
usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
   or: setup.py --help [cmd1 cmd2 ...]
   or: setup.py --help-commands
   or: setup.py cmd --help

error: invalid command 'bdist_deb'
Adding the files, scripts and permissions in the package
cp: cannot create directory 'deb_dist/thingsboard-gateway-2.1.0.1/debian/python3-thingsboard-gateway': No such file or directory
cp: cannot create directory 'deb_dist/thingsboard-gateway-2.1.0.1/debian/python3-thingsboard-gateway': No such file or directory
cp: cannot create directory 'deb_dist/thingsboard-gateway-2.1.0.1/debian/python3-thingsboard-gateway': No such file or directory
chown: cannot access 'deb_dist/thingsboard-gateway-2.1.0.1/debian/python3-thingsboard-gateway/': No such file or directory
chown: cannot access 'deb_dist/thingsboard-gateway-2.1.0.1/debian/python3-thingsboard-gateway/var/': No such file or directory
chmod: cannot access 'deb_dist/thingsboard-gateway-2.1.0.1/debian/python3-thingsboard-gateway/DEBIAN/preinst': No such file or directory
chown: cannot access 'deb_dist/thingsboard-gateway-2.1.0.1/debian/python3-thingsboard-gateway/DEBIAN/preinst': No such file or directory
generate_deb_package.sh: line 40: deb_dist/thingsboard-gateway-2.1.0.1/debian/python3-thingsboard-gateway/DEBIAN/control: No such file or directory
dpkg-deb: error: failed to open package info file 'deb_dist/thingsboard-gateway-2.1.0.1/debian/python3-thingsboard-gateway//DEBIAN/control' for reading: No such file or directory
cp: cannot stat 'deb_dist/thingsboard-gateway-2.1.0.1/debian/python3-thingsboard-gateway.deb': No such file or directory

zbeacon, 2020-02-10:
Hi @vimalgupta,

It's caused because something going wrong with debian package builder installation. It installs at start of script. Please share full output from script.

closed 2020-02-14 by zbeacon