thingsboard/thingsboard-gateway #243: Cannot connect to OPC server

komis1, 2020-02-06:
Hello,
Apologies for the noob question, I have setup an OPC server on a Rasberry Pi, and I'm trying to get the TB Gateway to connect to it, without success. The connector.log shows error
`''2020-02-06 13:22:50' - INFO - opcua_connector - 86 - Starting OPC-UA Connector'
''2020-02-06 13:22:50' - DEBUG - opcua_connector - 97 - error on connection to OPC-UA server.'
''2020-02-06 13:22:50' - ERROR - opcua_connector - 98 - 'Client' object has no attribute 'load_type_definitions''`

I have updated the opcua python package to the latest version (0.98.9). My TB installation runs on a Ubuntu VM, otherwise working without problems. I have setup the gateway as a device and configured the tb_gateway.yaml correctly.
On the Raspberry Pi, I am running the server_example.py of the python-opcua package from [https://github.com/FreeOpcUa/python-opcua/blob/master/examples/server-example.py](url). Using the ProSys OPC browser I can connect and retrieve the data successfully. Same applies when running the client_example.py code (from python-opcua). 
I have also tried running the Open62541 server, again I can connect successfully with the ProSys browser and the client_example.py script, but not with the gateway.

Any ideas what might be causing this? Thanks in advance!

zbeacon, 2020-02-06:
Hi @komis1 

Thanks for your interest in ThingsBoard IoT Gateway. Your problem looks really strange because I used the same server for development. Please share your configuration for opcua connector, I will try to repeat this situation. Also, please provide information about your Raspberry:
What system was used?
What version of Raspberry did you use?

komis1, 2020-02-06:
Hi @zbeacon,
Thank you for your reply. I know, I've been trying to get my head around it but can't find anything that would cause it. 
On the Linux VM (where tb is installed) I'm using Python 3.6.9.
On the Raspberry (RP4, 4Gb RAM), I'm using the latest Raspbian. Nothing else is running on the Pi. As I said, I don't think the problem is there, because I can see the OPCUA server just fine using a multitude of clients, both from the local machine, and from remote machines.

Here is the connector config for the gateway on the Linux VM
```{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "opc.tcp://<raspberry ip>:4840/freeopcua/server/",
    "scanPeriodInMillis": 10000,
    "timeoutInMillis": 5000,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Device1",
        "deviceNamePattern": "Device ${Root\\.Objects\\.Device1\\.serialNumber}",
        "attributes": [
          {
            "key": "temperature",
            "path": "${TemperatureAndHumiditySensor\\.Temperature}"
          }
        ],
        "timeseries": [
          {
            "key": "humidity",
            "path": "${Root\\.Objects\\.Device1\\.TemperatureAndHumiditySensor\\.Humidity}"
          },
          {
            "key": "batteryLevel",
            "path": "${Battery\\.batteryLevel}"
          }
        ],
        "rpc_methods": [
          {
            "method": "multiply",
            "arguments": [2, 4]
          }
        ],
        "attributes_updates": [
          {
            "attributeOnThingsBoard": "deviceName",
            "attributeOnDevice": "Root\\.Objects\\.Device1\\.serialNumber"
          }
        ]
      }
    ]
  }
}
```

And the general gateway config (yaml)
```
thingsboard:
  host: 127.0.0.1
  port: 1883
  remoteConfiguration: false
  security:
    accessToken: <my gateway access token>
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:

#  -
#    name: MQTT Broker Connector
#    type: mqtt
#    configuration: mqtt.json
#
#  -
#    name: Request Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
  -
    name: OPC-UA Connector
    type: opcua
    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector
```

zbeacon, 2020-02-07:
Hi @komis1 , 

Is the gateway also installed in VM?
I have installed the server-example.py in Raspberry Pi 4 (4GB). But i have used python3.7 on machine (real machine, not VM) with the gateway and it works properly, without connection error. 

Could you try to install the gateway in the real machine?
It can be some problem with network in VM.

komis1, 2020-02-07:
Hi,
Yes, the gateway is on the same vm as the Tb installation. Only the OPC
server is running on the raspberry pi.

I will try with python 3.7 and let you know...

komis1, 2020-02-07:
Ok I have installed Python3.7.6 on the UbuntuVM. Now, the gateway service fails completely to start.
```
● thingsboard-gateway.service - ThingsBoard Gateway
   Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-02-07 13:31:29 UTC; 22min ago
  Process: 29621 ExecStop=/bin/kill -INT $MAINPID (code=exited, status=0/SUCCESS)
  Process: 29946 ExecStart=/usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon() (code=exited, status=1/FAILURE)
 Main PID: 29946 (code=exited, status=1/FAILURE)

Feb 07 13:31:29 tb-i3t systemd[1]: thingsboard-gateway.service: Service hold-off time over, scheduling restart.
Feb 07 13:31:29 tb-i3t systemd[1]: thingsboard-gateway.service: Scheduled restart job, restart counter is at 5.
Feb 07 13:31:29 tb-i3t systemd[1]: Stopped ThingsBoard Gateway.
Feb 07 13:31:29 tb-i3t systemd[1]: thingsboard-gateway.service: Start request repeated too quickly.
Feb 07 13:31:29 tb-i3t systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
Feb 07 13:31:29 tb-i3t systemd[1]: Failed to start ThingsBoard Gateway.
```
When I try to run the client-example.py script using Python3.7.6 on the Ubuntu VM, it fails, although when executed with 3.6.9, it works.

Here it is, running with 3.6.9
```
host@tb-i3t:~$ python3.6 client-minimal.py 
urn:freeopcua:example
Objects node is:  Node(TwoByteNodeId(i=84))
Children of root are:  [Node(NumericNodeId(i=85)), Node(NumericNodeId(i=86)), Node(NumericNodeId(i=87))]
myvar is:  Node(NumericNodeId(ns=2;i=13))
myobj is:  Node(NumericNodeId(ns=2;i=12))
```
And here it is with 3.7.6
```
host@tb-i3t:~$ python3 client-minimal.py 
Traceback (most recent call last):
  File "client-minimal.py", line 5, in <module>
    from opcua import Client
  File "/usr/lib/python3/dist-packages/opcua/__init__.py", line 5, in <module>
    from opcua.common.node import Node
  File "/usr/lib/python3/dist-packages/opcua/common/node.py", line 6, in <module>
    from opcua import ua
  File "/usr/lib/python3/dist-packages/opcua/ua/__init__.py", line 6, in <module>
    from opcua.ua.uaprotocol_auto import *
  File "/usr/lib/python3/dist-packages/opcua/ua/uaprotocol_auto.py", line 8, in <module>
    from opcua.ua.uatypes import *
  File "/usr/lib/python3/dist-packages/opcua/ua/uatypes.py", line 21, in <module>
    from opcua.common.utils import Buffer
  File "/usr/lib/python3/dist-packages/opcua/common/utils.py", line 171
    task = asyncio.async(coro, loop=self.loop) 
 ```

Any ideas?

zbeacon, 2020-02-07:
You can see logs using following command:
```
sudo journalctl -xe
```
Please copy command output and share.

komis1, 2020-02-07:
Hi @zbeacon 

Here are the logs 

```
komninos@tb-i3t:~$ sudo systemctl restart thingsboard-gateway.service
komninos@tb-i3t:~$ sudo journalctl -xe
Feb 07 14:26:59 tb-i3t python3[2315]:     from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
Feb 07 14:26:59 tb-i3t python3[2315]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 27, in <module>
Feb 07 14:26:59 tb-i3t python3[2315]:     from thingsboard_gateway.gateway.tb_client import TBClient
Feb 07 14:26:59 tb-i3t python3[2315]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_client.py", line 18, in <module>
Feb 07 14:26:59 tb-i3t python3[2315]:     from thingsboard_gateway.tb_client.tb_gateway_mqtt import TBGatewayMqttClient
Feb 07 14:26:59 tb-i3t python3[2315]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_client/tb_gateway_mqtt.py", line 19, in <module>
Feb 07 14:26:59 tb-i3t python3[2315]:     from thingsboard_gateway.tb_client.tb_device_mqtt import TBDeviceMqttClient, DEVICE_TS_KV_VALIDATOR, KV_VALIDATOR
Feb 07 14:26:59 tb-i3t python3[2315]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_client/tb_device_mqtt.py", line 25, in <module>
Feb 07 14:26:59 tb-i3t python3[2315]:     from jsonschema import Draft7Validator
Feb 07 14:26:59 tb-i3t python3[2315]: ModuleNotFoundError: No module named 'jsonschema'
Feb 07 14:26:59 tb-i3t systemd[1]: thingsboard-gateway.service: Main process exited, code=exited, status=1/FAILURE
Feb 07 14:26:59 tb-i3t systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
Feb 07 14:26:59 tb-i3t systemd[1]: thingsboard-gateway.service: Service hold-off time over, scheduling restart.
Feb 07 14:26:59 tb-i3t systemd[1]: thingsboard-gateway.service: Scheduled restart job, restart counter is at 5.
-- Subject: Automatic restarting of a unit has been scheduled
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Automatic restarting of the unit thingsboard-gateway.service has been scheduled, as the result for
-- the configured Restart= setting for the unit.
Feb 07 14:26:59 tb-i3t systemd[1]: Stopped ThingsBoard Gateway.
-- Subject: Unit thingsboard-gateway.service has finished shutting down
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Unit thingsboard-gateway.service has finished shutting down.
Feb 07 14:26:59 tb-i3t systemd[1]: thingsboard-gateway.service: Start request repeated too quickly.
Feb 07 14:26:59 tb-i3t systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
Feb 07 14:26:59 tb-i3t systemd[1]: Failed to start ThingsBoard Gateway.
-- Subject: Unit thingsboard-gateway.service has failed
-- Defined-By: systemd
-- Support: http://www.ubuntu.com/support
-- 
-- Unit thingsboard-gateway.service has failed.
-- 
-- The result is RESULT.
```

komis1, 2020-02-07:
Funnily enough, the module it's complaining about is installed...

```
komninos@tb-i3t:~$ pip3 show jsonschema
Name: jsonschema
Version: 2.6.0
Summary: An implementation of JSON Schema validation for Python
Home-page: http://github.com/Julian/jsonschema
Author: Julian Berman
Author-email: Julian@GrayVines.com
License: MIT
Location: /usr/lib/python3/dist-packages
Requires: 
```

zbeacon, 2020-02-07:
@komis1 

Unfortunately, no. It isn't installed, pip3 related to python 3.6.9. Try to use pip3.7 show jsonschema. It is different package managers. Could you try to install the gateway in the real machine for test or using docker?

komis1, 2020-02-10:
hi @zbeacon 
Ok I got it working (sort of). Installed on a physical machine, same problems as the VM. Digging around, I figured it was a problem with parallel installations of Python3 (3.6 and 3.7), all the necessary packages were installed in the Python3.7 folders but the service file (/etc/systemd/system/thingsboard-gateway.service) was pointing to the 3.6 Python executable. Anyway I changed that to point to /usr/bin/python3.7 and now it's running.

However, I still have problems. It's finding and reading the attributes and telemetry data from the OPC server, but the converter is not passing them into the attributes or telemetry sections to send to ThingsBoard. Also, the attribute update part of the code is throwing an exception... Please see attached log. Any ideas what's wrong?

```
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:16' - INFO - opcua_connector - 86 - Starting OPC-UA Connector'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:16' - DEBUG - tb_gateway_service - 459 - Loaded devices:
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]:  {}'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:16' - DEBUG - tb_gateway_service - 265 - Send data Thread has been started successfully.'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:16' - DEBUG - opcua_connector - 93 - urn:freeopcua:example'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:16' - DEBUG - opcua_connector - 94 - 3'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:16' - INFO - opcua_connector - 104 - OPC-UA connector OPC-UA Connector connected to server 192.168.1.123:4840/freeopcua/server/'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:16' - DEBUG - opcua_connector - 173 - evaluating {'deviceNodePattern': 'Root\\.Objects\\.Device0001', 'deviceNamePattern': 'Device${Root\\.Objects\\.Device0001\\.device_id}', 'att
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:16' - DEBUG - opcua_connector - 306 - Search in Root\\.Objects'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:17' - DEBUG - opcua_connector - 298 - Found in node: Root\\.Objects\\.Device0001'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:17' - DEBUG - opcua_connector - 306 - Search in Root\\.Objects'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:17' - DEBUG - opcua_connector - 306 - Search in Root\\.Objects\\.Device0001'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:17' - DEBUG - opcua_connector - 298 - Found in node: Root\\.Objects\\.Device0001\\.device_id'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:17' - DEBUG - opcua_connector - 265 - Device name: Device0340'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:17' - DEBUG - opcua_connector - 306 - Search in Root\\.Objects\\.Device0001\\.controller'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:17' - DEBUG - opcua_connector - 298 - Found in node: Root\\.Objects\\.Device0001\\.controller\\.state'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:17' - DEBUG - opcua_connector - 205 - Node for attributes "State" with path: Root\.Objects\.Device0001\.controller\.state - FOUND! Current values is: Running'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:17' - DEBUG - opcua_connector - 222 - Data to ThingsBoard: {'deviceName': 'Device0340', 'deviceType': 'default', 'attributes': [], 'telemetry': []}'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:17' - DEBUG - opcua_connector - 298 - Found in node: Root\\.Objects\\.Device0001\\.sensor1'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:17' - DEBUG - opcua_connector - 205 - Node for timeseries "humidity" with path: Root\.Objects\.Device0001\.sensor1 - FOUND! Current values is: 1.0'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:17' - DEBUG - opcua_connector - 222 - Data to ThingsBoard: {'deviceName': 'Device0340', 'deviceType': 'default', 'attributes': [], 'telemetry': []}'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:17' - DEBUG - opcua_connector - 306 - Search in Root\\.Objects\\.Device0001\\.controller'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:17' - DEBUG - opcua_connector - 298 - Found in node: Root\\.Objects\\.Device0001\\.controller\\.state'
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: ''2020-02-10 15:04:17' - ERROR - opcua_connector - 251 - 'Device0340''
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: Traceback (most recent call last):
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 245, in __search_attribute_update_variables
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]:     if self.__available_object_resources[device_name].get("variables") is None:
Φεβ 10 15:04:17 administrator-OptiPlex-390 python3.7[11481]: KeyError: 'Device0340'
[... cut ...]
        "rpc_methods": [
          {
            "method": "multiply",
            "arguments": [2, 4]
          }
        ],
        "attributes_updates": [
          {
            "attributeOnThingsBoard": "State",
            "attributeOnDevice": "controller\\.state"
          }
        ]
      }
    ]
  }
}
```
and here is a screenshot of the OPCUA server config running on the Raspberry Pi (it's the server-example.py offered by the python-opcua library)

Thanks again for all your help!!!!
[image]

zbeacon, 2020-02-10:
Thank you for your interest in ThingsBoard IoT gateway one more time :)
I already know about this issue, please install 2.1.0.3 version of the gateway and try again.

To install it, you can use pip:
```
pip3.7 install thingsboard-gateway==2.1.0.3
```

komis1, 2020-02-10:
@zbeacon thank you so much for your hard work.
I'm afraid the problem persists. A new error appears in the log:
```
''2020-02-10 15:28:40' - ERROR - opcua_connector - 256 - 'dict' object has no attribute 'append''
```

zbeacon, 2020-02-10:
@komis1 

It's a different problem, ok, please install 2.1.0.4 version.

komis1, 2020-02-10:
@zbeacon I will try it in a moment. In the meantime, I found the bug, it was in the opcua_uplink_converter file. The regex fullmatch was returning always None. I changed this to a simple comparison (==) and now it's working perfectly.
Here some sample log output from when the error was happening, I used a simple statement to check what was being matched (pattern, string)

```
DEBUG - opcua_uplink_converter - 40 - no fullmatch for Root\.Objects\.Device0001\.controller\.state, Root\.Objects\.Device0001\.controller\.state'
```

zbeacon, 2020-02-10:
Ok, thanks, i will add 2 variants, with strict comparing and with fullmatch regex.

komis1, 2020-02-10:
@zbeacon happy to have helped! Let me know when you make another release and I can test it. I will leave my hacky code in place for now, as I want to try more things :)

komis1, 2020-02-10:
@zbeacon just another thing to say, I am noticing that after successful connection, the gateway reverts to an inactive state after a few seconds, and then doesn't seem to be polling the OPCUA server any more (active=false, lastconnecttime and lastdisconnecttime are always the same).
Time to go home now, I will look into this tomorrow.

zbeacon, 2020-02-10:
Your code is already in master branch (commit 22af314 ). If you want to develop something - you can create connectors, any you want/need and create a pull request. Currently I am going to create documentation about a Request connector and some improvements for already existing documents. New features will appear in about few weeks. 
The gateway (and any different device on ThingsBoard) goes to inactive state after 10 seconds (By default) after the last telemetry from device has been received. To keep the gateway in active state you should send/read data from device every 10 seconds. 
Or you can solve this using little hack - add shared attribute **RemoteLoggingLevel** and set it value to **DEBUG**. You can receive logs from the gateway to the your ThingsBoard instance ([Read more](https://thingsboard.io/docs/iot-gateway/guides/how-to-enable-remote-logging/)) and this telemetry will keep the gateway active.

zbeacon, 2020-02-10:
>> ...and then doesn't seem to be polling the OPCUA server any more (active=false, lastconnecttime and lastdisconnecttime are always the same)...

The gateway isn't polling the server, it subscribes to updates and send data at first time and only when it will be changed.

ki-do, 2020-02-12:
i'm afraid, still no luck 

>Feb 12 11:16:35 iot-pi python3[3172]: ''2020-02-12 11:16:35' - INFO - opcua_connector - 87 - Starting OPC-UA Connector'                                                                                                                                                                   >Feb 12 11:16:35 iot-pi python3[3172]: ''2020-02-12 11:16:35' - DEBUG - tb_gateway_service - 459 - Loaded devices:                                                                                                                                                                         >Feb 12 11:16:35 iot-pi python3[3172]:  {}'                                                                                                                                                                                                                                                >Feb 12 11:16:35 iot-pi python3[3172]: ''2020-02-12 11:16:35' - DEBUG - tb_gateway_service - 265 - Send data Thread has been started successfully.'                                                                                                                                        >Feb 12 11:16:35 iot-pi python3[3172]: ''2020-02-12 11:16:35' - DEBUG - opcua_connector - 94 - urn:freeopcua:example'                                                                                                                                                                      >Feb 12 11:16:35 iot-pi python3[3172]: ''2020-02-12 11:16:35' - DEBUG - opcua_connector - 95 - 3'                                                                                                                                                                                          >Feb 12 11:16:35 iot-pi python3[3172]: ''2020-02-12 11:16:35' - INFO - opcua_connector - 105 - OPC-UA connector OPC-UA Connector connected to server localhost:4840/freeopcua/server/'                                                                                                     >Feb 12 11:16:35 iot-pi python3[3172]: ''2020-02-12 11:16:35' - DEBUG - opcua_connector - 326 - Search in Root\\.Objects'                                                                                                                                                                  >Feb 12 11:16:36 iot-pi python3[3172]: ''2020-02-12 11:16:36' - ERROR - opcua_connector - 304 - Device name node not found with expression: multipl.'                                                                                                                                      >Feb 12 11:16:36 iot-pi python3[3172]: ''2020-02-12 11:16:36' - ERROR - opcua_connector - 190 - Device node is None, please check your configuration.'                                                                                                                                     >Feb 12 11:16:36 iot-pi python3[3172]: ''2020-02-12 11:16:36' - DEBUG - opcua_connector - 191 - Current device node is: Root\.Objects\.MyObject'                                                                                                                                           >Feb 12 11:16:36 iot-pi python3[3172]: ''2020-02-12 11:16:36' - DEBUG - opcua_connector - 195 - [{'Root\\.Objects\\.MyObject': {'deviceNodePattern': 'Root\\.Objects\\.MyObject', 'deviceNamePattern': 'multipl.', 'attributes': [{'key': 'State', 'path': '${Root\\.Objects\\.Device0001\ Feb 12 11:16:36 iot-pi python3[3172]: ''2020-02-12 11:16:36' - DEBUG - opcua_connector - 110 - Subscriptions: {}'                                                                                                                                                                         >Feb 12 11:16:36 iot-pi python3[3172]: ''2020-02-12 11:16:36' - DEBUG - opcua_connector - 111 - Available methods: {}'

komis1, 2020-02-12:
hi @zbeacon 
Coming back to this one more time. I installed 2.1.0.4, and applied my hack (== instead of fullmatch) to make it work. When starting the service for the first time, it reads all values, updates the generated device, and then the gateway "active" attribute goes to "false".
Next, I tried the ** RemoteLoggingLevel** tip as you mentioned, but it doesn't seem to keep the "active" attribute to "true". I'm not receiving any more debug output except the first message (
I wrote a little python client which randomly generates values for sensor1 (in the opcua server-example.py) every 2 seconds. This is working fine, as verified by the Prosys Browser app. However, still, the gateway remains at an "active=false" state, and doesn't update the timeseries with the new values.
The only thing that seems to change is the "lastActivityTime" server attribute of the gateway, this seems to reflect a time value change every minute.
I don't know if I am missing something here... any help is appreciated.

ki-do, 2020-02-12:
quickly comparing freeopcua examples with opcua?connector code, i assume, subscribe_data_change call is missing. Here is what they do in freeopcua example:

```
# subscribing to a variable node
        handler = SubHandler()
        sub = client.create_subscription(500, handler)
        handle = sub.subscribe_data_change(myvar)
```

komis1, 2020-02-12:
@ki-do Indeed, I just arrived at the same conclusion!!! Let's see if we can apply a fix to this.

komis1, 2020-02-12:
@ki-do @zbeacon 
Here is a fix, applied to opcua_connector.py (line 109)
```
        log.debug('Subscriptions: %s', self.subscribed)
        log.debug("Available methods: %s", self.__available_object_resources)
        for key in self.subscribed.keys():        
            log.debug("NODE %s",key)
            self.__sub.subscribe_data_change(key)
```

[image]

komis1, 2020-02-12:
Here is the problem with the attributes not updating, from what I gather.

In the initial scan of nodes, a converter is created. First, the converter is used for attributes. Then the converter is used for telemetry. 
When the converter is called again (after a data update), the converter is still configured to use the "timeseries" type, and doesn't understand how to do the conversion.

In the log below, the 2nd line output should read "attributes" but instead it's still thinking it should use "timeseries" because this was the last type converted.

@zbeacon my knowledge of Python is rather limited and I don't want to make a stupid fix - can you look into this please?

```
''2020-02-12 17:14:27' - DEBUG - opcua_connector - 345 - Python: New data change event on node Node(NumericNodeId(ns=2;i=11)), with val: skata'
''2020-02-12 17:14:27' - DEBUG - opcua_uplink_converter - 35 - Converting information_type timeseries'
''2020-02-12 17:14:27' - DEBUG - opcua_connector - 351 - Data to ThingsBoard: {'deviceName': 'Device0340', 'deviceType': 'default', 'attributes': [], 'telemetry': []}'
```

zbeacon, 2020-02-13:
@komis1 Thanks for your investigations, it looks like we have a problem with relative path to variables in timeseries and attributes section, i will fix this. Currently, please try to use the full path in the config to timeseries and attributes variables.

zbeacon, 2020-02-13:
@komis1, @ki-do 

Please try to install the gateway from sources (develop or master branch) and test relative path to variables in configurations.

komis1, 2020-02-13:
hi @zbeacon 
I just tried the latest pre-release (2.1.0.5). Thank you for fixing the subscriptions bug, I see where you have made the replacement in the code. The timeseries values are now updated regularly as should be.

The attributes are still not changing. The change event is picked up, but the uplink converter still only tries to update telemetry data - of course, the path is different, so no change is effected. I believe this is because the uplink converter is set to work with telemetry data after the initial load (since this is the last bit of information that is picked up by it). Therefore I think the fix should be something like an additional parameter in the uplink converter to say that we want to work with attribute or telemetry data types.

More specifically, in the uplink converter code, in line 33
```
for information in self.__config[information_type]:
```
this loop executes only once - the value of self.__config[information_type] is only "telemetry".

zbeacon, 2020-02-14:
@komis1 

Thank you so mus for your investigations and your edits, it was a bug.
I have add this code to the master branch. Also, I refactored converter a little bit to correct processing relative and absolute path of nodes.

komis1, 2020-02-14:
@zbeacon Glad to have helped :)

zbeacon, 2020-02-14:
I close this one issue. But if you have any other issues or suggestions for improvement, please open a new issue for them.

closed 2020-02-14 by zbeacon