thingsboard/thingsboard-gateway #244: Cannot connect to OPC UA server

Ming-Shu, 2020-02-15:
Hello,
I have setup an OPC UA server on Rasberry Pi 4, and I'm trying to connect to it ,but log show below:

**''2020-02-15 15:52:24' - DEBUG - tb_utility - 81 - Import OpcUaConnector from /usr/lib/python3/dist-packages/thingsboard_gateway/connectors/opcua.'
''2020-02-15 15:52:24' - INFO - opcua_connector - 87 - Starting OPC-UA Connector'
''2020-02-15 15:52:24' - DEBUG - tb_gateway_service - 489 - Loaded devices:
 {}'
''2020-02-15 15:52:24' - DEBUG - tb_gateway_service - 265 - Send data Thread has been started successfully.'
''2020-02-15 15:52:25' - DEBUG - opcua_connector - 100 - error on connection to OPC-UA server.'
''2020-02-15 15:52:25' - ERROR - opcua_connector - 101 - 'Client' object has no attribute 'load_type_definitions''
''2020-02-15 15:52:35' - DEBUG - opcua_connector - 100 - error on connection to OPC-UA server.'
''2020-02-15 15:52:35' - ERROR - opcua_connector - 101 - 'Client' object has no attribute 'load_type_definitions''**

What might be causing this? Thanks in advance!!!

My machine is real machine ,
**evironment* below*
OS-Ubuntu 18.04;
Thingsboard IoT Gateway version 2.1.06 ( latest )

My tb_gateway.yaml  is here
**
thingsboard:
  host: demo.thingsboard.io
  port: 1883
  remoteConfiguration: false
  security:
    accessToken: <My gateway access token>
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:
#  -
#    name: MQTT Broker Connector
#    type: mqtt
#    configuration: mqtt.json
#
#  -
#    name: Request Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
  -
    name: OPC-UA Connector
    type: opcua
    configuration: opcua.json

#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector
**
My opcua.json  is here
**
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "172.16.61.212:49320",
    "scanPeriodInMillis": 10000,
    "timeoutInMillis": 5000,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Device1",
        "deviceNamePattern": "Device ${Root\\.Objects\\.Device1\\.serialNumber}",
        "attributes": [
          {
            "key": "temperature",
            "path": "${TemperatureAndHumiditySensor\\.Temperature}"
          }
        ],
        "timeseries": [
          {
            "key": "humidity",
            "path": "${Root\\.Objects\\.Device1\\.TemperatureAndHumiditySensor\\.Humidity}"
          },
          {
            "key": "batteryLevel",
            "path": "${Battery\\.batteryLevel}"
          }
        ],
        "rpc_methods": [
          {
            "method": "multiply",
            "arguments": [2, 4]
          }
        ],
        "attributes_updates": [
          {
            "attributeOnThingsBoard": "deviceName",
            "attributeOnDevice": "Root\\.Objects\\.Device1\\.serialNumber"
          }
        ]
      }
    ]
  }
}
**

Ming-Shu, 2020-02-16:
Thank you for your help,I will try it. : )

zbeacon, 2020-02-17:
Hi @Ming-Shu,

Thank you for your interest in the ThingsBoard IoT gateway, please update to the version from the master branch and check gateway behavior. To install the gateway from the sources please follow [this guide](https://thingsboard.io/docs/iot-gateway/install/source-installation/). 

**Notate:** Remove previous version before install using following commands:
`sudo pip3 uninstall thingsboard-gateway -y` and `pip3 uninstall thingsboard-gateway -y`

Ming-Shu, 2020-02-17:
Thank you for all your reply again,I try it if any issue I will ask for you : )

closed 2020-02-18 by Ming-Shu