thingsboard/thingsboard-gateway #245: [BUG]

jcstarna, 2020-02-16:
**Description**
I'm taking a look to the this gateway and thingboard server.
I'm getting errors when polling modbus (TCP). 
1- It uses wrong function code when polling. It sends 4 (Read Input Register) when it must send 3 (Read Holding Register). I've checked that using wireshark
2 - It logs some error about response.
3 - It doesn't publish in MQTT (I guess because 1 and 2)

**Connector name:**
Modbus Connector

**Error traceback (If available):**
conector log show this records (repeated)
''2020-02-16 18:52:33' - INFO - modbus_connector - 59 - Modbus connected.'
''2020-02-16 18:52:34' - DEBUG - modbus_connector - 198 - To modbus device Medidor Potencia, 
{'byteOrder': 'LITTLE', 'tag': 'valCount', 'type': 'long', 'functionCode': 3, 'registerCount': 2, 'address': 2, 'deviceName': 'Medidor Potencia'}'
''2020-02-16 18:52:34' - DEBUG - modbus_connector - 199 - With result ReadRegisterResponse (2)'
''2020-02-16 18:52:34' - ERROR - modbus_connector - 154 - 'ReadInputRegistersResponse' object has no attribute 'isError''
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 116, in __process_devices
    if input_data.isError():
AttributeError: 'ReadInputRegistersResponse' object has no attribute 'isError'
```

**Versions (please complete the following information):**
 - OS: Ubuntu 18.04.4 LTS
 - Thingsboard IoT Gateway version 2.0-1
 - Python version 2.7.17

zbeacon, 2020-02-17:
Hi @jcstarna,

Thank you for interest in the ThingsBoard IoT gateway, please try to use the newer version of the gateway. You can update it using following commands:

if you run the gateway as a daemon
```
sudo pip3 uninstall thingsboard-gateway -y
sudo pip3 install thingsboard-gateway -y
```
or
```
pip3 uninstall thingsboard-gateway -y
pip3 install thingsboard-gateway -y --user
```

Please update your version and check its behavior.

jcstarna, 2020-02-17:
I've got errors on terminal when I tried those commands. I've reinstalled using the package from 
wget https://github.com/thingsboard/thingsboard-gateway/releases/download/2.1.0.6/python3-thingsboard-gateway.deb
(I changed 2.0 release number for 2.1.0.6)

**Now I've got this erros on logfile:**
''2020-02-17 10:27:34' - DEBUG - modbus_connector - 216 - To modbus device Mock de Esclavo, 
{'byteOrder': 'BIG', 'tag': 'testTS1', 'type': 'long', 'functionCode': 3, 'registerCount': 2, 'address': 2, 'deviceName': 'Mock de Esclavo'}'
''2020-02-17 10:27:34' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-02-17 10:27:34' - DEBUG - modbus_connector - 216 - To modbus device Mock de Esclavo, 
{'byteOrder': 'LITTLE', 'tag': 'testTS2', 'type': 'long', 'functionCode': 3, 'registerCount': 2, 'address': 4, 'deviceName': 'Mock de Esclavo'}'
''2020-02-17 10:27:34' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-02-17 10:27:34' - DEBUG - modbus_connector - 125 - Checking timeseries for device Mock de Esclavo'
''2020-02-17 10:27:34' - DEBUG - modbus_connector - 127 - {'timeseries': {'testTS1': {'data_sent': {'byteOrder': 'BIG', 'tag': 'testTS1', 'type': 'long', 'functionCode': 3, 'registerCount': 2, 'address': 2, 'deviceName': 'Mock de Esclavo'}, 'input_data': <pymodbus.register_read_message.ReadHoldingRegistersResponse object at 0x7fb7f17186a0>}, 'testTS2': {'data_sent': {'byteOrder': 'LITTLE', 'tag': 'testTS2', 'type': 'long', 'functionCode': 3, 'registerCount': 2, 'address': 4, 'deviceName': 'Mock de Esclavo'}, 'input_data': <pymodbus.register_read_message.ReadHoldingRegistersResponse object at 0x7fb7f17187f0>}}, 'attributes': {}}'
''2020-02-17 10:27:34' - DEBUG - bytes_modbus_uplink_converter - 33 - testTS1'
''2020-02-17 10:27:34' - DEBUG - bytes_modbus_uplink_converter - 36 - Called convert function from BytesModbusUplinkConverter with args'
''2020-02-17 10:27:34' - DEBUG - bytes_modbus_uplink_converter - 37 - {'byteOrder': 'BIG', 'tag': 'testTS1', 'type': 'long', 'functionCode': 3, 'registerCount': 2, 'address': 2, 'deviceName': 'Mock de Esclavo'}'
''2020-02-17 10:27:34' - DEBUG - bytes_modbus_uplink_converter - 38 - ReadRegisterResponse (2)'
''2020-02-17 10:27:34' - ERROR - bytes_modbus_uplink_converter - 61 - fromRegisters() got an unexpected keyword argument 'byteorder''
''2020-02-17 10:27:34' - ERROR - bytes_modbus_uplink_converter - 77 - local variable 'decoder' referenced before assignment'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 70, in convert
    result = decoder.decode_32bit_int()
UnboundLocalError: local variable 'decoder' referenced before assignment
''2020-02-17 10:27:34' - ERROR - modbus_connector - 172 - int() argument must be a string, a bytes-like object or a number, not 'list''
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 128, in __process_devices
    converted_data = self.__devices[device]["converter"].convert(config=None, data=device_responses)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 110, in convert
    elif int(result):
TypeError: int() argument must be a string, a bytes-like object or a number, not 'list'
''2020-02-17 10:27:35' - DEBUG - modbus_connector - 216 - To modbus device Mock de Esclavo, 
{'byteOrder': 'BIG', 'tag': 'testAtt', 'type': 'long', 'functionCode': 3, 'registerCount': 2, 'address': 0, 'deviceName': 'Mock de Esclavo'}'
''2020-02-17 10:27:35' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-02-17 10:27:35' - DEBUG - modbus_connector - 125 - Checking attributes for device Mock de Esclavo'
''2020-02-17 10:27:35' - DEBUG - modbus_connector - 127 - {'timeseries': {}, 'attributes': {'testAtt': {'data_sent': {'byteOrder': 'BIG', 'tag': 'testAtt', 'type': 'long', 'functionCode': 3, 'registerCount': 2, 'address': 0, 'deviceName': 'Mock de Esclavo'}, 'input_data': <pymodbus.register_read_message.ReadHoldingRegistersResponse object at 0x7fb7f1718f60>}}}'
''2020-02-17 10:27:35' - DEBUG - bytes_modbus_uplink_converter - 33 - testAtt'
''2020-02-17 10:27:35' - DEBUG - bytes_modbus_uplink_converter - 36 - Called convert function from BytesModbusUplinkConverter with args'
''2020-02-17 10:27:35' - DEBUG - bytes_modbus_uplink_converter - 37 - {'byteOrder': 'BIG', 'tag': 'testAtt', 'type': 'long', 'functionCode': 3, 'registerCount': 2, 'address': 0, 'deviceName': 'Mock de Esclavo'}'
''2020-02-17 10:27:35' - DEBUG - bytes_modbus_uplink_converter - 38 - ReadRegisterResponse (2)'
''2020-02-17 10:27:35' - ERROR - bytes_modbus_uplink_converter - 61 - fromRegisters() got an unexpected keyword argument 'byteorder''
''2020-02-17 10:27:35' - ERROR - bytes_modbus_uplink_converter - 77 - local variable 'decoder' referenced before assignment'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 70, in convert
    result = decoder.decode_32bit_int()
UnboundLocalError: local variable 'decoder' referenced before assignment
''2020-02-17 10:27:35' - ERROR - modbus_connector - 172 - int() argument must be a string, a bytes-like object or a number, not 'list''
Traceback (most recent call last):
[... cut ...]
        "rpc": {
          "turnLightOn": {
            "address": 4,
            "bit": 2,
            "value": true
          },
          "turnLightOff": {
            "address": 4,
            "bit": 2,
            "value": false
          },
          "getCPULoad": {
            "tag": "Integer",
            "value": 42,
            "functionCode": 16,
            "address": 0,
            "unitId": 1,
            "byteOrder": "BIG",
            "registerCount": 1
          }
        }
      }
    ]
  }
}

jcstarna, 2020-02-17:
Never mind. I upgraded it with **sudo pip3 install thingsboard-gateway --upgrade** and it works

closed 2020-02-17 by jcstarna