thingsboard/thingsboard-gateway #246: Connect to  OPC UA  Question

Ming-Shu, 2020-02-18:
Hello,
I'm trying to IoT Gateway installation from sources successful,and I have also commanded "python3 ./thingsboard_gateway/tb_gateway.py " is OK and show below:
**""2020-02-18 11:19:19" - INFO - [opcua_connector.py] - opcua_connector - 87 - Starting OPC-UA Connector"
""2020-02-18 11:19:19" - INFO - [opcua_connector.py] - opcua_connector - 109 - OPC-UA connector OPC-UA Connector connected to server localhost:49320"**

But I try to command "systemctl status thingsboard-gateway" is show below:

 thingsboard-gateway.service - ThingsBoard Gateway
   Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2020-02-18 11:08:30 CST; 59s ago
  Process: 5848 ExecStart=/usr/bin/python3 -c from thingsboard_gateway.tb_gateway import **daemon; daemon() (code=exited, status=1/FAIL
 Main PID: 5848 (code=exited, status=1/FAILURE)**

 二  18 11:08:30 orisol-Default-string systemd[1]: thingsboard-gateway.service: Service hold-off time over, scheduling restart.
 二  18 11:08:30 orisol-Default-string systemd[1]: thingsboard-gateway.service: Scheduled restart job, restart counter is at 5.
 二  18 11:08:30 orisol-Default-string systemd[1]: Stopped ThingsBoard Gateway.
 二  18 11:08:30 orisol-Default-string systemd[1]: thingsboard-gateway.service: Start request repeated too quickly.
 二  **18 11:08:30 orisol-Default-string systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.**
 二  18 11:08:30 orisol-Default-string systemd[1]: Failed to start ThingsBoard Gateway.

I don't know why show FAIL,and I can't sync data of  information  about OPC UA server in Thinkboard web.
  
Any ideas what might be causing this? Thanks in advance!

zbeacon, 2020-02-18:
Hi @Ming-Shu,

Unfortunately, i cannot explain this behavior without any output from the gateway.
Please share your system log, you can use following command to get the system log:
`sudo journalctl -xe`

Also you can open "service.log" in the gateway logs folder and looking for some errors there.

zbeacon, 2020-02-18:
Also please keep in mind that installation from sources and daemon installation are different installations. And when you ran the gateway using python3 command - you have used the configuration from the sources folder, instead this, when you ran the daemon, configs from /etc/thingsboard-gateway/config have been used. To upgrade the version of the gateway and run it as a daemon, please just use following command to install (from the sources directory)
`sudo python3 setup.py install` and restart the daemon (`sudo systemctl restart thingsboard-gateway`)

Ming-Shu, 2020-02-18:
Hi @zbeacon 

Apologies for the noob quesion, I'm trying  installation from sources and  restart the daemon (sudo systemctl restart thingsboard-gateway),but it still status fail.

Here are the system logs

[system_log.txt](https://github.com/thingsboard/thingsboard-gateway/files/4217964/system_log.txt)

My servive.log only show below :
""2020-02-18 14:27:27" - INFO - [tb_gateway_service.py] - tb_gateway_service - 125 - Stopping..."
""2020-02-18 14:27:27" - INFO - [tb_gateway_service.py] - tb_gateway_service - 127 - The gateway has been stopped."
""2020-02-18 15:24:10" - INFO - [tb_gateway_service.py] - tb_gateway_service - 125 - Stopping..."
""2020-02-18 15:24:10" - INFO - [tb_gateway_service.py] - tb_gateway_service - 127 - The gateway has been stopped."
""2020-02-18 15:31:37" - INFO - [tb_gateway_service.py] - tb_gateway_service - 125 - Stopping..."
""2020-02-18 15:31:37" - INFO - [tb_gateway_service.py] - tb_gateway_service - 127 - The gateway has been stopped."

zbeacon, 2020-02-18:
Hmm, it tries to start but something has gone wrong, please try to create folder with name "**logs**" and run the gateway using python3 and path from the directory with "**logs**" folder. By default if you ran the gateway from sources you need to create the logs folder manually. 

If previous step wasn't help - please try to increase the logging level (You can do it by change the level from **INFO** to **DEBUG** in "**logs.conf**" in configs directory. Please change it and post here the output from the gateway.

Ming-Shu, 2020-02-18:
OK, I'm trying again.
Glad to have helped : )

zbeacon, 2020-02-18:
@Ming-Shu 

Thank you, please give feedback after trying :).

Ming-Shu, 2020-02-19:
Hi @zbeacon
 
I succeed run BT gateway. Actually,I do reinstall my OS and  all Thingsboard step.
step below:
1. reinstall OS
2. install Thingsboard& BT gateway(2.1.0.6-1)
3. run "sudo pip3 install thingsboard-gateway --upgrade" and  installation from sources.
4. Edit yaml and json 
5. run sudo python3 -c 'from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService; TBGatewayService("tb_gateway.yaml")'

But I still question,I don't know write Mapping on Thingsboard web,
e.g.

 My opcua.json :
"deviceNodePattern": "Root\\.Objects\\.Folder",
        "deviceNamePattern": "Device",
        "attributes": [
          {
            "key": "T001",
            "path": "${T001}"
          }
        ]

Thinksboard below: 
deviceNodePattern : Channel1\.Device\d+$ --> Root\.Objects\.Folder
deviceNamePattern: Device ${_System._DeviceId} --> "Device"
Attributes: key:Tag1-->T001
Attributes: Type:String-->String
Attributes: Value:${Tag1}-->${T001}
but Thinksboard Extensions show **Not sync**

Any ideas about it ,Thank you!

zbeacon, 2020-02-19:
Hi @Ming-Shu ,

It's normal, the extension tab had been developed for old gateway, written on Java. We develop the remote configuration for the new gateway. It will be available in one of the next releases of ThingsBoard.

Ming-Shu, 2020-02-19:
OK, I  understand.
I was discovering  the new problem, if  running thingsboard and BT gateway,and interrupt OPCUA server then restart OPCUA server,  thingsboard and BT gateway(_run "sudo python3 -c 'from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService; TBGatewayService("tb_gateway.yaml")' "_) that will stop update value .But I restart BT gateway ,it will back to normal status,and show new value. Is it normal or bug?

zbeacon, 2020-02-24:
Hi @Ming-Shu,

It was a bug, a have fixed it in the new pre-release (2.1.0.8), please try to install it and check this behavior.

Ming-Shu, 2020-02-24:
@zbeacon 

OK, I'm glad to receive your reply, and I  will try it. : )

Ming-Shu, 2020-02-25:
@zbeacon 

I'm trying new pre-release (2.1.0.8),but show 
**ERROR - [tb_gateway_service.py] - tb_gateway_service - 341 - 'NoneType' object has no attribute 'in_process'"**

my log  below:

[service.log](https://github.com/thingsboard/thingsboard-gateway/files/4247881/service.log)

[tb_connection.log](https://github.com/thingsboard/thingsboard-gateway/files/4247883/tb_connection.log)

Any ideas what might be causing this? Thanks !

zbeacon, 2020-02-25:
@Ming-Shu 

Got it, i fixed it in 2.1.0.9, please try this version :)

Ming-Shu, 2020-02-25:
Good! I'm trying it.

Ming-Shu, 2020-02-25:
@zbeacon 

I'm running it is OK! 
But still have a problem,I saw fixed accumulation value of LATEST TELEMETRY on thingksboard web(1122-->1132--->1142---->1152........etc),then I see value is not do it now(1122-->1132--->1127---->1142---->1137---->1152........etc). Actually, The result is correct finally,Is it  normal?

 The log from var/log/ thingsboard-gateway
[connector.log](https://github.com/thingsboard/thingsboard-gateway/files/4248616/connector.log)

zbeacon, 2020-02-25:
@ Ming Shu

I'm glad to hear that. Information depends on data from the OPC-UA server. The server sends information to the gateway after the gateway has been subscribed to some node.

Ming-Shu, 2020-02-25:
@zbeacon 

I don't understand about  value of information why re-back.
Because, I don't tough  OPC-UA server, that value was regular add before.But it is not do it.

connector.log  from var/log/ thingsboard-gateway
'attributes': [{'T001': '1634'}]---> [{'T001': '1639'}]--->[{'T001': '1650'}]--->[{'T001': '1644'}]--->[{'T001': '1650'}]--->[{'T001': '1654'}] 

Is this  data from the  OPC-UA server error action ?

Ming-Shu, 2020-02-26:
Hi @zbeacon 
I have observed the something status,

[image]
attributes': [{'T001': '13740'}]-->attributes': [{'T001': '13741'}]--->attributes': [{'T001': '13752'}]
then 
Python: New data change event on node Node(StringNodeId(ns=1;s=T001)), with val: 13748
[image]

--->attributes': [{'T001': '13752'}]--->--->attributes': [{'T001': '13748'}]
Python: New data change event on node Node(StringNodeId(ns=1;s=T001)), with val: 13752

[image]

So,I think it maybe is not normal action,what do you think about it?

Other case below

[image]

zbeacon, 2020-02-26:
Hi @Ming-Shu 

I think this is caused by double reading of one node, first the gateway subscribes to node and waits for a callback from the server. After that it re-scans the server and reads the current value of the node. But the information from the subscription was obtained after the second scan. I can add a parameter to the configuration to disable the subscription and only read the values ​​when scanning.
What do you think about this?

Ming-Shu, 2020-02-26:
@zbeacon 

Let me make sure I have it right, 
step below: 
1.  gateway subscribes to node,but not get current value of the node.
2.  scans the server and reads the current value of the node.
3.  the information from the subscription(step 1 requst) was obtained is step 2  value of the node
Because,it caused by double reading of one node,right?

If my understand is correct, your ideas is good and I think should do it.

zbeacon, 2020-02-26:
@Ming-Shu 

No, its flow is little bit different:
1. Scan server, read current value and subscribe
2. Scan server, read current value
3. The information from the subscription has been obtained.

I mean that the server sends information to the subscription after the gateway has already read node with current value.

I will add two parameters:
`disableSubscriptions` - gives ability to disable subscriptions
`subCheckPeriodMillis` - gives ability to install the period for subscription check

Ming-Shu, 2020-02-27:
@zbeacon 

I'm glad to hear your detail expound that,I understand eventually.

I  appreciate your reply again,and  look forward to releasing new version.

closed 2020-03-13 by Ming-Shu