thingsboard/thingsboard-gateway #247: OPC-UA: Get all variables from folder

david114, 2020-02-20:
Hey there,
i've never worked with OPC-UA and i need a little bit of help on how to receive all the sensor data.
**My OPC-UA Server Structure looks like this:**

[image]

I want to read out all variables that are in the main_folder (sensor1, sensor2, ...) **dynamically**
My opc.json currently looks like this:

[image]

Another problem (maybe because I'm using invalid syntax) is that the opcua-connector keeps reconnecting to the opc-ua server, and i really don't know why:

[image]

I'd be really happy if you could help me out with this one. 
Thanks!

Ming-Shu, 2020-02-21:
Hello @david114 

Actually,I have the similar experience too!
Your opc.json------>"deviceNodePattern":"Root\\.Objects\\.man_folder",
About reconnecting to the opc-ua server, maybe  you can check your host IP of "tb_gateway.yaml"

Hope this would help you.

david114, 2020-02-23:
Thanks. i got it to work right now.

closed 2020-02-23 by david114