thingsboard/thingsboard-gateway #249: [BUG] Server connection timeout is always set to 10 seconds

david114, 2020-02-23:
I'm experiencing a weird behaviour using the OPC-UA Connector.
Although i set `timeoutInMillis: 2000` the timeout is always 10 seconds.
i want to update my telemetry data every 2 or 3 seconds, is there any other parameter that i need to set?

david114, 2020-02-24:
Here some additional info:

**My opcua.json:**
```
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "localhost:4840/freeopcua/server/",
    "scanPeriodInMillis": 1000,
    "timeoutInMillis": 1000,
    "showMap": true,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.main_folder",
        "deviceNamePattern": "Machine",
        "attributes": [
          {
            "key": "Sensor_2",
            "path": "${sensor2}"
          }
        ],
        "timeseries": [
          {
            "key": "Sensor_1",
            "path": "${sensor1}"
          }
        ]
      }
    ]
  }
}
```

**My tb_gateway.yaml:**
```
thingsboard:
  host: <my-ip>
  port: 1883
  remoteConfiguration: true
  security:
    accessToken: <my-accesstoken>
storage:
  type: file
  data_folder_path: ./data/
  max_file_count: 100
  max_read_records_count: 100
  max_records_per_file: 10000
connectors:
#  -
#   name: MQTT Broker Connector
#   type: mqtt
#   configuration: mqtt.json
#
#  -
#    name: Request Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
  -
    name: OPC-UA Connector
    type: opcua
    configuration: opcua.json

#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector
```

With this configuration, is it not possible to send data every second to thingsboard, why?

zbeacon, 2020-02-24:
Hi @david11, 

Thanks for your interest in the ThingsBoard IoT gateway. We were going to implement the regular polling of the OPC-UA server, but at the moment we have released subscription for the interesting nodes. And the gateway subscribes to the node and when the node updates its value, the gateway receives notification about this event and sends new value to the ThingsBoard. We will add rescan ability in one of the future releases.

david114, 2020-02-24:
My nodes are updating every second. So the gateway should receive a notification and send new values to thingsboard every second, or am I wrong?

zbeacon, 2020-02-24:
You are right, i will examine this behavior. I have added ability to use the rescan feature in the new pre-release, please try to use the version 2.1.0.8 and give your feedback.

david114, 2020-02-24:
I tried it, unfortunately im experiencing the same behaviour.

```
""2020-02-24 14:12:12" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 185 - connection SUCCESS"
""2020-02-24 14:12:12" - INFO - [tb_client.py] - tb_client - 78 - TB client <paho.mqtt.client.Client object at 0x75b53ef0> has been disconnected."
""2020-02-24 14:12:23" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 185 - connection SUCCESS"
""2020-02-24 14:12:23" - INFO - [tb_client.py] - tb_client - 78 - TB client <paho.mqtt.client.Client object at 0x75b53ef0> has been disconnected."
""2020-02-24 14:12:33" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 185 - connection SUCCESS"
""2020-02-24 14:12:33" - INFO - [tb_client.py] - tb_client - 78 - TB client <paho.mqtt.client.Client object at 0x75b53ef0> has been disconnected."
""2020-02-24 14:12:43" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 185 - connection SUCCESS"
""2020-02-24 14:12:43" - INFO - [tb_client.py] - tb_client - 78 - TB client <paho.mqtt.client.Client object at 0x75b53ef0> has been disconnected."
""2020-02-24 14:12:53" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 185 - connection SUCCESS"
""2020-02-24 14:12:53" - INFO - [tb_client.py] - tb_client - 78 - TB client <paho.mqtt.client.Client object at 0x75b53ef0> has been disconnected."
""2020-02-24 14:13:03" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 185 - connection SUCCESS"
""2020-02-24 14:13:03" - INFO - [tb_client.py] - tb_client - 78 - TB client <paho.mqtt.client.Client object at 0x75b53ef0> has been disconnected."
""2020-02-24 14:13:13" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 185 - connection SUCCESS"
""2020-02-24 14:13:13" - INFO - [tb_client.py] - tb_client - 78 - TB client <paho.mqtt.client.Client object at 0x75b53ef0> has been disconnected."
""2020-02-24 14:13:23" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 185 - connection SUCCESS"
""2020-02-24 14:13:23" - INFO - [tb_client.py] - tb_client - 78 - TB client <paho.mqtt.client.Client object at 0x75b53ef0> has been disconnected."
""2020-02-24 14:13:33" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 185 - connection SUCCESS"
""2020-02-24 14:13:34" - INFO - [tb_client.py] - tb_client - 78 - TB client <paho.mqtt.client.Client object at 0x75b53ef0> has been disconnected."

```

I'm using[ the installation from source](https://thingsboard.io/docs/iot-gateway/install/source-installation/), if that somehow helps you out

zbeacon, 2020-02-24:
This is a little different, please try to replace logging level "INFO" in logs.conf to "DEBUG" and you will see something like:
`""2020-02-24 15:40:02" - DEBUG - [opcua_connector.py] - opcua_connector - 455 - Data to ThingsBoard: {'deviceName': 'Device Boiler 2', 'deviceType': 'default', 'attributes': [], 'telemetry': [{'Temperature': '74.5'}]}"`

This indicates sending data to ThingsBoard.

david114, 2020-02-24:
Its seems to work right now, at least when looking at the logs:

```
""2020-02-24 14:41:13" - DEBUG - [opcua_connector.py] - opcua_connector - 377 - Python: New data change event on node Node(StringNodeId(ns=1;s=sensor2)), with val: 105.0"
""2020-02-24 14:41:13" - DEBUG - [opcua_connector.py] - opcua_connector - 383 - Data to ThingsBoard: {'deviceName': 'Davids_Aufwickler', 'deviceType': 'default', 'attributes': [{'Sensor_2': '105.0'}], 'telemetry': [{'Sensor_1': '105.0'}]}"
""2020-02-24 14:41:18" - DEBUG - [opcua_connector.py] - opcua_connector - 377 - Python: New data change event on node Node(StringNodeId(ns=1;s=sensor2)), with val: 106.0"
""2020-02-24 14:41:18" - DEBUG - [opcua_connector.py] - opcua_connector - 383 - Data to ThingsBoard: {'deviceName': 'Davids_Aufwickler', 'deviceType': 'default', 'attributes': [{'Sensor_2': '106.0'}], 'telemetry': [{'Sensor_1': '106.0'}]}"
""2020-02-24 14:41:18" - DEBUG - [opcua_connector.py] - opcua_connector - 377 - Python: New data change event on node Node(StringNodeId(ns=1;s=sensor2)), with val: 106.0"
""2020-02-24 14:41:18" - DEBUG - [opcua_connector.py] - opcua_connector - 383 - Data to ThingsBoard: {'deviceName': 'Davids_Aufwickler', 'deviceType': 'default', 'attributes': [{'Sensor_2': '106.0'}], 'telemetry': [{'Sensor_1': '106.0'}]}"
```
However, when i look in thingsboard, the updated values won't really show up randomly. Sometimes every value and sometimes no value at all for minutes. Maybe the server im running thingsboard on isn't powerful enough, i'll have to investigate.

zbeacon, 2020-02-24:
Glad to hear that, please let me know about results of investigation.

david114, 2020-02-25:
I've checked to connection to my thingsboard server via wireshark.
Although the gateway is logging that data is being sent every 2 seconds like in the logs:
```
""2020-02-25 09:42:47" - DEBUG - [opcua_connector.py] - opcua_connector - 377 - Python: New data change event on node Node(StringNodeId(ns=1;s=sensor1)), with val: 384"
""2020-02-25 09:42:47" - DEBUG - [opcua_connector.py] - opcua_connector - 383 - Data to ThingsBoard: {'deviceName': 'Davids_Aufwickler', 'deviceType': 'default', 'attributes': [], 'telemetry': [{'Sensor_1': '384'}]}"
```
there is nothing really being sent to my thingsboard server. Wireshark only loggs SUBACK and PUBACK every 10 seconds.
```
""2020-02-25 09:42:22" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f88bed3a610>, None, 16, 'Received SUBACK')"
""2020-02-25 09:42:22" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f88bed3a610>, None, 16, 'Received SUBACK')"
""2020-02-25 09:42:22" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f88bed3a610>, None, 16, 'Received SUBACK')"
""2020-02-25 09:42:22" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f88bed3a610>, None, 16, 'Received SUBACK')"
""2020-02-25 09:42:22" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f88bed3a610>, None, 16, 'Received SUBACK')"
""2020-02-25 09:42:22" - DEBUG - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 66 - Service subscription to topic v1/gateway/attributes - successfully completed."
""2020-02-25 09:42:22" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f88bed3a610>, None, 16, 'Received SUBACK')"
""2020-02-25 09:42:22" - DEBUG - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 66 - Service subscription to topic v1/gateway/attributes/response - successfully completed."
""2020-02-25 09:42:22" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f88bed3a610>, None, 16, 'Received SUBACK')"
""2020-02-25 09:42:22" - DEBUG - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 66 - Service subscription to topic v1/gateway/rpc - successfully completed."
""2020-02-25 09:42:22" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f88bed3a610>, None, 16, 'Received SUBACK')"
""2020-02-25 09:42:22" - DEBUG - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 66 - Service subscription to topic v1/gateway/rpc/response - successfully completed."
""2020-02-25 09:42:22" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f88bed3a610>, None, 16, 'Received PUBACK (Mid: 34)')"
""2020-02-25 09:42:22" - INFO - [tb_client.py] - tb_client - 78 - TB client <paho.mqtt.client.Client object at 0x7f88bed3a610> has been disconnected."
""2020-02-25 09:42:22" - DEBUG - [tb_device_mqtt.py] - tb_device_mqtt - 170 - <paho.mqtt.client.Client object at 0x7f88bed3a610>"
```
 So I'm basically still facing the old issue here. Is there any port needed to be openend besides `1883`?

david114, 2020-02-25:
**[This](https://thingsboard.io/docs/iot-gateway/install/source-installation/)** installation via pip doesn't work for me.
i reinstalled thingsboard-gateway as package from **[this](https://thingsboard.io/docs/iot-gateway/install/deb-installation/)** tutorial and now its working.

@zbeacon could you provide the newest version (2.1.0.9) as a .deb package?

zbeacon, 2020-02-25:
Please write an output after running command:
`sudo pip3 show thingsboard-gateway`

The debian package on installation takes the latest version of the gateway package from the pypi at the moment.

david114, 2020-02-25:
```
Name: thingsboard-gateway
Version: 2.1.0.9
Summary: Thingsboard Gateway for IoT devices.
Home-page: https://github.com/thingsboard/thingsboard-gateway
Author: ThingsBoard
Author-email: info@thingsboard.io
License: Apache Software License (Apache Software License 2.0)
Location: /usr/lib/python3.8/site-packages
Requires: cffi, jsonpath-rw, regex, pip, jsonschema, lxml, opcua, paho-mqtt, pymodbus, pyserial, pytz, PyYAML, simplejson, pyrsistent
Required-by: 

```

zbeacon, 2020-02-28:
Hi @david114 

We have released a new version (2.2) and added the deb package in the release.

Thank you for your help in improving the gateway. :smile: :+1: 
I close this issue if you have any question please write to us.

closed 2020-02-28 by zbeacon