thingsboard/thingsboard-gateway #250: Modbus TCP connection problems

pehedima, 2020-02-25:
I installed the new TB gateway python version 2.1.0.9 on Ubuntu 18.04. and want to receive data from one modbus tcp device. The error message is as follows:

`Feb 25 08:20:45 lp042396-VirtualBox python3[2668]: Traceback (most recent call last):
Feb 25 08:20:45 lp042396-VirtualBox python3[2668]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 128, in __process_devices
Feb 25 08:20:45 lp042396-VirtualBox python3[2668]:     converted_data = self.__devices[device]["converter"].convert(config=None, data=device_responses)
Feb 25 08:20:45 lp042396-VirtualBox python3[2668]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 80, in convert
Feb 25 08:20:45 lp042396-VirtualBox python3[2668]:     result = decoder.decode_32bit_float()
Feb 25 08:20:45 lp042396-VirtualBox python3[2668]: UnboundLocalError: local variable 'decoder' referenced before assignment
Feb 25 08:20:46 lp042396-VirtualBox python3[2668]: ''2020-02-25 08:20:46' - DEBUG - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f811194a9b0>, None, 16, "Sending PUBLISH (d0, q1,
Feb 25 08:20:46 lp042396-VirtualBox python3[2668]: ''2020-02-25 08:20:46' - DEBUG - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f811194a9b0>, None, 16, "Sending PUBLISH (d0, q1,
Feb 25 08:20:46 lp042396-VirtualBox python3[2668]: ''2020-02-25 08:20:46' - DEBUG - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f811194a9b0>, None, 16, 'Received PUBACK (Mid: 54
Feb 25 08:20:46 lp042396-VirtualBox python3[2668]: ''2020-02-25 08:20:46' - DEBUG - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f811194a9b0>, None, 16, 'Received PUBACK (Mid: 54
l`

The modbus device is defined in the following file:
[modbus-ph-config-new.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/4248704/modbus-ph-config-new.json.txt)

Currently I use the following tb_gateway.yaml
[tb_gateway.yaml.txt](https://github.com/thingsboard/thingsboard-gateway/files/4248705/tb_gateway.yaml.txt)

Someone an idea what goes wrong here?

zbeacon, 2020-02-25:
Hi @pehedima,

Thanks for your insterest in the ThingsBoard IoT gateway. It looks like some problem with receiving information from device. Please try to use long instead of the double in configuration and check is it will send some information to the ThingsBoard? I will investigate this behavior and let you know about results.

pehedima, 2020-02-25:
Thanks for your reply.
Ich changed the format from double to long and restart the service.
The data was send to ThingsBoard 👍 , but the value was wrong.
Ich changed it back to double and now the korrekt double value is send.
It seems to be working now, but I am wondering why is was not working at the first attempt?

zbeacon, 2020-02-25:
Thanks for information. Unfortunately, i don't know why this happened. I will investigate this problem, maybe it was a one-time problem with reading configuration or something else. If something happens like this, please let me know.

closed 2020-03-02 by zbeacon