thingsboard/thingsboard-gateway #251: MQTToverSSL problem

pehedima, 2020-02-25:
MQTToverSSL is not working for me.

ThingsBoard.yml:

> ....
>     bind_port: "${MQTT_BIND_PORT:8883}"
> ....
>     # MQTT SSL configuration
>     ssl:
>       # Enable/disable SSL support
>       enabled: "${MQTT_SSL_ENABLED:true}"
>       # SSL protocol: See http://docs.oracle.com/javase/8/docs/technotes/guides/security/StandardNames.html#SSLContext
>       protocol: "${MQTT_SSL_PROTOCOL:TLSv1.2}"
>       # Path to the key store that holds the SSL certificate
>       key_store: "${MQTT_SSL_KEY_STORE:mqttserver_iot.jks}"
>       # Password used to access the key store
>       key_store_password: "${MQTT_SSL_KEY_STORE_PASSWORD:serverems12345}"
>       # Password used to access the key
>       key_password: "${MQTT_SSL_KEY_PASSWORD:serverems12345}"
>       # Type of the key store
>       key_store_type: "${MQTT_SSL_KEY_STORE_TYPE:JKS}"
>   

The port 8883 ist working and the 'old' java version of tb-gateway is able to connect with the two 'old' parameters:

>           truststore: mqttclient.jks
>           truststorePassword: clientems12345

The python version needs caCert and cert.

I tried the following tb_gateway.yaml

>     port: 8883
> security:
>     accessToken: g1ynS7ftk0YKbNXR8OBq
>     caCert: /etc/thingsboard-gateway/config/mqttserver_iot.pub.pem
>     cert: none
> storage:
> 

The file mqttserver_iot.pub.pem ist located in directory /etc/thingsboard-gateway/config.
No other pem files are available.

But no connection to ThingsBoard is possbile any more.

Reports:
[tb_connection.log](https://github.com/thingsboard/thingsboard-gateway/files/4249281/tb_connection.log)
[thingsboard.txt](https://github.com/thingsboard/thingsboard-gateway/files/4249282/thingsboard.txt)

Someone an idea what is wrong here?
Thanks in advance.

zbeacon, 2020-02-25:
Hi @pehedima 

We don't longer support the Java-based gateway, please try to use the new Python-based gateway for your purposes. You can use the SSL, in the new gateway.  
To add certificates to the gateway please use [this article](https://thingsboard.io/docs/iot-gateway/config/mqtt/?mqttConnectorCredentialsConfig=tls#subsection-security). If you will have any questions please write us.

pehedima, 2020-02-25:
Thanks, I know, that the java version is not supported any more. That was only a hit, that the java version has no problem to connect the Server instead of the python version.
 
As mentioned above, I checked the following configuration in tb_gateway.yaml

```
    ...
    port: 8883
    ...
    security:
    accessToken: g1ynS7ftk0YKbNXR8OBq
    caCert: /etc/thingsboard-gateway/config/mqttserver_iot.pub.pem
    cert: none
    storage:

```

Refer to your last link, I understood, that the security section could be filled in mgtt.json!?

Therfore I tried the second configuration:

tb_gateway.yaml:
```
    ...
    port: 8883
    ...
    security:
    accessToken: g1ynS7ftk0YKbNXR8OBq
    storage:

```
mqtt.json:

```
  "broker": {
    "name":"Default Local Broker",
    "host":"<my-ip>",
    "port":8883,
    "security": {
    "caCert": "/etc/thingsboard-gateway/conf/mqttserver_iot.pub.pem",
    }
```
In both cases no contection to Thingsboard is possible
Error:

```
Feb 25 13:53:36 lp042396-VirtualBox python3[16681]: ''2020-02-25 13:53:36' - DEBUG - bytes_modbus_uplink_converter - 36 - Called convert function from BytesModbusUplinkConverter with args'
Feb 25 13:53:36 lp042396-VirtualBox python3[16681]: ''2020-02-25 13:53:36' - DEBUG - bytes_modbus_uplink_converter - 37 - {'byteOrder': 'BIG', 'tag': 'Freq (Hz)', 'type': 'double', 'functio
Feb 25 13:53:36 lp042396-VirtualBox python3[16681]: ''2020-02-25 13:53:36' - DEBUG - bytes_modbus_uplink_converter - 38 - ReadRegisterResponse (2)'
Feb 25 13:53:36 lp042396-VirtualBox python3[16681]: ''2020-02-25 13:53:36' - DEBUG - bytes_modbus_uplink_converter - 115 - {'deviceName': 'Phoenix-3N-Python', 'deviceType': 'ModbusDevice',
Feb 25 13:53:36 lp042396-VirtualBox python3[16681]: ''2020-02-25 13:53:36' - DEBUG - tb_client - 108 - connecting to ThingsBoard'
Feb 25 13:53:36 lp042396-VirtualBox python3[16681]: ''2020-02-25 13:53:36' - DEBUG - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f62a36d7be0>, None, 16, "Sending CONNECT (u1, p0
Feb 25 13:53:37 lp042396-VirtualBox python3[16681]: ''2020-02-25 13:53:37' - DEBUG - tb_client - 108 - connecting to ThingsBoard'
Feb 25 13:53:37 lp042396-VirtualBox python3[16681]: ''2020-02-25 13:53:37' - DEBUG - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f62a36d7be0>, None, 16, "Sending CONNECT (u1, p0
Feb 25 13:53:38 lp042396-VirtualBox python3[16681]: ''2020-02-25 13:53:38' - DEBUG - tb_client - 108 - connecting to ThingsBoard'
Feb 25 13:53:38 lp042396-VirtualBox python3[16681]: ''2020-02-25 13:53:38' - DEBUG - tb_client - 61 - (<paho.mqtt.client.Client object at 0x7f62a36d7be0>, None, 16, "Sending CONNECT (u1, p0
~
```

and addtional error messages on thindboard:
```
2020-02-25 14:03:53,395 [nioEventLoopGroup-4-2] ERROR o.t.s.t.mqtt.MqttTransportHandler - [2bee9cff-a1f9-45f1-b535-3855d6e5595b] Unexpected Exception
io.netty.handler.codec.DecoderException: io.netty.handler.ssl.NotSslRecordException: not an SSL/TLS record: 102200044d5154540482003c000000146731796e533766746b30594b624e5852384f4271
        at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:470)
        at io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:276)
        at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:374)
        at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:360)
        at io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:352)
        at io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1408)
        at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:374)
        at io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:360)
        at io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:930)
        at io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:163)
        at io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:697)
        at io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:632)
        at io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:549)
        at io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:511)
        at io.netty.util.concurrent.SingleThreadEventExecutor$5.run(SingleThreadEventExecutor.java:918)
        at io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
        at io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
        at java.lang.Thread.run(Thread.java:748)
Caused by: io.netty.handler.ssl.NotSslRecordException: not an SSL/TLS record: 102200044d5154540482003c000000146731796e533766746b30594b624e5852384f4271
        at io.netty.handler.ssl.SslHandler.decodeJdkCompatible(SslHandler.java:1203)
        at io.netty.handler.ssl.SslHandler.decode(SslHandler.java:1271)
        at io.netty.handler.codec.ByteToMessageDecoder.decodeRemovalReentryProtection(ByteToMessageDecoder.java:500)
        at io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:439)
        ... 17 common frames omitted
^
```

May be you can send me a concret example. I've only the one *.pem file. 
Thanks for your support

zbeacon, 2020-02-27:
Unfortunately i cannot understand why it doesn't work with your certificate. It looks like your file contains not correct certificate. ```Caused by: io.netty.handler.ssl.NotSslRecordException: not an SSL/TLS record```. Please try to regenerate the certificates on device or try to use another way to generate the certificate.

david114, 2020-02-28:
@zbeacon 
@pehedima 

i faced the same issue and my solution was to add` tls: true `to the security section of `tb_gateway.yml`
The` tb_client.py `awaits a tls parameter:

`        self.__tls = True if credentials.get('tls', False) else False
`
This config works for me:

```
thingsboard:
  host: xxxxxxx.de
  port: 8883
  remoteConfiguration: true
  security:
    tls: true
    accessToken: xxxxxxxxx
    caCert: trustid-x3.pem
```

zbeacon, 2020-02-28:
Hi @david114 

Thank you for your investigations.

closed 2020-03-05 by zbeacon

pehedima, 2020-03-06:
Perfekt. Thanks. 
MQTToverSSL is working now.