thingsboard/thingsboard-gateway #252: How to use RPC to write device registers via Modbus protocol?

laoli6666, 2020-02-26:
I am using TB2.4.3 and GW (Python) to implement the Modbus function these days.
I have successfully implemented the function of reading device registers, but the function of **Writing device registers via Modbus protocol** has not been implemented.
 I know that I should use the RPC function, but I do n’t know The specific implementation method, tried many times without success, I hope the author can provide a complete implementation documentation, thank you very much!
Finally, thank you again for such an excellent product!

zbeacon, 2020-02-26:
Hi @laoli6666 

Thank you for your interest in the ThingsBoard IoT Gateway.
Modbus functions 5, 6, 15, 16 are implemented. Maybe it is some bug.
I need to investigate this issue.
Please attach your config file or describe, which register, bit address, etc you want to change? - I will configure the gateway and update an example in the documentation.

laoli6666, 2020-02-26:
Thanks a lot！
-----------------------
My “tb_gateway.yaml” :
thingsboard:
  host: 127.0.0.1
  port: 1883
  security:
    accessToken: MyGateWay_TOKEN
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:

  -
    name: Modbus Connector
    type: modbus
    configuration: modbus.json
---------------------------------------------
My "modbus.json" :
{
  "server": {
    "name": "Modbus 172.18.0.3",
    "type": "tcp",
    "host": "172.18.0.3",
    "port": 502,
    "timeout": 350,
    "rtuOverTcp": false,
    "devices": [
      {
        "unitId": 3,
        "deviceName": "ModbusDVC",
        "attributesPollPeriod": 120000,
        "timeseriesPollPeriod": 120000,
        "sendDataOnlyOnChange": false,
        "timeseries": 
          {
            "byteOrder": "BIG",
            "tag": "Temp",
            "type": "long",
            "functionCode": 3,
            "registerCount": 1,
            "address": 0
          },
        "rpc": {
          "setValue": {
            "tag": "Integer",
            "value": 6666,
            "functionCode": 16,
            "address": 6,
            "unitId": 3,
            "byteOrder": "BIG",
            "registerCount": 1
          }
        }
      }
    ]
  }
}
-------------------------------------------------------------
Then I use a "RPC debug terminal" to debug it in Dashboard, But no matter what command I send, such as "setValue ...", I always receive "Device is offline"
I think I should be doing something wrong, but I don't know where.

zbeacon, 2020-02-26:
Ok, i will try to emulate this behavior in my test server. Also please check logs (/var/log/thingsboard/connector.log and /var/log/thingsboard/service.log) - Are there any exceptions?

laoli6666, 2020-02-26:
OK, thank you very much!
In addition, I would like to ask a question, what is the relationship between the various items defined in GW's "modbus.json" and the parameters that need to be transmitted during the RPC call?
Can I understand it this way:
1. Will the options defined in "modbus.json" be replaced by the options passed?
2. There is no need to define it completely in "modbus.json", as long as it can be supplemented by passed parameters.

I really need you to provide a complete example document and operation steps, as long as it can write registers, this function is very important.
Thanks again, hope it didn't take up much of your time.

zbeacon, 2020-02-26:
In dashboard you need to specify device that was created by the gateway as an entity for RPC terminal. Please check that you use the dashboard with correct entity. You need to send the RPC commands to the device instead of the gateway.

          "setValue": {
            "tag": "Integer",
            "value": 6666,
            "functionCode": 16,
            "address": 6,
            "unitId": 3,
            "byteOrder": "BIG",
            "registerCount": 1
          } 

With these configuration the gateway will try to send value 6666 to register with address 6, device id - 3. Also you can try to use function with code 6 instead 16.

laoli6666, 2020-02-26:
I tried again, and the data did write, but "Device is offline" was still displayed in the "RPC debug terminal".
-------------------------
 I would like to ask a question, what is the relationship between the various items defined in GW's "modbus.json" and the parameters that need to be transmitted during the RPC call?
Can I understand it this way？
1.Will the options defined in "modbus.json" be replaced by the options passed?
2.There is no need to define it completely in "modbus.json", as long as it can be supplemented by passed parameters？

zbeacon, 2020-02-26:
1. Options from the Modbus file are used for processing RPC call, there is no ability to use parameters from RPC call as configuration yet.
2. Parameters from the configuration needs to Modbus downlink converter, it converts the data received from the ThingsBoard in the RPC request to methods that will work in the Modbus server. If you are familiar with Python, you can implement converter for this or refactor existing and create pull request. Perhaps we will implement this feature in one of the future releases.

laoli6666, 2020-02-27:
Thank you very much for your reply, and we look forward to your new version of GW very much. After all, Modbus devices often have many registers and need more flexible operation.

I also hope that GW can have a primitive function, such as telling him the port or IP address to send data through TB, and it can send a primitive Modbus instruction, and the remaining work can be done by powerful JS, because we still hope Keep the same with the official original software, which will facilitate future version upgrades.

laoli6666, 2020-02-27:
In addition, I thought of another way, I don't know if it can be achieved:
1. I want to know if the communication between GW and TB about RPC is through the rule chain?
2. If yes, can outbound messages pass through the rule chain? In this case, can I shape or process the RPC message so that he can meet my requirements?

zbeacon, 2020-02-27:
1. Yes, the relation in the Root rule chain is "RPC Request to Device".
2. You can change the message to/from device using rule chain.

chad007, 2020-04-18:
Hi @laoli6666
I have a question which is not quite relevant to yours: how to make the GW(python) status "sync"? My TB can already receives values of device attributes reported by the GW(python), which are defined in the "modbus.json" of the GW, but the status of GW(python) is always "Not sync". With the "Not sync" status, seems like I cannot provision new device from TB to GW(python), this is a big problem in my case. Thanks!

zbeacon, 2020-05-15:
Hi @chad007

We've implemented a new configuration and RPC requests processing in the gateway, please upgrade your gateway and try to use the new [Modbus configuration](https://thingsboard.io/docs/iot-gateway/config/modbus/). If you will have any questions, please create a new version.

closed 2020-05-15 by zbeacon