thingsboard/thingsboard-gateway #254: change opcua.json (connector config) remotely

david114, 2020-03-04:
Hey,

i have read all docs but i couldn't really figure out if its possible to change the connector config file remotely. In my case the `opcua.json`

To visualize what I'm trying to achieve i made a small graphic:

[image]

I can't access the RaspberryPi (Thingsboard IoT Gateway) directly anymore, because it's located remotely. The problem I'm facing is that the configuration from the OPC-UA Server may change from time to time. So i need to be able to configure the `opcua.json `to new telemetry data.

How can I achieve this?

Thanks in advance!

zbeacon, 2020-03-04:
Hi @david114 

I have released this feature in the gateway, but unfortunately, there is no correct dashboard for remote configuration in the ThingsBoard yet, it will be added in the next release.

Also to use this feature you need to add/set configuration parameter ** remoteConfiguration** to **true** in the tb_gateway.yaml

david114, 2020-03-04:
Allright, i'll be patient then :)
Thanks for answering quickly.

closed 2020-03-04 by david114