thingsboard/thingsboard-gateway #255: Thingsboard-gateway V2.2 Communication with Server

mirtaheri, 2020-03-05:
Hello!
we have installed and followed the instructions for the thingsboard gateway but failed to stablish communication with Thingsboardf server. It is worth to mention that with previous versions (JAVA) the implementation is successfull, so I wonder if I am missing correct configuration for the current (python) version, with some slight format changes w.r.t to the past.
The gateway should interface with a sensor (namely weather station) device through mqtt connector and send the data to the server.
The weather station -through `weewex` mqtt interface- is publishing successfully the sensor values.

The `tb_gateway.yaml` -communication to sever- is configured as following:

```
thingsboard:
  host: mydomain.it
  port: 1883
  remoteConfiguration: false
  security:
    accessToken: UVXXXXXXXXXXXX
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
connectors:
  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json
```

The  `mqtt.json` which is used as gateway connector, is set (minimal setup) like below:

```
{
  "broker": {
    "name": "Weather Station Broker",
    "host": "localhost",
    "port": 1883,
    "security": {
      "type": "anonymous"
    }
  },
  "mapping": [
    {
      "topicFilter": "WeatherSensors",
      "converter": {
        "type": "json",
        "deviceTypeJsonExpression": "${sensorType}",
        "timeout": 60000,
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${outTemp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${outHumidity}"
          }
        ],
        "deviceNameJsonExpression": "${WeatherStationName}"
      }
    }
  ]
}
```
WeatherSensors is the topic on which `weewx-mqtt`, and `outTemp`, `outHumidity` are fields in the payload.

I receive in the dashboard (server-side) these keys: `eventsProduced` and `eventsSent` and haveing all fields in the `mqtt.json` (`connectRequests`, `disconnectRequests`, `attributeUpdates `and `serverSideRpc`) I get also `mqttbrokerconnectorEventsProduced` and `mqttbrokerconnectorEventsSent`.
So, I assume that connectin to server is not a problem, but at the same time I am sure about the sensor (weewex) topic and payload.

zbeacon, 2020-03-05:
Hi @mirtaheri 

Thank you for your interest in ThingsBoard IoT.

Unfortunately, I cannot find the problem without the logs, please try to find exceptions in the logs and attach them to your message. At the moment, I can only say that the gateway has successfully connected to ThingsBoard and initialized the connectors. But it seems that some kind of problem occurred at the time of subscription or when the gateway received some information. We need to examine “connector.log” and “converter.log” to find the problem.
Also, please try using "/" in the subject of the Filter parameter - "/WeatherSensors" instead of "WeatherSensors"

mirtaheri, 2020-03-05:
Dear Ilya,
thanks you very much for the promp reply!
I restarted the thingsboard-gateway and update this thread with the latest logs.
`connector.log`:

''2020-03-05 16:02:12' - DEBUG - tb_logger - 37 - Added remote handler to log extension'
''2020-03-05 16:02:12' - DEBUG - tb_logger - 37 - Added remote handler to log converter'
''2020-03-05 16:02:12' - DEBUG - tb_logger - 37 - Added remote handler to log connector'

`converter.log` is empty,
but in service.log there are some contents (errors) that might give the clue;
```
''2020-03-05 16:02:12' - DEBUG - tb_logger - 37 - Added remote handler to log service'
''2020-03-05 16:02:12' - DEBUG - tb_utility - 68 - ModuleSpec(name='MqttConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0xb60b4270>, origin='/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/mqtt_connector.py')'
''2020-03-05 16:02:12' - DEBUG - tb_utility - 74 - <module 'MqttConnector' from '/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/mqtt_connector.py'>'
''2020-03-05 16:02:12' - DEBUG - tb_utility - 81 - Import MqttConnector from /usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt.'
''2020-03-05 16:02:12' - ERROR - tb_gateway_service - 225 - 'connectRequests''
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 220, in _connect_with_connectors
    connector_type)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 42, in __init__
    self.__get_service_config(config)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 212, in __get_service_config
    self.__attribute_updates = config[service_config]
KeyError: 'connectRequests'
''2020-03-05 16:02:12' - DEBUG - tb_gateway_service - 505 - Loaded devices:
 {}'
''2020-03-05 16:02:12' - DEBUG - tb_gateway_service - 269 - Send data Thread has been started successfully.'
''2020-03-05 16:02:13' - DEBUG - tb_gateway_service - 151 - Received data: {}'
''2020-03-05 16:02:13' - DEBUG - tb_gateway_service - 152 - (None,)'
```

P.N. I also had tried `/WeatherSensors` and various alternatives, but apparently this was not the problem.

Thanks again for your availability.

mirtaheri, 2020-03-05:
Update!
After I shared the logs here out of a system start with "minimum inputs" in mqtt.json (as shared before, I had removed some fields), I also restarted the service with the complete configuration (what is provided as a template in [Thingsboard instructions](https://thingsboard.io/docs/iot-gateway/config/mqtt/)). The new logs are as followings:

`connector.log`:

```
''2020-03-05 17:02:52' - DEBUG - tb_logger - 37 - Added remote handler to log extension'
''2020-03-05 17:02:52' - DEBUG - tb_logger - 37 - Added remote handler to log converter'
''2020-03-05 17:02:52' - DEBUG - tb_logger - 37 - Added remote handler to log connector'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, "Sending CONNECT (u0, p0, wr0, wq0, wf0, c1, k60) client_id=b'jwlowetjutuekorsbftdfoh'")'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, 'Received CONNACK (0, 0)')'
''2020-03-05 17:02:52' - INFO - mqtt_connector - 144 - MQTT Broker Connector connected to localhost:1883 - successfully.'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, "Sending SUBSCRIBE (d0, m1) [(b'WeatherSensors', 0)]")'
''2020-03-05 17:02:52' - INFO - mqtt_connector - 170 - Connector "MQTT Broker Connector" subscribe to WeatherSensors'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, "Sending SUBSCRIBE (d0, m2) [(b'/sensor/+/data', 0)]")'
''2020-03-05 17:02:52' - INFO - mqtt_connector - 170 - Connector "MQTT Broker Connector" subscribe to /sensor/+/data'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 152 - Custom converter for topic /custom/sensors/+ - found!'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, "Sending SUBSCRIBE (d0, m3) [(b'/custom/sensors/+', 0)]")'
''2020-03-05 17:02:52' - INFO - mqtt_connector - 170 - Connector "MQTT Broker Connector" subscribe to /custom/sensors/+'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, "Sending SUBSCRIBE (d0, m4) [(b'sensor/connect', 0)]")'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, "Sending SUBSCRIBE (d0, m5) [(b'sensor/+/connect', 0)]")'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, "Sending SUBSCRIBE (d0, m6) [(b'sensor/disconnect', 0)]")'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, "Sending SUBSCRIBE (d0, m7) [(b'sensor/+/disconnect', 0)]")'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, 'Received SUBACK')'
''2020-03-05 17:02:52' - INFO - mqtt_connector - 201 - "MQTT Broker Connector" subscription success to topic WeatherSensors, subscription message id = 1'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, 'Received SUBACK')'
''2020-03-05 17:02:52' - INFO - mqtt_connector - 201 - "MQTT Broker Connector" subscription success to topic /sensor/+/data, subscription message id = 2'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, 'Received SUBACK')'
''2020-03-05 17:02:52' - INFO - mqtt_connector - 201 - "MQTT Broker Connector" subscription success to topic /custom/sensors/+, subscription message id = 3'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, 'Received SUBACK')'
''2020-03-05 17:02:52' - INFO - mqtt_connector - 201 - "MQTT Broker Connector" subscription success to topic sensor/connect, subscription message id = 4'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, 'Received SUBACK')'
''2020-03-05 17:02:52' - INFO - mqtt_connector - 201 - "MQTT Broker Connector" subscription success to topic sensor/+/connect, subscription message id = 5'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, 'Received SUBACK')'
''2020-03-05 17:02:52' - INFO - mqtt_connector - 201 - "MQTT Broker Connector" subscription success to topic sensor/disconnect, subscription message id = 6'
''2020-03-05 17:02:52' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, 'Received SUBACK')'
''2020-03-05 17:02:52' - INFO - mqtt_connector - 201 - "MQTT Broker Connector" subscription success to topic sensor/+/disconnect, subscription message id = 7'
''2020-03-05 17:03:15' - DEBUG - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0xb605fcf0>, None, 16, "Received PUBLISH (d0, q0, r0, m0), 'WeatherSensors', ...  (663 bytes)")'
''2020-03-05 17:03:15' - DEBUG - json_mqtt_uplink_converter - 64 - "sensorModel" key not found in message: {'outHumidity': 78.62785651801163, 'humidex': -1.072269222545263, 'consBatteryVoltage': 11.953168077537022, 'WeatherStationName': 'VantagePRO2_Device', 'rainRate': 0.0, 'heatindex': -1.072269222545263, 'radiation': 61.655498441002415, 'forecastRule': 0, 'inTemp': 17.7188012304434, 'inDewpoint': -0.9235889071950036, 'barometer': 1051.617847245405, 'windchill': -1.0722692225452 [... cut ...] 51801163"}]}'
''2020-03-05 17:03:15' - ERROR - tb_gateway_service - 232 - Data from MQTT Broker Connector connector is invalid.'
''2020-03-05 17:04:15' - ERROR - tb_utility - 41 - deviceType is empty in data: {"deviceName": "VantagePRO2_Device", "deviceType": null, "attributes": [], "telemetry": [{"temperature": "-1.068114048896523"}, {"humidity": "78.60822077600085"}]}'
''2020-03-05 17:04:15' - ERROR - tb_gateway_service - 232 - Data from MQTT Broker Connector connector is invalid.'
```
Thanks!

zbeacon, 2020-03-05:
The problem is in the configuration, the gateway needs all sections inside the configuration. 
Just copy following configuration and try again:

`{
  "broker": {
    "name": "Weather Station Broker",
    "host": "localhost",
    "port": 1883,
    "security": {
      "type": "anonymous"
    }
  },
  "mapping": [
    {
      "topicFilter": "WeatherSensors",
      "converter": {
        "type": "json",
        "deviceTypeJsonExpression": "${sensorType}",
        "timeout": 60000,
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${outTemp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${outHumidity}"
          }
        ],
        "deviceNameJsonExpression": "${WeatherStationName}"
      }
    }
  ],
  "connectRequests": [],
  "disconnectRequests": [],
  "attributeUpdates": [],
  "serverSideRpc": []
}`

mirtaheri, 2020-03-05:
Well, this is done, now remains one error in logs;
```
''2020-03-05 19:03:15' - ERROR - tb_utility - 41 - deviceType is empty in data: {"deviceName": "VantagePRO2_Device", "deviceType": null, "attributes": [], "telemetry": [{"temperature": "0.8752049217736079"}, {"humidity": "75.33100167591637"}]}'
''2020-03-05 19:03:15' - ERROR - tb_gateway_service - 232 - Data from MQTT Broker Connector connector is invalid.'
```
But actually the json message from sensor (weewx) doesn't contain `"deviceType"`. So if I leave  like         `"deviceTypeJsonExpression": "${sensorType}"` I give an error. If I refer it to a non-related field like `"deviceTypeJsonExpression": "${WeatherStationName}"` doesn't give error but no sensor value gets published to the server. I tried to use "default" and "gateway" instead of variable, it gave again error. I am not even sure if this error for `sensorType` causes the problem.

zbeacon, 2020-03-05:
@mirtaheri 

Ok, it's good. If the message from device doesn't contain device type just setup this value to "default" in configuration file instead of "${deviceType}". 
If you use construction ${} construction, the gateway tries to get the type from the incoming message.

mirtaheri, 2020-03-06:
When I set `deviceNameJsonExpression` and `deviceTypeJsonExpression` manually like this:

`{
  "broker": {
    "name": "Weather Station Broker",
    "host": "localhost",
    "port": 1883,
    "security": {
      "type": "anonymous"
    }
  },
  "mapping": [
    {
      "topicFilter": "WeatherSensors",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "odroid_VantagePRO_device",
        "deviceTypeJsonExpression": "default",
        "timeout": 60000,
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${outTemp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${outHumidity}"
          }
        ]
      }
    }
  ],
  "connectRequests": [],
  "disconnectRequests": [],
  "attributeUpdates": [],
  "serverSideRpc": []
}
`
it is not recognized at all:

`''2020-03-06 01:07:15' - ERROR - tb_utility - 38 - deviceName is empty in data {"deviceName": null, "deviceType": null, "attributes": [], "telemetry": [{"temperature": "16.498213791299385"}, {"humidity": "56.290688817195736"}]}'
''2020-03-06 01:07:15' - ERROR - tb_gateway_service - 232 - Data from MQTT Broker Connector connector is invalid.'`

zbeacon, 2020-03-06:
Hi @mirtaheri 

I was the issue with mqtt converter, please update your gateway and try again.

To update - you can [install the gateway from sources](https://thingsboard.io/docs/iot-gateway/install/source-installation/) or just run the following command `sudo pip3 install thingsboard-gateway --upgrade`

mirtaheri, 2020-03-06:
Dear Ilya,
It is working! 
Thank you  very much and best regards.

closed 2020-03-06 by mirtaheri

mirtaheri, 2020-03-06:
Just another question, to build from source, how we can have the debian `python3-thingsboard-gateway.deb` with the upgraded w.r.t this issue (#255)? Following the same instruction for 2.2.1?

krishna194, 2020-06-30:
I have to send data to the Thingsboard server using Thingsboard gateway with a mosquitto application
I have successfully connected to  Thingsboard gateway
My logs are attached below
[connector.txt](https://github.com/thingsboard/thingsboard-gateway/files/4850035/connector.txt)
[mqtt.txt](https://github.com/thingsboard/thingsboard-gateway/files/4850037/mqtt.txt)
[service.txt](https://github.com/thingsboard/thingsboard-gateway/files/4850038/service.txt)
[tb_gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/4850039/tb_gateway.txt)