thingsboard/thingsboard-gateway #257: cannot send data to thingsboard

Tracy6465, 2020-03-10:
I use the latest thingsboard gateway code, but I can't send data to thingsboard，
[image]
in the thingsboard,I can see the lastest telmetry
[image]
but the extension
[image]

zbeacon, 2020-03-10:
Hi @Tracy6465 ,

Thanks for your interest in the ThingsBoard IoT gateway.
Please set a logging level DEBUG instead INFO in file logs.conf in folder with tb_gateway.yaml. Run the gateway again and please attach logs.

Tracy6465, 2020-03-10:
@zbeacon 
/usr/local/bin/python3.7 /Users/iotinall/project/thingsboard-gateway/thingsboard_gateway/tb_gateway.py
""2020-03-10 16:39:16" - INFO - [tb_gateway_service.py] - tb_gateway_service - 48 - Gateway starting..."
""2020-03-10 16:39:16" - DEBUG - [tb_client.py] - tb_client - 108 - connecting to ThingsBoard"
""2020-03-10 16:39:16" - DEBUG - [tb_device_mqtt.py] - tb_device_mqtt - 353 - Subscribed to * with id 1"
""2020-03-10 16:39:16" - DEBUG - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 180 - Subscribed to *|* with id 1"
""2020-03-10 16:39:16" - DEBUG - [tb_logger.py] - tb_logger - 37 - Added remote handler to log service"
""2020-03-10 16:39:16" - DEBUG - [tb_logger.py] - tb_logger - 37 - Added remote handler to log storage"
""2020-03-10 16:39:16" - DEBUG - [tb_logger.py] - tb_logger - 37 - Added remote handler to log extension"
""2020-03-10 16:39:16" - DEBUG - [tb_logger.py] - tb_logger - 37 - Added remote handler to log converter"
""2020-03-10 16:39:16" - DEBUG - [tb_logger.py] - tb_logger - 37 - Added remote handler to log connector"
""2020-03-10 16:39:16" - DEBUG - [tb_logger.py] - tb_logger - 37 - Added remote handler to log tb_connection"
""2020-03-10 16:39:16" - DEBUG - [tb_utility.py] - tb_utility - 68 - ModuleSpec(name='MqttConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x11098c850>, origin='/Users/iotinall/project/thingsboard-gateway/thingsboard_gateway/connectors/mqtt/mqtt_uplink_converter.py')"
""2020-03-10 16:39:16" - DEBUG - [tb_utility.py] - tb_utility - 74 - <module 'MqttConnector' from '/Users/iotinall/project/thingsboard-gateway/thingsboard_gateway/connectors/mqtt/mqtt_uplink_converter.py'>"
""2020-03-10 16:39:16" - DEBUG - [tb_utility.py] - tb_utility - 68 - ModuleSpec(name='MqttConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x1109a84d0>, origin='/Users/iotinall/project/thingsboard-gateway/thingsboard_gateway/connectors/mqtt/mqtt_connector.py')"
""2020-03-10 16:39:16" - DEBUG - [tb_utility.py] - tb_utility - 74 - <module 'MqttConnector' from '/Users/iotinall/project/thingsboard-gateway/thingsboard_gateway/connectors/mqtt/mqtt_connector.py'>"
""2020-03-10 16:39:16" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, "Sending CONNECT (u1, p0, wr0, wq0, wf0, c1, k60) client_id=b''")"
""2020-03-10 16:39:16" - DEBUG - [tb_utility.py] - tb_utility - 81 - Import MqttConnector from /Users/iotinall/project/thingsboard-gateway/thingsboard_gateway/connectors/mqtt."
""2020-03-10 16:39:16" - DEBUG - [tb_gateway_service.py] - tb_gateway_service - 554 - Loaded devices:
 {}"
""2020-03-10 16:39:16" - DEBUG - [mqtt_connector.py] - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0x110a05610>, None, 16, "Sending CONNECT (u1, p1, wr0, wq0, wf0, c1, k60) client_id=b'spbxwpkzkhepwtcujhtivja'")"
""2020-03-10 16:39:16" - DEBUG - [tb_gateway_service.py] - tb_gateway_service - 290 - Send data Thread has been started successfully."
""2020-03-10 16:39:16" - INFO - [tb_gateway_service.py] - tb_gateway_service - 105 - Gateway started."
""2020-03-10 16:39:16" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, 'Received CONNACK (1, 0)')"
""2020-03-10 16:39:16" - DEBUG - [tb_client.py] - tb_client - 73 - TB client <paho.mqtt.client.Client object at 0x11098cbd0> connected to ThingsBoard"
""2020-03-10 16:39:16" - DEBUG - [mqtt_connector.py] - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0x110a05610>, None, 16, 'Received CONNACK (0, 0)')"
""2020-03-10 16:39:16" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 185 - connection SUCCESS"
""2020-03-10 16:39:16" - INFO - [mqtt_connector.py] - mqtt_connector - 144 - MQTT Broker Connector connected to 127.0.0.1:1893 - successfully."
""2020-03-10 16:39:16" - DEBUG - [tb_device_mqtt.py] - tb_device_mqtt - 186 - <paho.mqtt.client.Client object at 0x11098cbd0>"
""2020-03-10 16:39:16" - DEBUG - [mqtt_connector.py] - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0x110a05610>, None, 16, "Sending SUBSCRIBE (d0, m1) [(b'/sensor/data', 0)]")"
""2020-03-10 16:39:16" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, "Sending SUBSCRIBE (d0, m1) [(b'v1/devices/me/attributes', 1)]")"
""2020-03-10 16:39:16" - INFO - [mqtt_connector.py] - mqtt_connector - 170 - Connector "MQTT Broker Connector" subscribe to /sensor/data"
""2020-03-10 16:39:16" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, "Sending SUBSCRIBE (d0, m2) [(b'v1/devices/me/attributes/response/+', 1)]")"
""2020-03-10 16:39:16" - DEBUG - [mqtt_connector.py] - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0x110a05610>, None, 16, "Sending SUBSCRIBE (d0, m2) [(b'/sensor/+/data', 0)]")"
""2020-03-10 16:39:16" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, "Sending SUBSCRIBE (d0, m3) [(b'v1/devices/me/rpc/request/+', 0)]")"
""2020-03-10 16:39:16" - INFO - [mqtt_connector.py] - mqtt_connector - 170 - Connector "MQTT Broker Connector" subscribe to /sensor/+/data"
""2020-03-10 16:39:16" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, "Sending SUBSCRIBE (d0, m4) [(b'v1/devices/me/rpc/response/+', 1)]")"
[... cut ...]
""2020-03-10 16:39:48" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, 'Received PUBACK (Mid: 14)')"
""2020-03-10 16:39:54" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, "Sending PUBLISH (d0, q1, r0, m15), 'b'v1/devices/me/telemetry'', ... (116 bytes)")"
""2020-03-10 16:39:54" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, 'Received PUBACK (Mid: 15)')"

zbeacon, 2020-03-10:
Ok, try to send data to the topic "/sensor/data" instead of "sensor/connect".

Tracy6465, 2020-03-10:
@zbeacon 
""2020-03-10 17:23:14" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, "Sending PUBLISH (d0, q1, r0, m447), 'b'v1/devices/me/telemetry'', ... (116 bytes)")"
""2020-03-10 17:23:14" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, 'Received PUBACK (Mid: 447)')"
""2020-03-10 17:23:21" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, "Sending PUBLISH (d0, q1, r0, m448), 'b'v1/devices/me/telemetry'', ... (116 bytes)")"
""2020-03-10 17:23:21" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, 'Received PUBACK (Mid: 448)')"
""2020-03-10 17:23:27" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, "Sending PUBLISH (d0, q1, r0, m449), 'b'v1/devices/me/telemetry'', ... (116 bytes)")"
""2020-03-10 17:23:27" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, 'Received PUBACK (Mid: 449)')"
""2020-03-10 17:23:27" - DEBUG - [mqtt_connector.py] - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0x110a05610>, None, 16, "Received PUBLISH (d0, q0, r0, m0), '/sensor/data', ...  (134 bytes)")"
""2020-03-10 17:23:27" - DEBUG - [mqtt_connector.py] - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0x110a05610>, None, 8, 'Caught exception in on_message: Expecting property name enclosed in double quotes: line 1 column 2 (char 1)')"
""2020-03-10 17:23:28" - DEBUG - [mqtt_connector.py] - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0x110a05610>, None, 16, 'Sending PINGREQ')"
""2020-03-10 17:23:28" - DEBUG - [mqtt_connector.py] - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0x110a05610>, None, 16, 'Received PINGRESP')"
""2020-03-10 17:23:33" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, "Sending PUBLISH (d0, q1, r0, m450), 'b'v1/devices/me/telemetry'', ... (116 bytes)")"
""2020-03-10 17:23:33" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, 'Received PUBACK (Mid: 450)')"

zbeacon, 2020-03-10:
""2020-03-10 17:23:27" - DEBUG - [mqtt_connector.py] - mqtt_connector - 193 - (<paho.mqtt.client.Client object at 0x110a05610>, None, 8, 'Caught exception in on_message: Expecting property name enclosed in double quotes: line 1 column 2 (char 1)')"

It looks like something went wrong with message, please attach a message you sent.

Correct message will look like:
`
{
"serialNumber": "Test 1",
"sensorType": "default",
"temp": 26.5,
"sensorModel": "5F1C3E2A"
}
`

If you use mosquitto_pub - try to use the following command:
`
mosquitto_pub -h localhost -p 1893 -u mosquttio_test -P mosquttio_test -t '/sensor/data' -m '{ "serialNumber": "Test 1", "sensorType": "default", "temp": 26.5, "sensorModel": "5F1C3E2A" }'
`

Tracy6465, 2020-03-10:
@zbeacon 
I'm sorry, but I found that the content I copied from thingsboard's official website is in the wrong format.
`{“serialNumber”: “SN-001”, “sensorType”: “Thermometer”, “sensorModel”: “T1000”, “temp”: 42, “hum”: 58}`
I changed it to this：
`{"serialNumber":"SN-001","sensorType":"Thermometer","sensorModel":"T1000","temp":42,"hum":58}
`
the logs is：
`""2020-03-10 18:29:48" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, "Sending PUBLISH (d0, q1, r0, m1092), 'b'v1/gateway/attributes'', ... (30 bytes)")"
""2020-03-10 18:29:48" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, "Sending PUBLISH (d0, q1, r0, m1093), 'b'v1/gateway/telemetry'', ... (86 bytes)")"
""2020-03-10 18:29:48" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, 'Received PUBACK (Mid: 1092)')"
""2020-03-10 18:29:48" - DEBUG - [tb_client.py] - tb_client - 61 - (<paho.mqtt.client.Client object at 0x11098cbd0>, None, 16, 'Received PUBACK (Mid: 1093)')"`

closed 2020-03-10 by Tracy6465

zbeacon, 2020-03-10:
@Tracy6465 

Thanks, I will check it and fix it.