thingsboard/thingsboard-gateway #258: [BUG] Connectors are imported multiple times

acetylen, 2020-03-10:
Local implementations of thingsboard-provided connectors are overridden by the package provided ones.

I have implemented CustomSerialConnector and put my version in `/var/lib/thingsboard_gateway/serial/serialconnector.py`, as instructed in the documentation. When the gateway starts up, the following happens:

```
''2020-03-10 10:21:55' - ERROR - tb_utility - 88 - Import CustomSerialConnector failed, path /usr/local/lib/python3.7/dist-packages/thingsboard_gateway/connectors/serial doesn't exist'
''2020-03-10 10:21:55' - DEBUG - tb_utility - 68 - ModuleSpec(name='CustomSerialConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fb7b185ad68>, origin='/var/lib/thingsboard_gateway/serial/serialconnector.py')'
''2020-03-10 10:21:55' - DEBUG - tb_utility - 74 - <module 'CustomSerialConnector' from '/var/lib/thingsboard_gateway/serial/serialconnector.py'>'
''2020-03-10 10:21:55' - DEBUG - tb_utility - 68 - ModuleSpec(name='CustomSerialConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7fb7b07ef198>, origin='/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/extensions/serial/custom_serial_connector.py')'
''2020-03-10 10:21:55' - DEBUG - tb_utility - 74 - <module 'CustomSerialConnector' from '/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/extensions/serial/custom_serial_connector.py'>'
''2020-03-10 10:21:55' - DEBUG - tb_utility - 81 - Import CustomSerialConnector from /usr/local/lib/python3.7/dist-packages/thingsboard_gateway/extensions/serial.'
''2020-03-10 10:21:55' - INFO - custom_serial_connector - 36 - Starting Custom Custom Serial Connector connector'
```
If it's not obvious, what's happening here is that the gateway imports *every instance it can find* of CustomSerialConnector, with the last one taking priority. Since the last one defined is the one in the thingsboard_gateway library, overriding connectors as described in the documentation isn't actually possible.

**Versions (please complete the following information):**
 - OS: Debian 10
 - Thingsboard IoT Gateway version: 2.2.2
 - Python version: 3.7.3

zbeacon, 2020-03-10:
Hi @acetylen,

Thanks for your interest in the ThingsBoard IoT gateway.
This behavior caused by connector class names are duplicated (CustomSerialConnector is already in the gateway). 
To use your connector, please change your class name for connector and connector class parameter in the general configuration file ("tb_gateway.yaml"). 

Please try and give feedback.

acetylen, 2020-03-10:
Of course, that is what causes it. However, I have a feeling it's not intentional. Note in the log that `custom_serial_connector.py` is not imported from 'gateway/connectors/serial` but from `gateway/extensions/serial`. If I understand the documentation correctly, that path is for converters only?

The way you import modules is also not how it's usually done, which makes it harder to figure out what should really be happening. Since the gateway just continues on after this, I get an error further down the line when initializing a converter.

zbeacon, 2020-03-10:
The gateway first tries to import the default connectors. And if, after successful import, he finds a connector with the same name in the extensions folder, he will import it and overwrite the previous loaded connector. 
This logic is used because the "extensions" folder was created to store custom connectors and converters, and they can overwrite default connectors, if you need. 
The "connectors" folder uses for default connectors.

acetylen, 2020-03-10:
So you're shipping non-default connectors in the default distribution? Is that really intentional?

zbeacon, 2020-03-10:
@acetylen 

Yes, it is intentional, we want to indicate place for custom connectors.

acetylen, 2020-03-10:
The documentation indicates that they should be in `/var/lib/thingsboard-gateway`, but the file is in `/usr/local/lib/python3/dist-packages/thingsboard-gateway` if installed through pip.

zbeacon, 2020-03-10:
@acetylen  
Thanks, I will add option into the documentation as soon as possible.

closed 2020-04-28 by zbeacon