thingsboard/thingsboard-gateway #260: [BUG] Serial connector

acetylen, 2020-03-10:
**Describe the bug**
Clear and concise description of what the bug is.

**Connector name (If bug in the some connector):**
[e.g. OPC-UA Connector]

**Error traceback (If available):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.0]
 - Python version[e.g. 3.7]

zbeacon, 2020-03-10:
Hi @acetylen,

This connector created to explain how connectors work and it's general goal - show how to create custom connector. Also if you watch a little bit upper in code, you can see that this connector has for loop to loop over devices in the configuration:

```
            ...
            while True:
                for device in self.devices:
                    serial = self.devices[device]["serial"]
                    ...
```

Symbol "\n" is hard coded to simplify configuration in the example.

acetylen, 2020-03-10:
The outer loop will never advance while a read is in progress, and a read will never finish if a device disconnects.

I realize that the connector is an example but it should at least be a complete one.

closed 2020-05-05 by zbeacon