thingsboard/thingsboard-gateway #261: Problems with NaN responces from the device

pehedima, 2020-03-12:
I tried to read out an electrical meter, which is connected to one phase (L1) only.

During the reading of the current the following error occurs:

```
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'U_L1_V', 'type': 'double', 'functionCode': 3, 'address': 19000, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'U_L2_V', 'type': 'double', 'functionCode': 3, 'address': 19002, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'U_L3_V', 'type': 'double', 'functionCode': 3, 'address': 19004, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'P_L1_W', 'type': 'double', 'functionCode': 3, 'address': 19020, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'P_L2_W', 'type': 'double', 'functionCode': 3, 'address': 19022, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'P_L3_W', 'type': 'double', 'functionCode': 3, 'address': 19024, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'Q_L1_Var', 'type': 'double', 'functionCode': 3, 'address': 19036, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'Q_L2_Var', 'type': 'double', 'functionCode': 3, 'address': 19038, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'Q_L3_Var', 'type': 'double', 'functionCode': 3, 'address': 19040, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'S_L1_VA', 'type': 'double', 'functionCode': 3, 'address': 19028, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'S_L2_VA', 'type': 'double', 'functionCode': 3, 'address': 19030, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'S_L3_VA', 'type': 'double', 'functionCode': 3, 'address': 19032, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'PF_L1', 'type': 'double', 'functionCode': 3, 'address': 820, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'PF_L2', 'type': 'double', 'functionCode': 3, 'address': 822, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'PF_L3', 'type': 'double', 'functionCode': 3, 'address': 824, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'WH_L1_Wh', 'type': 'double', 'functionCode': 3, 'address': 19054, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'WH_L2_Wh', 'type': 'double', 'functionCode': 3, 'address': 19056, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'WH_L3_Wh', 'type': 'double', 'functionCode': 3, 'address': 19058, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'QH_L1_Warh', 'type': 'double', 'functionCode': 3, 'address': 19086, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'QH_L2_Warh', 'type': 'double', 'functionCode': 3, 'address': 19088, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'QH_L3_Warh', 'type': 'double', 'functionCode': 3, 'address': 19090, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 217 - With result ReadRegisterResponse (2)'
''2020-03-12 14:49:43' - DEBUG - modbus_connector - 216 - To modbus device J-Home-python-balena-serial,
{'tag': 'THDU_L1_proc', 'type': 'double', 'functionCode': 3, 'address': 19110, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
[... 30 lines cut ...]
''2020-03-12 14:49:44' - DEBUG - bytes_modbus_uplink_converter - 36 - Called convert function from BytesModbusUplinkConverter with args'
''2020-03-12 14:49:44' - DEBUG - bytes_modbus_uplink_converter - 37 - {'tag': 'U_L3_V', 'type': 'double', 'functionCode': 3, 'address': 19004, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-12 14:49:44' - DEBUG - bytes_modbus_uplink_converter - 38 - ReadRegisterResponse (2)'
[... cut ...]
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 128, in __process_devices
    converted_data = self.__devices[device]["converter"].convert(config=None, data=device_responses)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 118, in convert
    self.__result[config_data].append({tag: int(result, 16)})
TypeError: int() can't convert non-string with explicit base
```

Maybe someone can help me to handle such 'NaN' responces.
Thanks.

zbeacon, 2020-03-13:
Hi @pehedima 

Thank you for your feedback. I think I have fixed this, but unfortunately I don't have ability to check this on real device. Could you install a version from branch develop/2.4-python and check it? 
The issue with NaN will be resolved, probably. But it is possible to get new error with transferring data to the ThingsBoard. 

Anyway, please check and give feedback :)

pehedima, 2020-03-13:
Hi @zbeacon

Thanks for your support :)

I checked the new files from develop/2.4-python and there are additional errors refer to the 'handling' of the values:

I tested two different tags the first was 'U_L1_V'. This is a normal 'double' response and contains always a value. Non NaN value. The second tag ist 'I_L1_A' and that contains always a NaN value. 

Here the logfile for both cases: 
```
''2020-03-13 11:31:41' - DEBUG - bytes_modbus_uplink_converter - 55 - Tag: U_L1_V Config: {'tag': 'U_L1_V', 'type': 'double', 'functionCode': 3, 'address': 19000, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'} registers: [17247, 44758]'
''2020-03-13 11:31:41' - ERROR - modbus_connector - 177 - '
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 131, in __process_devices
    converted_data = self.__devices[device]["converter"].convert(config=None, data=device_responses)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 62, in convert
    decoded_data = self.__decode_from_registers(decoder, configuration)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 101, in __decode_from_registers
    assert decoder_functions.get(type_) is not None
AssertionError
''2020-03-13 11:31:46' - DEBUG - modbus_connector - 221 - To modbus device J-Home-python-balena-serial,
{'tag': 'U_L1_V', 'type': 'double', 'functionCode': 3, 'address': 19000, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-13 11:31:46' - DEBUG - modbus_connector - 222 - With result ReadRegisterResponse (2)'
''2020-03-13 11:31:46' - DEBUG - modbus_connector - 128 - Checking timeseries for device J-Home-python-balena-serial'
''2020-03-13 11:31:46' - DEBUG - modbus_connector - 130 - {'timeseries': {'U_L1_V': {'data_sent': {'tag': 'U_L1_V', 'type': 'double', 'functionCode': 3, 'address': 19000, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}, 'input_data': <pymodbus.register_read_message.ReadHoldingRegistersResponse object at 0x74f3ff10>}}, 'attributes': {}}'
''2020-03-13 11:31:46' - DEBUG - bytes_modbus_uplink_converter - 55 - Tag: U_L1_V Config: {'tag': 'U_L1_V', 'type': 'double', 'functionCode': 3, 'address': 19000, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'} registers: [17247, 50621]'
''2020-03-13 11:31:46' - ERROR - modbus_connector - 177 - '
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 131, in __process_devices
    converted_data = self.__devices[device]["converter"].convert(config=None, data=device_responses)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 62, in convert
    decoded_data = self.__decode_from_registers(decoder, configuration)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 101, in __decode_from_registers
    assert decoder_functions.get(type_) is not None
AssertionError
''2020-03-13 11:31:51' - DEBUG - modbus_connector - 221 - To modbus device J-Home-python-balena-serial,
{'tag': 'U_L1_V', 'type': 'double', 'functionCode': 3, 'address': 19000, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-13 11:31:51' - DEBUG - modbus_connector - 222 - With result ReadRegisterResponse (2)'
''2020-03-13 11:31:51' - DEBUG - modbus_connector - 128 - Checking timeseries for device J-Home-python-balena-serial'
''2020-03-13 11:31:51' - DEBUG - modbus_connector - 130 - {'timeseries': {'U_L1_V': {'data_sent': {'tag': 'U_L1_V', 'type': 'double', 'functionCode': 3, 'address': 19000, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}, 'input_data': <pymodbus.register_read_message.ReadHoldingRegistersResponse object at 0x74f3fb90>}}, 'attributes': {}}'
''2020-03-13 11:31:51' - DEBUG - bytes_modbus_uplink_converter - 55 - Tag: U_L1_V Config: {'tag': 'U_L1_V', 'type': 'double', 'functionCode': 3, 'address': 19000, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'} registers: [17248, 8149]'
''2020-03-13 11:31:51' - ERROR - modbus_connector - 177 - '
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 131, in __process_devices
    converted_data = self.__devices[device]["converter"].convert(config=None, data=device_responses)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 62, in convert
    decoded_data = self.__decode_from_registers(decoder, configuration)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 101, in __decode_from_registers
    assert decoder_functions.get(type_) is not None
AssertionError
''2020-03-13 11:31:56' - DEBUG - modbus_connector - 221 - To modbus device J-Home-python-balena-serial,
{'tag': 'U_L1_V', 'type': 'double', 'functionCode': 3, 'address': 19000, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
[... cut ...]
''2020-03-13 16:14:03' - DEBUG - bytes_modbus_uplink_converter - 39 - {'tag': 'I_L1_A', 'type': 'double', 'functionCode': 3, 'address': 19012, 'registerCount': 2, 'byteOrder': 'BIG', 'deviceName': 'J-Home-python-balena-serial'}'
''2020-03-13 16:14:03' - DEBUG - bytes_modbus_uplink_converter - 40 - ReadRegisterResponse (2)'
''2020-03-13 16:14:03' - DEBUG - bytes_modbus_uplink_converter - 117 - {'deviceName': 'J-Home-python-balena-serial', 'deviceType': 'ModbusDevice', 'telemetry': [{'I_L1_A': 0}], 'attributes': []}'

```
We should match the versions then everthing should work as expected.

commit ca43caff (2020-03-17): Fix #261
closed 2020-03-18 by zbeacon