thingsboard/thingsboard-gateway #262: One Bug about thingsboard-gateway  2.2.2.1

Ming-Shu, 2020-03-13:
Hi @zbeacon 

I'm trying  pre-release (2.2.2.1)  and discovering a old problem, just running thingsboard and BT gateway,and interrupt OPCUA server then restart OPCUA server, thingsboard  must stop update value .But I restart BT gateway ,it will back to normal status.I had remember this bug on version 1.0.7,and you fixed it on   version 1.0.9.
Hope you can fix it , thinks  : )

zbeacon, 2020-03-13:
Hi @Ming-Shu  

Please add logs from file "connector.log" in period when the problem occurs..

Ming-Shu, 2020-03-13:
Hi @zbeacon 

A part of connector.log below:

''2020-03-13 12:59:49' - DEBUG - opcua_connector - 466 - Python: New data change event on node Node(StringNodeId(ns=1;s=T001)), with val: 18708'
''2020-03-13 12:59:49' - DEBUG - opcua_connector - 472 - Data to ThingsBoard: {'deviceName': 'Device', 'deviceType': 'default', 'attributes': [{'T001': '18708'}], 'telemetry': [{'T001': '18708'}]}'
''2020-03-13 12:59:50' - DEBUG - opcua_connector - 466 - Python: New data change event on node Node(StringNodeId(ns=1;s=T001)), with val: 18709'
''2020-03-13 12:59:50' - DEBUG - opcua_connector - 472 - Data to ThingsBoard: {'deviceName': 'Device', 'deviceType': 'default', 'attributes': [{'T001': '18709'}], 'telemetry': [{'T001': '18709'}]}'
''2020-03-13 12:59:50' - DEBUG - opcua_connector - 466 - Python: New data change event on node Node(StringNodeId(ns=1;s=T001)), with val: 18709'
''2020-03-13 12:59:50' - DEBUG - opcua_connector - 472 - Data to ThingsBoard: {'deviceName': 'Device', 'deviceType': 'default', 'attributes': [{'T001': '18709'}], 'telemetry': [{'T001': '18709'}]}'
''2020-03-13 12:59:51' - DEBUG - opcua_connector - 466 - Python: New data change event on node Node(StringNodeId(ns=1;s=T001)), with val: 18710'
''2020-03-13 12:59:51' - DEBUG - opcua_connector - 472 - Data to ThingsBoard: {'deviceName': 'Device', 'deviceType': 'default', 'attributes': [{'T001': '18710'}], 'telemetry': [{'T001': '18710'}]}'
''2020-03-13 12:59:51' - DEBUG - opcua_connector - 466 - Python: New data change event on node Node(StringNodeId(ns=1;s=T001)), with val: 18710'
''2020-03-13 12:59:51' - DEBUG - opcua_connector - 472 - Data to ThingsBoard: {'deviceName': 'Device', 'deviceType': 'default', 'attributes': [{'T001': '18710'}], 'telemetry': [{'T001': '18710'}]}'
''2020-03-13 12:59:57' - ERROR - opcua_connector - 146 - Connection refused on connection to OPC-UA server with url localhost:49320'
''2020-03-13 13:00:07' - ERROR - opcua_connector - 146 - Connection refused on connection to OPC-UA server with url localhost:49320'
''2020-03-13 13:00:17' - ERROR - opcua_connector - 146 - Connection refused on connection to OPC-UA server with url localhost:49320'
''2020-03-13 13:00:28' - ERROR - opcua_connector - 146 - Connection refused on connection to OPC-UA server with url localhost:49320'
''2020-03-13 13:00:38' - ERROR - opcua_connector - 146 - Connection refused on connection to OPC-UA server with url localhost:49320'
''2020-03-13 13:00:48' - ERROR - opcua_connector - 146 - Connection refused on connection to OPC-UA server with url localhost:49320'
''2020-03-13 13:02:49' - DEBUG - tb_logger - 37 - Added remote handler to log extension'
''2020-03-13 13:02:49' - DEBUG - tb_logger - 37 - Added remote handler to log converter'
''2020-03-13 13:02:49' - DEBUG - tb_logger - 37 - Added remote handler to log connector'
''2020-03-13 13:02:51' - INFO - opcua_connector - 90 - Starting OPC-UA Connector'
''2020-03-13 13:02:51' - DEBUG - opcua_connector - 101 - uri://MatrikonOpcUaServer'
''2020-03-13 13:02:51' - DEBUG - opcua_connector - 102 - 1'
''2020-03-13 13:02:51' - INFO - opcua_connector - 115 - OPC-UA connector OPC-UA Connector connected to server localhost:49320'
''2020-03-13 13:02:51' - DEBUG - opcua_connector - 367 - Device name: Device'
''2020-03-13 13:02:51' - DEBUG - opcua_connector - 270 - Node for attributes "T001" with path: Root\\.Objects\\.OnsFolder\\.T001 - FOUND! Current values is: 18890'
''2020-03-13 13:02:51' - DEBUG - opcua_connector - 295 - Added subscription to node: Node(StringNodeId(ns=1;s=T001))'
''2020-03-13 13:02:51' - DEBUG - opcua_connector - 296 - Data to ThingsBoard: {'deviceName': 'Device', 'deviceType': 'default', 'attributes': [{'T001': '18890'}], 'telemetry': [{'T001': '18890'}]}'
''2020-03-13 13:02:51' - DEBUG - opcua_connector - 466 - Python: New data change event on node Node(StringNodeId(ns=1;s=T001)), with val: 18890'
''2020-03-13 13:02:51' - DEBUG - opcua_connector - 472 - Data to ThingsBoard: {'deviceName': 'Device', 'deviceType': 'default', 'attributes': [{'T001': '18890'}], 'telemetry': [{'T001': '18890'}]}' 

**Setp 1.**  OPCUA server is running, log ------->New data change event on node Node(StringNodeId(ns=1;s=T001)), with val: 18708'......etc

**Setp 2.**  interrupt OPCUA server,  log --------> Connection refused on connection to OPC-UA server with url localhost:49320' (My OPC-UA server install local this time)

**Setp 3.**  restart OPCUA server,  log --------> 
''2020-03-13 13:02:49' - DEBUG - tb_logger - 37 - Added remote handler to log extension'
''2020-03-13 13:02:49' - DEBUG - tb_logger - 37 - Added remote handler to log converter'
''2020-03-13 13:02:49' - DEBUG - tb_logger - 37 - Added remote handler to log connector'

''2020-03-13 13:02:51' - INFO - opcua_connector - 90 - Starting OPC-UA Connector'
''2020-03-13 13:02:51' - DEBUG - opcua_connector - 101 - uri://MatrikonOpcUaServer'
''2020-03-13 13:02:51' - DEBUG - opcua_connector - 102 - 1'
''2020-03-13 13:02:51' - INFO - opcua_connector - 115 - OPC-UA connector OPC-UA Connector connected to server localhost:49320'
''2020-03-13 13:02:51' - DEBUG - opcua_connector - 367 - Device name: Device'
''2020-03-13 13:02:51' - DEBUG - opcua_connector - 270 - Node for attributes "T001" with path: Root\\.Objects\\.OnsFolder\\.T001 - FOUND! Current values is: 18890'
''2020-03-13 13:02:51' - DEBUG - opcua_connector - 295 - Added subscription to node: Node(StringNodeId(ns=1;s=T001))'
''2020-03-13 13:02:51' - DEBUG - opcua_connector - 296 - Data to ThingsBoard: {'deviceName': 'Device', 'deviceType': 'default', 'attributes': [{'T001': '18890'}], 'telemetry': [{'T001': '18890'}]}'

zbeacon, 2020-03-13:
Hi @Ming-Shu 

It was problem with subscriptions. I pushed fix in the development branch (develop/2.4-python). Could you install a version from the development branch and check it? Or you also can wait next release, with this fix.

Ming-Shu, 2020-03-16:
Hi @zbeacon  

I will wait next release,thank you for your help. : )

closed 2020-03-20 by Ming-Shu