thingsboard/thingsboard-gateway #264: About thingsboard-gateway 2.2.3.2 test

Ming-Shu, 2020-03-20:
Hi @zbeacon

I'm trying pre-release (2.2.3.2) and discovering maybe a old problem!

I have three opc-ua server

1. localhost test
2. raspberry pi 
3. IPC

I running thingsboard and BT gateway,and interrupt OPCUA server then restart OPCUA server,then
1. localhost test is OK,it will back to normal status.
2. raspberry pi  and IPC is dead,and must  restart BT gateway or runing cmd

sudo python3 -c 'from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService; TBGatewayService("tb_gateway.yaml")'

Log below:
[connector.log](https://github.com/thingsboard/thingsboard-gateway/files/4358640/connector.log)

Hope you can fix it again , thinks ><~~~~

zbeacon, 2020-03-20:
Hi @Ming-Shu,

Thanks for your feedback.
Unfortunately, there was no changes in the OPC-UA Connector in this release. It looks like a new issue. I will try to repeat this and fix it.

zbeacon, 2020-03-20:
Could you try to repeat this situation by waiting about a minute before restarting? - The situation may be caused by some timeouts at the network level.

Ming-Shu, 2020-03-23:
OK ,I will try it and feedback for you.

Ming-Shu, 2020-03-23:
Hi @zbeacon  

I got some test result that feedback you.

This test one OPC UA server on IPC,and Node number is 60 . 
All problem below:
1.  I was waiting about a 55 minute at  first time connect (thingsboard show value,but runing cmd sudo python3 -c 'from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService; TBGatewayService("tb_gateway.yaml")' is show correct value ),so I reduced node from 60 to 10 in the .josn file. Result, I only need to wait about 5 minute.
2. If interrupt OPCUA server then restart OPCUA server,thingsboard must don't update value no matter how long.However,I restart BT gateway or runing cmd is update value on thingsboard web.But if I cancel runing cmd after update value that will  don't update value on thingsboard .

So,I suggest that is probably have a bug  in  BT gateway  connect  to thingsboard.

I hope this could be help : )

Ming-Shu, 2020-03-23:
@zbeacon 

I record some LOG 
[connector.log](https://github.com/thingsboard/thingsboard-gateway/files/4367389/connector.log)

16116 line is value at last ,then  restart OPCUA server.

16188 line is update  value but thingsboard web no any  update.

zbeacon, 2020-03-23:
Hi @Ming-Shu,

Thanks, I am investigating this issue.

Ming-Shu, 2020-03-24:
OK ,I will try it again,thank you for your help :)

Ming-Shu, 2020-03-24:
> Hi @Ming-Shu,
> 
> Thanks, I am investigating this issue.

OK ,I will try it again,thank you for your help :)

Ming-Shu, 2020-03-25:
Hi @zbeacon 

I have some test result and feedback  you.

Actually, you have  reached   restart OPCUA server and TB gateway normal work,but don't immediate updating on thingsboard web again  , it update about 25 minutes now .In addition, If  OPC UA server have node number is 60 has become first time connect must No updating on   thingsboard web.

Node only 10 number ,but don't immediate updating on thingsboard web again below:
Look line 84613

 [image]

Look line 84656 of T0101 and   thingsboard web  of T0101 
[image]

Look line 85207
 [image]
 of T0101 and   thingsboard web  of T0101 

Node  60 number ,but first time connect must No updating on thingsboard web again below:

[image]

zbeacon, 2020-03-25:
@Ming-Shu

zbeacon, 2020-03-25:
Hi @Ming-Shu 

Thanks for you feedback.
It can be caused by logic when the gateway sends attributes and telemetry to the ThingsBoard using pack of value. Try to set parameter "read_records_count" in the general configuration file to 1 and try again.

The speed of reconnect processing so slow, because the gateway read all required nodes on server every time, when connected. Unfortunately, they can be changed during server disconnected time and if the gateway won't read nodes on connect, it won't know actually state of the OPC-UA server. Also, if you use subscription model, it must subscribes to get information from interested nodes. I have idea, how to solve this. 
Probably, it will be solved, if we will read again interesting node when exception is being raised. I need to do some research and change logic of subscribing and reading.

Ming-Shu, 2020-03-25:
Hi @zbeacon 

Thank you for your prompt reply,I look forward to trying your next version : )

Ming-Shu, 2020-03-26:
Hi @zbeacon 

Today, I noticed something interesting.I create two (XXX.json)  to read some node(T0101),and use 'Prosys OPC UA Client' too.

A.json screenshot below:
[image]

B.json screenshot below:
[image]

A.json time is 10:38,and value is 2155 .But,
B.json time is 10:37,and value is 2764
Unm.....................Actually,'Prosys OPC UA Client' value is 2832 at 10:39.

closed 2020-03-31 by Ming-Shu