thingsboard/thingsboard-gateway #266: ERROR - [modbus_connector.py] - modbus_connector - 177 - 'ModbusIOException' object has no attribute 'bits'

bashole, 2020-03-25:
### **Describe**
Error happen when using modbus connector

### **Evn Versions**
 - OS: [Ubuntu 18.04]
 - Thingsboard IoT Gateway version [master branch]
 - Python version[ 3.7]

### **Error traceback**
I follow the installation instruction step by step:

https://thingsboard.io/docs/iot-gateway/install/source-installation/

my cofiguration files are listed below:

**<tb_gateway.yaml>**
thingsboard:
  host: **my-site******
  port: 1883
  remoteConfiguration: false
  security:
    accessToken: **x0W**********
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000

connectors:

  -
    name: Modbus Connector
    type: modbus
    configuration: modbus_serial.json

**<modbus_serial.json>**
{
  "server": {
    "name": "Modbus Default Server",
    "type": "serial",
    "method": "rtu",
    "port": "/dev/ttyUSB0",
    "baudrate": **9600**,
    "timeout": 35,
    "devices": [
      {
        "unitId": 1,
        "deviceName": "**mc power state**",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": true,
        "attributes": [
          {
            "byteOrder": "BIG",
            "tag": "test",
            "type": "long",
            "functionCode": **1**,
            "registerCount": 1,
            "address": **16385**
          }
        ]
      }
    ]
  }
}

when running the gateway,corresponse error were appear.and here is the logfiles:
**<tb_connection.log>**
""2020-03-25 21:49:55" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 175 - Subscribed to *|* with id 1"
""2020-03-25 21:49:57" - ERROR - [tb_device_mqtt.py] - tb_device_mqtt - 202 - connection FAIL with error 5 not authorised"
""2020-03-25 21:49:57" - INFO - [tb_client.py] - tb_client - 81 - TB client <paho.mqtt.client.Client object at 0x7f0948d0cd30> has been disconnected."

**<service.log>**
""2020-03-25 21:49:55" - INFO - [tb_gateway_service.py] - tb_gateway_service - 50 - Gateway starting..."
""2020-03-25 21:49:56" - WARNING - [tb_gateway_service.py] - tb_gateway_service - 561 - Device mc power state connector not found, maybe it had been disabled."
""2020-03-25 21:49:56" - INFO - [tb_gateway_service.py] - tb_gateway_service - 108 - Gateway started."
""2020-03-25 21:51:31" - INFO - [tb_gateway_service.py] - tb_gateway_service - 152 - Stopping..."
""2020-03-25 21:51:31" - INFO - [tb_gateway_service.py] - tb_gateway_service - 154 - The gateway has been stopped."

**<connector.log>**
""2020-03-25 21:49:56" - INFO - [modbus_connector.py] - modbus_connector - 57 - Starting Modbus connector"
""2020-03-25 21:49:56" - INFO - [modbus_connector.py] - modbus_connector - 63 - Modbus connected."
""2020-03-25 21:50:31" - ERROR - [modbus_connector.py] - modbus_connector - 177 - 'ModbusIOException' object has no attribute 'bits'"
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.2.3.2-py3.6.egg/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 131, in __process_devices
    converted_data = self.__devices[device]["converter"].convert(config=None, data=device_responses)
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.2.3.2-py3.6.egg/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 42, in convert
    result = response.bits
AttributeError: 'ModbusIOException' object has no attribute 'bits'
""2020-03-25 21:51:06" - ERROR - [modbus_connector.py] - modbus_connector - 177 - 'ModbusIOException' object has no attribute 'bits'"
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.2.3.2-py3.6.egg/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 131, in __process_devices
    converted_data = self.__devices[device]["converter"].convert(config=None, data=device_responses)
  File "/usr/local/lib/python3.6/dist-packages/thingsboard_gateway-2.2.3.2-py3.6.egg/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 42, in convert
    result = response.bits
AttributeError: 'ModbusIOException' object has no attribute 'bits'
""2020-03-25 21:51:31" - INFO - [modbus_connector.py] - modbus_connector - 101 - Modbus Connector has been stopped."

Any help is appreciated. Thanks in advance.

zbeacon, 2020-03-25:
Hi @bashole 

Thanks for your interest in the ThingsBoard IoT gateway.
This problem caused by wrong type in the configuration, please chenge the type from "long" to "bit" and try again. Also please let me know about results.

bashole, 2020-03-26:
Hi @zbeacon 
Thank you for your support.

Same errors still exist after changing the type to "bit".
Btw, what is the most possible reasons cause the error logs "TB client <paho.mqtt.client.Client object at 0x7f0948d0cd30> has been disconnected."" in **<tb_connection.log>** and "WARNING - [tb_gateway_service.py] - tb_gateway_service - 561 - Device mc power state connector not found, maybe it had been disabled."" in **<service.log>** ?

The modbus server which I connected to has the serial port cfg below：
 baudrate		: 9600
data length	: 8
start bit		: 1
stop bit		: 1
parity		: **even**
check sum	: none
rs/cs flow control	: none
address		: 1

Does the above have any influence on the communication?

I download a test tool below:
http://www.modbusdriver.com/modpoll.html

The command of mine is:
sudo ./modpoll -b 9600 -t 0 -r 16385 -l 100 /dev/ttyUSB0

Test result is ok.

zbeacon, 2020-03-26:
Thanks for your reply, please look for exceptions in your connector.log file and copy them here, I will try to understand the problem.

bashole, 2020-03-26:
here is my current logs:
[tb_connection.log](https://github.com/thingsboard/thingsboard-gateway/files/4388018/tb_connection.log)
[service.log](https://github.com/thingsboard/thingsboard-gateway/files/4388021/service.log)
[connector.log](https://github.com/thingsboard/thingsboard-gateway/files/4388023/connector.log)

here is my latest configs:
**<tb_gateway.yaml>**
thingsboard:
  host: 127.0.0.1
  port: 1883
  remoteConfiguration: false
  security:
    accessToken: MgANcxztUpD5d9XwFpxk
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000

connectors:

  -
    name: bb
    type: modbus
    configuration: modbus_serial.json

**<modbus_serial.json>**
{
  "server": {
    "name": "Modbus Default Server",
    "type": "serial",
    "method": "rtu",
    "port": "/dev/ttyUSB0",
    "baudrate": 9600,
    "parity": "PARITY_EVEN",
    "timeout": 35,
    "devices": [
      {
        "unitId": 2,
        "deviceName": "bb",
        "attributesPollPeriod": 100,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
        "attributes": [
          {
            "byteOrder": "LITTLE",
            "tag": "test",
            "type": "bit",
            "functionCode": 1,
            "registerCount": 1,
            "address": 16385
          }
        ]
      }
    ]
  }
}

[image]

bashole, 2020-03-29:
1. **ERROR - [modbus_connector.py] - modbus_connector - 177 - 'ModbusIOException' object has no attribute 'bits'**
The error has been fixed by set my modbus server device's parity even to none, because the pymodbus's default parity setting is none, and tb-gateway has no interface to modify it.
I think it is better to give the interface to modify parity argument to add more flexibility.
Plus: change attributes type from 'long' to 'bits' in <modbus-serial.json>is also nessesary [note: 'bits' not 'bit'] 

2. **WARNING - [tb_gateway_service.py] - tb_gateway_service - 561 - Device mc power state connector not found, maybe it had been disabled."**
This warning is weird, I changed serve name from the default 'Modbus Default Server' to my specified connector name 'dd', then the warning was gone away.
That is to say, I have to make the connnector name in <tb.gateway.yaml> and the server name in <modbus-serial.json> to be the same.

3. **INFO - [tb_client.py] - tb_client - 81 - TB client <paho.mqtt.client.Client object at 0x7f0948d0cd30> has been disconnected."**
After tb-gateway working with my modbus server device, it couldn't connnet to the thingboard server. I thought it will have some relationships with the way I establishing the tb server. I established the tb server by docker msa, it may cause some 503 issues , so I change the normal installation way which is metioned at tb documentation "Installing ThingsBoard CE on Ubuntu Server", then it was successed! However that was still not very stable, and this INFO still exists.

closed 2020-03-29 by bashole