thingsboard/thingsboard-gateway #267: TB Gateway 2.2.4 test feedback

Ming-Shu, 2020-03-31:
Hi @zbeacon 

I am done some test result below:

1. I have a localhost opcua server virtual machine and IPC  opcua server  real machine.
2.  localhost opcua server test is so perfect, restart opcua server and  read value is no problem.
3.Unfortunately, IPC  opcua server is completely different than localhost opcua.
   I create 5 opcua.json that set read_records_count: 1,max_records_count: 100 and  enable on tb_gateway.yaml.
   
  one.json is 60 nodes,two.json is 10 nodes, A.json  nodes is 8,and B.json is 12 ,C.json nodes is 15 .
  
result:
 one.json is very long time  to connect thingsboard web at first time,it about 35 minutes.
 two.json  to connect thingsboard web about 4.5 minutes.
 A.json  to connect thingsboard web about 4 minutes.
 B.json  to connect thingsboard web about 4 minutes.
 C.json  to connect thingsboard web about 6 minutes.

but all don't do reconnect  when  opcua server restart on this version.

I have some picture about log below:

[image]

The terminal value and web value are the same and time is too(T0101 is 4222).

-----------------------------------------------------------------------------------------------------------------------------------
[image]

The terminal value and web value are not the same. ( terminal  T0101 is 4228,web T0101 is 4222). 

-----------------------------------------------------------------------------------------------------------------------------------

[image]

attritubes value get empty  
-----------------------------------------------------------------------------------------------------------------------------------
 
[image]

one.json terminal value  is 5731,one.json web value  is 4222 and two.json web value is 5128.

The above is my test result for you : )

Ming-Shu, 2020-04-01:
Hi,@zbeacon

I did some tests,have been observed  in all test.
so far,the best stable version is 2.2.2.1,because do the same test as above that two.json and A.json and B.json and C.json read value is same and time is same too.But  test is no OK about restart opcua server and 60 nodes in IPC. Maybe  for your reference.  : )

Notes: If use  "Fix for #262 Bug was with subscriptions behavior model
355212d " on localhost opcua server test is so perfect,but only for this.

zbeacon, 2020-04-03:
Hi @Ming-Shu 

Thank you for your feedback, but I have not solved this problem yet, I will write to you when it is fixed and you can check.

Ming-Shu, 2020-04-03:
That's OK.Thank you for your reply,I hope everything is going well with you. : )

chad007, 2020-04-17:
Hi @Ming-Shu 

I wonder which version of TB are you using to test with TB Gateway 2.2.4. 

I encounter a problem when using TB 2.4.2 with TB Gateway 2.2.4.1. Although the TB Gateway is able to connect to the mqtt port 1883 of TB successfully, it fails to subscribe any of the mqtt topics:

`tb-gateway_1  | ''2020-04-17 13:33:01' - ERROR - mqtt_connector - 190 - "MQTT Broker Connector" subscription failed to topic /sensor/data subscription message id = 1'`
`tb-gateway_1  | ''2020-04-17 13:33:01' - ERROR - mqtt_connector - 190 - "MQTT Broker Connector" subscription failed to topic /sensor/+/data subscription message id = 2'`
`tb-gateway_1  | ''2020-04-17 13:33:01' - ERROR - mqtt_connector - 190 - "MQTT Broker Connector" subscription failed to topic /custom/sensors/+ subscription message id = 3'`
`tb-gateway_1  | ''2020-04-17 13:33:01' - ERROR - mqtt_connector - 190 - "MQTT Broker Connector" subscription failed to topic sensor/connect subscription message id = 4'`
`tb-gateway_1  | ''2020-04-17 13:33:01' - ERROR - mqtt_connector - 190 - "MQTT Broker Connector" subscription failed to topic sensor/+/connect subscription message id = 5'`
`tb-gateway_1  | ''2020-04-17 13:33:01' - ERROR - mqtt_connector - 190 - "MQTT Broker Connector" subscription failed to topic sensor/disconnect subscription message id = 6'`
`tb-gateway_1  | ''2020-04-17 13:33:01' - ERROR - mqtt_connector - 190 - "MQTT Broker Connector" subscription failed to topic sensor/+/disconnect subscription message id = 7'

Thanks!
`

Ming-Shu, 2020-04-18:
Hi @chad007

I suggest that you could check json file at config folder : )

chad007, 2020-04-18:
@Ming-Shu Thank you for your suggestion.
I  checked "config/tb_gateway.yaml" and found that I was using the mqtt connector in the TB Gateway for a modbus gateway configured in TB. I switched the connector from mqtt to modbus, and the error described above resolved.

samson0v, 2022-01-20:
Hi @Ming-Shu, there are fixes for the OPC-UA speed, so you can update your Gateway to the newest version via the master branch and retest it.