thingsboard/thingsboard-gateway #268: [BUG] mqtt.json - access to json object attributes

affoltep, 2020-03-31:
**Describe the bug**
I am not able to get data out of the object metadata:
`{
  "metadata": {
      "value1": "hello world",
      "value2": 123
   }
  }
`
by using the following syntax in mqtt.json

`
...
"timeseries": [
  {
      "type": "string",
      "key": "value1",
      "value": "${metadata.value1}"
   }
`
What should I use instead of ${metadata.value1}?

thingsboard-gateway 2.2.4

zbeacon, 2020-03-31:
Hi @affoltep 

Thank you for your interest in the ThingsBoard IoT gateway.
You try to use JSONPath Expression, but you need to give root position in it (*$.*). 
To solve this issue just replace `"${metadata.value1}"` with `"${$.metadata.value1}"`.

affoltep, 2020-03-31:
Hi @zbeacon
Thank you for your assistance. I tried `${$.metatdata.value1} before, but it doesn't work:
with this mqtt.json:
`{
  "broker": {
    "name":"demo_weatherstation",
    "host":"eu.thethings.network",
    "port":1883,
    "security": {
      "type": "basic",
      "username": "openthings_meteo_001",
      "password": "**********"
    }
  },
  "mapping": [
    {
      "topicFilter": "openthings_meteo_001/devices/+/up",
      "converter": {
        "type": "json",
        "filterExpression": "",
        "deviceNameJsonExpression": "${dev_id}",
        "deviceTypeJsonExpression": "${$.payload_fields.dev_type}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "dev_eui",
            "value": "${hardware_serial}"
          },{
            "type": "string",
            "key": "app_id",
            "value": "${app_id}"
          },{
            "type": "long",
            "key": "port",
            "value": "${port}"
          },{
            "type": "long",
            "key": "counter",
            "value": "${counter}"
          },{
            "type": "double",
            "key": "pos_latitude",
            "value": "${$.metadata.latitude}"
          },{
            "type": "double",
            "key": "pos_longitude",
            "value": "${$.metadata.longitude}"
          },{
            "type": "string",
            "key": "dev_id",
            "value": "${dev_id}"
          }
        ],
        "timeseries": [
         {
           "type": "double",
           "key": "air_dewpoint",
           "value": "${$.payload_fields.air_dew}"
         },{
           "type": "long",
           "key": "air_humidity",
           "value": "${$.payload_fields.air_humid}"
         },{
           "type": "double",
           "key": "air_temperature",
           "value": "${$.payload_fields.air_temp}"
         },{
           "type": "double",
           "key": "battery",
[... 90 lines cut ...]
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.payload_fields.air_humid''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.payload_fields.air_temp''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.payload_fields.batt''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.payload_fields.delta_t''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.payload_fields.raincount''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.payload_fields.soil_temp''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.metadata.time''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.metadata.frequency''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.metadata.modulation''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.metadata.data_rate''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.metadata.coding_rate''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.metadata.gateways[*].gtw_id''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.metadata.gateways[*].timestamp''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.metadata.gateways[*].time''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.metadata.gateways[*].channel''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.metadata.gateways[*].rssi''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.metadata.gateways[*].snr''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.metadata.gateways[*].latitude''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.metadata.gateways[*].longitude''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.metadata.gateways[*].altitude''
''2020-03-31 11:59:56' - ERROR - tb_utility - 119 - '$.metadata.airtime''
''2020-03-31 11:59:56' - ERROR - tb_utility - 35 - deviceType is empty in data: {"deviceName": "openthings_weatherstation_001", "deviceType": null, "attributes": [{"dev_eui": "70B3D5499B62C911"}, {"app_id": "openthings_meteo_001"}, {"port": 1}, {"counter": 0}, {"pos_latitude": null}, {"pos_longitude": null}, {"dev_id": "openthings_weatherstation_001"}], "telemetry": [{"air_dewpoint": null}, {"air_humidity": null}, {"air_temperature": null}, {"battery": null}, {"delta_t": null}, {"raincount": null}, {"soil_temperature": null}, {"time": null}, {"frequency": null}, {"modulation": null}, {"data_rate": null}, {"coding_rate": null}, {"gtw_id": null}, {"gtw_timestamp": null}, {"gtw_time": null}, {"gtw_channel": null}, {"gtw_rssi": null}, {"gtw_snr": null}, {"gtw_latitude": null}, {"gtw_longitude": null}, {"gtw_altitude": null}, {"airtime": null}]}'
''2020-03-31 11:59:56' - ERROR - tb_gateway_service - 179 - Data from demo_weatherstation connector is invalid.'
`
Any ideas?

zbeacon, 2020-03-31:
Problem is: ERROR - tb_utility - 35 - deviceType is empty in data

It looks like there is no **dev_type** in response. For test purposes you can replace `"deviceTypeJsonExpression": "${$.payload_fields.dev_type}",` with `"deviceTypeJsonExpression": "default",`.

affoltep, 2020-03-31:
@zbeacon - It's bizarre. Even I use `"deviceTypeJsonExpression": "default",` 

> the ERROR - tb_utility - 35 - deviceType is empty in data 
remains

affoltep, 2020-03-31:
...first ERROR I get:

> ''2020-03-31 12:24:49' - ERROR - tb_utility - 119 - 'default''

affoltep, 2020-03-31:
I'm a step further:
If I use `"deviceTypeJsonExpression": "${app_id}"` it does not fail with deviceType empty.
If I use `"deviceTypeJsonExpression": "${app_id}"` it does fail with ERROR tb_utility -35

lines similar to `"value": "${$.metadata.latitude}"` are failing always with:
> ''2020-03-31 12:41:19' - ERROR - tb_utility - 119 - '$.metadata.latitude''

zbeacon, 2020-03-31:
Please update to the latest version and try again. [Guide](https://thingsboard.io/docs/iot-gateway/install/upgrade-instructions/)

We will check and update version in the docker.

affoltep, 2020-03-31:
Looks like I already use the latest:

sudo docker pull thingsboard/tb-gateway
Using default tag: latest
latest: Pulling from thingsboard/tb-gateway
Digest: sha256:ca9ef1e2e228a649842cc5ebef92973181f1c29ee143bbc23581d001bd3f48df
Status: Image is up to date for thingsboard/tb-gateway:latest
docker.io/thingsboard/tb-gateway:latest

affoltep, 2020-03-31:
...ah, probably I was too early...will try to update a little bit later

affoltep, 2020-04-01:
it seems to be the same problem as #255 
I have no glue whether tb-gateway 2.2 version is still used at docker. Actually the docjer version is actual:
`sudo docker pull thingsboard/tb-gateway
Using default tag: latest
latest: Pulling from thingsboard/tb-gateway
Digest: sha256:ca9ef1e2e228a649842cc5ebef92973181f1c29ee143bbc23581d001bd3f48df
Status: Image is up to date for thingsboard/tb-gateway:latest
docker.io/thingsboard/tb-gateway:latest`
but it doesn't still work

zbeacon, 2020-04-03:
Hi @affoltep 

We have released a new version 2.2.4.1 - Could you try to install it. Docker is also updated.

affoltep, 2020-04-03:
Hi @zbeacon - I could do the installation - thx
But it is not really better. I don't get any ERROR yet, but it looks like it as well doesn't detect any message anymore..
connector.log:
`''2020-04-03 20:17:27' - INFO - mqtt_connector - 138 - MQTT Broker Connector connected to eu.thethings.network:1883 - successfully.'
''2020-04-03 20:17:27' - DEBUG - mqtt_connector - 139 - Client <paho.mqtt.client.Client object at 0x7f11ac6a8710>, userdata None, flags {'session present': 0}, extra_params ()'
''2020-04-03 20:17:27' - INFO - mqtt_connector - 162 - Connector "MQTT Broker Connector" subscribe to openthings_meteo_001/devices/+/up'
''2020-04-03 20:17:28' - INFO - mqtt_connector - 192 - "MQTT Broker Connector" subscription success to topic openthings_meteo_001/devices/+/up, subscription message id = 1'`
converter.log
`''2020-04-03 20:17:27' - DEBUG - tb_logger - 37 - Added remote handler to log converter'
`
service.log
`''2020-04-03 20:17:27' - INFO - tb_gateway_service - 52 - Gateway starting...'
''2020-04-03 20:17:27' - INFO - tb_gateway_service - 54 - ThingsBoard IoT gateway version: 2.2.4.1'
''2020-04-03 20:17:27' - DEBUG - tb_logger - 37 - Added remote handler to log service'
''2020-04-03 20:17:27' - DEBUG - tb_utility - 70 - ModuleSpec(name='MqttConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f11ac6ff898>, origin='/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/json_mqtt_uplink_converter.py')'
''2020-04-03 20:17:27' - DEBUG - tb_utility - 77 - <module 'MqttConnector' from '/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/json_mqtt_uplink_converter.py'>'
''2020-04-03 20:17:27' - DEBUG - tb_utility - 70 - ModuleSpec(name='MqttConnector', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f11ac6ff160>, origin='/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/mqtt_connector.py')'
''2020-04-03 20:17:27' - DEBUG - tb_utility - 77 - <module 'MqttConnector' from '/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/mqtt_connector.py'>'
''2020-04-03 20:17:27' - DEBUG - tb_utility - 81 - Import MqttConnector from /usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt.'
''2020-04-03 20:17:27' - DEBUG - tb_gateway_service - 551 - Loaded devices:
 {}'
''2020-04-03 20:17:27' - DEBUG - tb_gateway_service - 296 - Send data Thread has been started successfully.'
''2020-04-03 20:17:27' - INFO - tb_gateway_service - 112 - Gateway started.'
`
Nothing else happens, even if I send a message. I didn't have done any changes on mqtt.json

zbeacon, 2020-04-09:
Hi @affoltep 

Thank you for your patience, please check the logs during startup and sending data through the gateway (for verification, use the command `docker attach tb-gateway` after the command` docker start`). It will show you all the information about the launch in real time, and also, please send a message to the topic and check the output, this may cause some errors during processing. If there are any errors, we can investigate this. If the gateway does not respond to the message, this means that it is not subscribed to the desired topic. In this case, please share the command you sent and mqtt.json configuration file.

affoltep, 2020-04-09:
Hi @zbeacon 
This is exactly my problem. There are no more debug information as I published above.
The only I get from connector.log is:
`''2020-04-09 21:27:46' - DEBUG - mqtt_connector - 185 - (<paho.mqtt.client.Client object at 0x7fa5e0fe22b0>, None, 16, "Received PUBLISH (d0, q0, r0, m0), 'openthings_meteo_001/devices/openthings_weatherstation_001/up', ...  (396 bytes)")'
`

zbeacon, 2020-04-10:
@affoltep 

Ok, the gateway successfully subscribed to topic openthings_meteo_001/devices/openthings_weatherstation_001/up and received messages.

But you didn't provide configuration (mapping for this toping from mqtt.json file) and message you have been sent. I cannot help you without these information.

closed 2020-04-21 by zbeacon